daniestevez/gr-satellites #211: clang-cl support

gvanem, 2020-12-20:
A simple "failure" in `lib/golay24.c` using clang-cl on Windows:
```
golay24.c(28,24): error: functions that differ only in their return type cannot be overloaded
static inline uint32_t __builtin_parity(uint32_t x)
              ~~~~~~~~ ^
```

Which I fixed by:
```diff
--- a/golay24.c 2020-10-31 12:10:31
+++ b/golay24.c 2020-12-20 08:12:01
@@ -23,15 +23,15 @@

 #include "golay24.h"

-// Replacement for __builtin_parity in compilers other than GCC
-#ifndef __GNUC__
+// Replacement for __builtin_parity in compilers other than GCC and clang.
+#if !defined(__GNUC__) && !defined(__clang__)
 static inline uint32_t __builtin_parity(uint32_t x)
 {
     uint32_t p;
     volk_32u_popcnt(&p, x);
     return p & 1;
 }
-#endif /* __GNUC__ */
+#endif /* !__GNUC__ && !__clang__ */
```

daniestevez, 2020-12-20:
Thanks @gvanem! Can you please send this fix in a pull request?

gvanem, 2020-12-20:
Sorry no. I'm not friends with git.

commit 194bba96 (2020-12-27): Fix build problem with clang

The Golay implementation uses __builtin_parity() from GCC, and
gives an alternative implementation using volk_32u_popcount()
for other compilers. The current #define to detect GCC also
matched clang, which doesn't have __builtin_parity().

This closes #211
closed 2020-12-27 by daniestevez