daniestevez/gr-satellites #212: I sometimes get an error "certificate verify failed" because of having 5 characters callsign (DV2JB)

jharwinbarrozo, 2020-12-27:
[image]

daniestevez, 2020-12-27:
I don't think this has anything to do with your callsing. The telemetry server you're connecting to is using an SSL certificate that your machine doesn't like. I'm guessing this server is SatNOGS DB, since this particular message is produced by the SIDS submit code.

I'm not sure what to do with this information. The problem is most likely not with gr-satellites, but with the SatNOGS DB server, your network infrastructure (in case you have proxies or other things acting upon SSL connections), or your machine's certificates. Perhaps a wireshark capture while the problem happens would be useful, but I suggest trying to debug the problem directly with `openssl s_client`.

K4KDR, 2020-12-27:
Please note that with any unexplained SSL errors, it is wise to verify that the date & time is correct on the local computer.

kng, 2020-12-27:
I got this error earlier, and solved it with `sudo update-ca-certificates --fresh`

jharwinbarrozo, 2020-12-27:
Okay everyone, thanks for all your suggestion and advice. This community always amazes me!

I tried to reinstall the gr-satellites out of frustration, but that didn't solve the problem, but atleast got a chance to update my gr-satellites which is good!

@K4KDR In all my linux computer/rpi, I always have a script running 30 seconds after boot to check if the local time is correct if not it will re-sync automatically. 

@kng I saw this suggestion from a stackoverflow, I tried this, but it didn't solve my problem. 

But I will share here how I solved my problem, again @daniestevez is correct - THIS IS NOT RELATED to gr-satellites. This might be helpful for someone in case ohers will encounter the same issue.

I solved it by downloading this [install certificates command](https://gist.github.com/marschhuynh/31c9375fc34a3e20c2d3b9eb8131d8f3) and renaming it obvisouly to install-certicates.py
and then
`sudo python3 install-certificates.py`

closed 2020-12-27 by jharwinbarrozo

daniestevez, 2020-12-27:
Thanks @jharwinbarrozo for sharing your solution. I wonder why it is not enough to run `update-ca-certificates --fresh` and the script you mentioned needs to be run. Perhaps Python takes the SSL certificates from somewhere else.