daniestevez/gr-satellites #213: Deviation can be negative

dernasherbrezon, 2020-12-27:
Commented here:
https://github.com/daniestevez/gr-satellites/commit/6e12394e55cfd8ed1762f79be79ea7f744025e62

+ AFSK decoder might be affected. It stopped working in jradio, but might be as well in gr-satellites (i'm using the same set of blocks).

daniestevez, 2020-12-27:
Thanks for spotting this! I just added an `abs()` to the deviation, tested it, and it seems to work. I closed the issue directly with the commit message. Please re-open if there are still any problems.

dernasherbrezon, 2020-12-27:
What about AFSK? Have you noticed anything unusual? Mine implementation stopped working after additional low pass filter

daniestevez, 2020-12-27:
I haven't done extensive testing, but I just run the GOMX-1 test file
```
gr_satellites GOMX-1 --wavfile satellite-recordings/gomx_1.wav --samp_rate 48e3
```
with f5f56c3afe3df3c6a8d4dfd6ac95d0cec1da0186 and it decodes one packet correctly.

GOMX-1 uses AFSK with a negative devation.

I think that it would also be good to test a 1k2 AFSK AX.25 signal, but I'm not sure I have one of those in [satellite-recordings](https://github.com/daniestevez/satellite-recordings). Do you have an example where gr-satellites fails for an AFSK signal?

dernasherbrezon, 2020-12-27:
Not really. If gr-satellites works fine there, then there might be an issue in my AFSK implementation. Closing this one.

closed 2020-12-27 by dernasherbrezon

daniestevez, 2020-12-27:
I've just noticed I do have `swiatowid-ax25.wav` in satellites recordings. This recording contains two 1k2 AFSK AX.25 packets, and it decodes fine with f5f56c3afe3df3c6a8d4dfd6ac95d0cec1da0186, so I think I can be confident that this works.

commit f5f56c3a (2020-12-27): Fix deviation bug in FSK demodulator filter

In
6e12394e55cfd8ed1762f79be79ea7f744025e62
a lowpass filter was added before quadrature demod in the FSK
demodulator. The filter uses Carson's bandwidth rule to decide
the cut-off. However, there was a bug in the formula for Carson's
rule, since it didn't take into account that the deviation can be
negative.

In gr-satellites, a postive deviation means that the bit 0 is
the low tone and the bit 1 is the high tone, and a negative deviation
means that the bit 1 is the low tone and the bit 0 is the high tone
(in other words, the deviation is how many  [... cut]