daniestevez/gr-satellites #215: GNU Radio 3.9 support

daniestevez, 2020-12-27:
By suggestion of @nickoe, I have created this issue to coalesce all the issues regarding support for GNU Radio 3.9 (#190 and #101), and keep track what will need to be done to support GNU Radio 3.9.

The main reference for this work is the [GNU Radio 3.9 OOT Module Porting Guide](https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide). The recommendation is that support for 3.9 should be done in a different branch than 3.8, since it is very difficult to have a single code base supporting both.

My current take on this is that right now I don't have time to maintain a 3.9 branch, and I prefer to use my time to improving the functionality of the 3.8 branch. However, if other people want to take the effort to do the port, there is a `gr39` branch where they can send pull requests (send me a heads up, since it's probably better to rebase that branch to the current master if someone wants to start working with this).

daniestevez, 2020-12-27:
An issue related to GNU Radio 3.9 is #170 about OGG input support. In GNU Radio 3.9 the Audio source block can read OGG, so this will need to be tested and then the #170 can be closed:

* [ ] Test OGG file input with GNU Radio 3.9

daniestevez, 2021-01-20:
Now that GNU Radio 3.9 is out, this issue gets higher priority, and a port of gr-satellites to 3.9 will be done soon.

I'm thinking about the best way to maintain both 3.8 and 3.9 versions, since I guess that for several months many people will keep using 3.8. My initial idea is to move the `master` branch to 3.9, add a `maint-3.9` for 3.9 releases, which will use the version numbering v4, and keep the `maint-3.8` and v3 numbering for 3.8 releases. For the time being, releases will be done simultaneously on both branches and have the same features.

N6RFM, 2021-01-20:
Thanks Dani.  Sounds good.  Looking forward to having some of the new 3.9 functionality realized in gr-satellites.

nickoe, 2021-01-20:
@daniestevez  That sounds reasonable to me. I just tried to re-read the README of the repo but it is unclear what version master of gr-satellites is intended to mactch in terms of gnuradio version. Is this to be master of gnuradio or latest maint branch?

daniestevez, 2021-01-20:
Currently `master` is compatible with 3.8, since a codebase for gr-satellites compatible with 3.9 doesn't exist yet. When the codebase exists, it will be merged into master, leaving `maint-3.8` as the GNU Radio 3.8 compatible branch. I agree that the README could be more clear, so I'll change this right now.

gr-satellites wouldn't run (or even build) against 3.9 without modifications. For example, it is necessary to write the pybind11 bindings.

nickoe, 2021-01-22:
For reference https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

daniestevez, 2021-01-23:
I have pushed several commits to the [gr39 branch](https://github.com/daniestevez/gr-satellites/tree/gr39). With the last of them, c77fbdb7bb4e5413296584772922eeb1316434e4 gr-satellites seems to be working correctly with GNU Radio 3.9 as far as I can tell.

I have run `test.sh` to check that the decodes produced in the master branch (49db56534ee0ee75c623e9e4695794e6e8b5365b) and c77fbdb7bb4e5413296584772922eeb1316434e4 are the same. They are, except for a few formatting differences and some packets which get decoded in one but not in the other (I guess that the reason is slight changes in the signal processing blocks in GNU Radio). All the satellite decoders seem to be working fine as far as I could tell.

Tests and feedback are welcome because in a few days I'll be merging these changes into master and releasing v4.0.0 as the first GNU Radio 3.9 version (the release will come together with v3.7.0, having the same functionality but supporting GNU Radio 3.8).

N6RFM, 2021-01-23:
Hi Dani,
 Was able to upgrade from GNU 3.8 to 3.9.  Have hit a roadblock with the gr39 branch.  Any suggestions?  Thanks. Bob
bob@bob:~/gr-satellites-gr39/build$ cmake ..
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/local/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   chrono
--   atomic
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.6.9") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable exact version "3.6.9") 
-- Found Git: /usr/bin/git  
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components:  doxygen dot 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- Found pybind11: /usr/local/include (found version "2.6.2" dev1)
-- Using install prefix: /usr/local
-- Building for version: v3.7-compat-xxx-xunknown / 3.7.0git
-- No C++ unit tests... skipping
-- Python checking for pygccxml - found
CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:203 (message):
  Python bindings for decode_rs.h are out of sync
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:51 (GR_PYBIND_MAKE_OOT)

-- Configuring incomplete, errors occurred!
See also "/home/bob/gr-satellites-gr39/build/CMakeFiles/CMakeOutput.log".
See also "/home/bob/gr-satellites-gr39/build/CMakeFiles/CMakeError.log".
bob@bob:~/gr-satellites-gr39/build$

daniestevez, 2021-01-23:
My bad, sorry.

I think that clang-format modified one of the `.h` files for the C++ blocks, so the Python bindings need to be updated. I didn't realise because the check for the hashes is only done during `cmake`. i've just pushed a commit that fixes this.

N6RFM, 2021-01-23:
Thank you.  re-cloned gr-satellites, then git checkout gr39
Now this -
-- Python checking for pygccxml - found
CMake Error: File /home/bob/gr-satellites/docs/doxygen/pydoc_macros.h does not exist.
CMake Error at /usr/local/lib/cmake/gnuradio/GrPybind.cmake:241 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  python/bindings/CMakeLists.txt:51 (GR_PYBIND_MAKE_OOT)

CMake Error at python/bindings/CMakeLists.txt:56 (install):
  install TARGETS given target "satellites_python" which does not exist in
  this directory.

daniestevez, 2021-01-23:
Thanks for testing this. It turns out that a couple files were missing from the git repo. I've just added them.

N6RFM, 2021-01-23:
Thanks Dani.  So far majority of decodes being tested using test.sh tool run as expected, up until the last two namely sokrat and beesat-9, and what look like a memory leak?  Cheers.

SOKRAT
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 225, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 211, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 116, in __init__
    dump_path = options.dump_path, **sat)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 144, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 234, in _init_demodulator_deframer
    deframer = self.get_deframer(transmitter['framing'])(options = self.options, **deframer_additional_options)
  File "/usr/local/lib/python3/dist-packages/satellites/components/deframers/mobitex_deframer.py", line 46, in __init__
    self.connect(self, self.invert, self.slicer, self.mobitex)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.hier_block2_pb object at 0x7fdd01cf35a8>, <gnuradio.gr.gr_python.basic_block object at 0x7fdd01cf3dc0>, 0, <Swig Object of type 'gr::basic_block_sptr *' at 0x7fdd01cbd510>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.
BEESAT-9
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 225, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 211, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 116, in __init__
    dump_path = options.dump_path, **sat)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 144, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 234, in _init_demodulator_deframer
    deframer = self.get_deframer(transmitter['framing'])(options = self.options, **deframer_additional_options)
  File "/usr/local/lib/python3/dist-packages/satellites/components/deframers/mobitex_deframer.py", line 46, in __init__
    self.connect(self, self.invert, self.slicer, self.mobitex)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 37, in wrapped
    func(self, src, src_port, dst, dst_port)
  File "/usr/local/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 100, in connect
    self.primitive_connect(*args)
TypeError: primitive_connect(): incompatible function arguments. The following argument types are supported:
    1. (self: gnuradio.gr.gr_python.hier_block2_pb, block: gnuradio.gr.gr_python.basic_block) -> None
    2. (self: gnuradio.gr.gr_python.hier_block2_pb, src: gnuradio.gr.gr_python.basic_block, src_port: int, dst: gnuradio.gr.gr_python.basic_block, dst_port: int) -> None

Invoked with: <gnuradio.gr.gr_python.hier_block2_pb object at 0x7f6fcd346b20>, <gnuradio.gr.gr_python.basic_block object at 0x7f6fcd346df8>, 0, <Swig Object of type 'gr::basic_block_sptr *' at 0x7f6fcd310510>, 0
swig/python detected a memory leak of type 'gr::basic_block_sptr *', no destructor found.

daniestevez, 2021-01-23:
I also get those errors.

The problem is that those satellites use the BEESAT protocol, which needs the [beesat-sdr](https://github.com/kappiman/beesat-sdr) OOT module to be installed. That module is not yet ported to GNU Radio 3.9. gr-satellites is smart and will print some error if the OOT module is not installed, but I have the OOT module for GNU Radio 3.8 installed in my system and gr-satellites loads that up, which gives all sorts of conflicts. I'm guessing that the same is happening to you.

The solution would be to port beesat-sdr to GNU Radio 3.9 (I'll probably have to do that another day) and then install the 3.9 version.

N6RFM, 2021-01-23:
Was wondering the same.  Hard to keep up with so many moving parts!  Thanks great support.  Now gr-satellites 3.9 deployed for real life testing. Thanks again.  Stay Safe!

N6RFM, 2021-01-23:
Something else going on..  Looking at several flowgraphs which utilize IQ to satyaml decoder flow.  Getting same error related to WIN_Hamming.
Traceback (most recent call last):
  File "/home/bob/Desktop/Working_GRCs/LUME1.py", line 336, in <module>
    main()
  File "/home/bob/Desktop/Working_GRCs/LUME1.py", line 314, in main
    tb = top_block_cls(freq=options.freq, gpredict_port=options.gpredict_port, nfreq=options.nfreq, offset=options.offset)
  File "/home/bob/Desktop/Working_GRCs/LUME1.py", line 205, in __init__
    firdes.WIN_HAMMING,
AttributeError: type object 'gnuradio.filter.filter_python.firdes' has no attribute 'WIN_HAMMING'

daniestevez, 2021-01-23:
Can you share the flowgraph you're using? You can .zip it and attach it to a message here.

I'm taking a look at the flowgraphs from gr-frontends and the Signal Source block is giving
```
Param - Waveform(waveform):
	Expression <gr_waveform_t.GR_COS_WAVE: 102> is invalid for type 'int'.
```
Not sure what the problem is. Maybe my GNU Radio build or a bug.

The RTL SDR blocks belong to gr-osmosdr, which is another OOT module. I'm not sure if it has been ported to 3.9 yet.

N6RFM, 2021-01-24:
[Airspy_AAUSAT-4.grc.zip](https://github.com/daniestevez/gr-satellites/files/5861548/Airspy_AAUSAT-4.grc.zip)
Yes, osmocom source seems still at 3.8.

daniestevez, 2021-01-24:
I have the same problem with your flowgraph. Apparently this is a bug in 3.9.0 that is already reported as https://github.com/gnuradio/gnuradio/issues/4141

N6RFM, 2021-01-25:
Aha.  Will keep an eye out for this to be addressed.  Thanks.

daniestevez, 2021-02-07:
* [ ] Update Manifest to list 3.9 in `gr_supported_versions` when a stable release of gr-satellites for GR 3.9 is released.

daniestevez, 2021-03-06:
I've just released [gr-satellites v4.0.0](https://github.com/daniestevez/gr-satellites/releases/tag/v4.0.0), which is the first stable release supporting GNU Radio 3.9. I'll be closing this issue. Please report any problems you find in a new issue.

closed 2021-03-06 by daniestevez