daniestevez/gr-satellites #217: AAUSAT-4 decoder and hexdump seemingly not working

nickoe, 2021-01-08:
I was trying to decode AAUSAT-4 in preparation of testing a new radio, but I can't seem to make it dump any PDU's from live data or your test data. I tried to enable the hexdump option:

https://gr-satellites.readthedocs.io/en/latest/command_line.html?highlight=zmq#hex-dump

But when I run the test it seem that no data is printed. Is this expected?

```
gr_satellites AAUSAT-4 --wavfile satellite-recordings/aausat_4.wav --samp_rate 48e3 --hexdump
gr::log :DEBUG: correlate_access_code_tag_ff0 - Access code: 4f5a34435542
gr::log :DEBUG: correlate_access_code_tag_ff0 - Mask: ffffffffffff
gr::log :DEBUG: correlate_access_code_tag_ff1 - Access code: 4f5a34435542
gr::log :DEBUG: correlate_access_code_tag_ff1 - Mask: ffffffffffff
gr::log :DEBUG: correlate_access_code_tag_ff0 - writing tag at sample 2804
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
```

Is there an issue with the rs decode?

This is tested on `maint-3.8`. Other satellites with the test script do print decoded stuff.

janvgils, 2021-01-08:
This is strange, when I use the following command line there is no problem with generating the PDU.

I guess there is a missing dependency, what did you alter in your build environment? From earlier conversation I understood you where building "everything" in a virtual or non standard `/usr/local` environment.

`gr_satellites AAUSAT-4 --wavfile satellite-recordings/aausat_4.wav --samp_rate 48e3 --hexdump`

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 2k4 FSK downlink) (rs_errors . 0) (iterations . -1))
pdu_length = 92
contents = 
0000: 00 56 00 b1 92 48 27 00 03 00 00 54 14 57 1f 01 
0010: 26 6e 0e db c7 fe f8 7f 10 11 a3 00 04 00 3e 02 
0020: 38 ff a5 18 00 11 3b 03 43 f7 1a 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 01 57 00 00 00 00 ff ff 00 00 
0050: 00 00 00 00 00 00 00 00 00 00 3b 00 
***********************************
```

daniestevez, 2021-01-08:
Hi @nickoe, this works for me. I suspect that the your problem may be caused by an elusive issue that we have detected with the Volk Viterbi decoder in some systems/machines (so the Reed-Solomon decoder would be working fine, but the Viterbi decoder would output incorrect data).

Please see [this comment](https://github.com/daniestevez/gr-satellites/issues/88#issuecomment-655624568) and try to see whether changing the Volk Viterbi decoder implementation in `~/.volk/volk_config` fixes your problem.

It is also relevant whether `make test` runs successfully for you, since that includes a full test of the AO-40 decoder, which uses almost the same Viterbi + Reed-Solomon as AAUSAT-4.

nickoe, 2021-01-08:
@daniestevez Ok, I just tried to run `volk_profile` and I got:
```
#this file is generated by volk_profile.
#the function name is followed by the preferred architecture.
volk_64u_popcnt a_sse4_2 generic
volk_64u_popcnt generic generic
volk_64u_popcnt a_sse4_2 generic
volk_16u_byteswap a_avx2 u_avx2
volk_32u_byteswap a_avx2 generic
volk_32u_popcnt_32u generic generic
volk_64u_byteswap a_avx2 u_avx2
volk_32fc_s32fc_x2_rotator_32fc u_avx_fma u_avx_fma
volk_8u_x4_conv_k7_r2_8u avx2 avx2
volk_32f_s32f_32f_fm_detect_32f a_avx u_avx
volk_16ic_s32f_deinterleave_real_32f a_avx2 u_avx2
volk_16ic_deinterleave_real_8i a_avx2 u_avx2
volk_16ic_deinterleave_16i_x2 a_avx2 u_avx2
volk_16ic_s32f_deinterleave_32f_x2 a_avx2 generic
volk_16ic_deinterleave_real_16i a_avx2 generic
volk_16ic_magnitude_16i a_avx2 u_avx2
volk_16ic_s32f_magnitude_32f a_avx2 u_avx2
volk_16ic_convert_32fc a_avx2 u_avx2
volk_16ic_x2_multiply_16ic a_avx2 u_avx2
volk_16ic_x2_dot_prod_16ic a_axv2 u_axv2
volk_16i_s32f_convert_32f a_avx2 u_avx2
volk_16i_convert_8i a_avx2 u_sse2
volk_16i_32fc_dot_prod_32fc a_avx2 u_avx2_fma
volk_32f_accumulator_s32f a_avx u_avx
volk_32f_x2_add_32f a_avx generic
volk_32f_index_max_16u a_avx u_avx
volk_32f_index_max_32u a_avx u_avx
volk_32fc_32f_multiply_32fc a_avx u_orc
volk_32fc_32f_add_32fc a_avx u_avx
volk_32f_log2_32f a_avx2 u_avx2_fma
volk_32f_expfast_32f a_avx u_avx_fma
volk_32f_x2_pow_32f a_avx2 u_avx2
volk_32f_sin_32f a_avx2_fma u_avx2_fma
volk_32f_cos_32f a_avx2 u_avx2
volk_32f_tan_32f a_avx2 u_avx2
volk_32f_atan_32f a_avx2_fma u_avx
volk_32f_asin_32f a_avx u_avx2_fma
volk_32f_acos_32f a_avx u_avx2_fma
volk_32fc_s32f_power_32fc generic generic
volk_32f_s32f_calc_spectral_noise_floor_32f a_avx u_avx
volk_32fc_s32f_atan2_32f a_sse4_1 generic
volk_32fc_x2_conjugate_dot_prod_32fc a_avx u_avx
volk_32fc_deinterleave_32f_x2 generic generic
volk_32fc_deinterleave_64f_x2 u_avx u_avx
volk_32fc_s32f_deinterleave_real_16i a_avx2 u_avx2
volk_32fc_deinterleave_imag_32f a_avx u_avx
volk_32fc_deinterleave_real_32f a_avx2 u_avx2
volk_32fc_deinterleave_real_64f a_avx2 u_avx2
volk_32fc_x2_dot_prod_32fc a_avx_fma u_avx
volk_32fc_32f_dot_prod_32fc a_avx u_avx2_fma
volk_32fc_index_max_16u a_avx2 u_avx2
volk_32fc_index_max_32u a_avx2 u_avx2
volk_32fc_s32f_magnitude_16i a_avx2 u_avx2
volk_32fc_magnitude_32f a_avx u_sse
volk_32fc_magnitude_squared_32f a_avx u_avx
volk_32fc_x2_add_32fc u_avx u_avx
volk_32fc_x2_multiply_32fc a_avx2_fma generic
volk_32fc_x2_multiply_conjugate_32fc u_avx u_avx
volk_32fc_x2_divide_32fc a_avx generic
volk_32fc_conjugate_32fc a_avx u_avx
volk_32f_s32f_convert_16i a_avx2 u_avx2
volk_32f_s32f_convert_32i a_avx u_avx
volk_32f_convert_64f a_avx u_avx
volk_32f_s32f_convert_8i a_avx2 u_avx2
volk_32fc_convert_16ic a_avx2 u_avx2
volk_32fc_s32f_power_spectrum_32f generic generic
volk_32fc_x2_square_dist_32f a_avx2 u_avx2
volk_32fc_x2_s32f_square_dist_scalar_mult_32f a_avx2 u_avx2
volk_32f_x2_divide_32f a_avx u_avx
volk_32f_x2_dot_prod_32f a_avx u_avx2_fma
volk_32f_x2_s32f_interleave_16ic a_avx2 u_avx2
volk_32f_x2_interleave_32fc a_avx generic
volk_32f_x2_max_32f a_avx generic
volk_32f_x2_min_32f a_avx u_avx
volk_32f_x2_multiply_32f generic generic
volk_32f_64f_multiply_64f a_avx u_avx
volk_32f_64f_add_64f a_avx u_avx
volk_32f_s32f_normalize a_avx u_avx
volk_32f_s32f_power_32f generic generic
volk_32f_sqrt_32f a_avx u_avx
volk_32f_s32f_stddev_32f a_avx u_avx
volk_32f_stddev_and_mean_32f_x2 a_avx u_avx
volk_32f_x2_subtract_32f a_sse u_orc
volk_32f_x3_sum_of_poly_32f a_avx2_fma u_avx
volk_32i_x2_and_32i u_orc u_orc
volk_32i_s32f_convert_32f a_avx2 generic
volk_32i_x2_or_32i a_avx2 u_avx2
volk_32f_x2_dot_prod_16i a_avx2_fma u_avx2_fma
volk_64f_convert_32f a_avx u_avx
volk_64f_x2_max_64f a_avx u_avx
volk_64f_x2_min_64f a_avx u_avx
volk_64f_x2_multiply_64f u_avx u_avx
volk_64f_x2_add_64f a_avx generic
volk_8ic_deinterleave_16i_x2 a_avx2 generic
volk_8ic_s32f_deinterleave_32f_x2 a_avx2 generic
volk_8ic_deinterleave_real_16i a_avx2 u_avx2
volk_8ic_s32f_deinterleave_real_32f a_avx2 u_avx2
volk_8ic_deinterleave_real_8i a_avx2 generic
volk_8ic_x2_multiply_conjugate_16ic a_avx2 u_avx2
volk_8ic_x2_s32f_multiply_conjugate_32fc a_avx2 u_avx2
volk_8i_convert_16i a_avx2 u_avx2
volk_8i_s32f_convert_32f a_avx2 u_avx2
volk_32fc_s32fc_multiply_32fc a_avx_fma u_avx_fma
volk_32f_s32f_multiply_32f a_avx u_sse
volk_32f_s32f_add_32f a_avx u_orc
volk_32f_binary_slicer_32i a_avx generic
volk_32f_binary_slicer_8i a_avx2 u_avx2
volk_32u_reverse_32u bintree_permute_bottom_up bintree_permute_bottom_up
volk_32f_tanh_32f u_avx u_avx
volk_32fc_x2_s32fc_multiply_conjugate_add_32fc u_avx u_avx
volk_32f_s32f_s32f_mod_range_32f a_avx u_avx
volk_8u_x3_encodepolar_8u_x2 a_avx2 u_ssse3
volk_32f_8u_polarbutterfly_32f u_avx2 u_avx2
volk_32fc_s32f_x2_power_spectral_density_32f generic generic
```
There is no mention of spiral as you mentioned in the linked comment. This is an x280 Thinkpad with  i5-8350U CPU. I still see the reed solomon decoder fail. What can I do next?

----
I have just installed the OOT module like any other, so I didn't think about checking your docs for dependencies. I don't have `liborc` installed yet.

daniestevez, 2021-01-08:
In `~/.volk/volk_config`, try changing the line

```
volk_8u_x4_conv_k7_r2_8u avx2 avx2
```

to

```
volk_8u_x4_conv_k7_r2_8u spiral spiral
```

or to

```
volk_8u_x4_conv_k7_r2_8u generic generic
```

and try again.

nickoe, 2021-01-08:
@daniestevez Ahh, ok. That does seem to work. It decodes one packet. I assume that is what you got as well?

For reference:
```gr_satellites AAUSAT-4 --wavfile satellite-recordings/aausat_4.wav --samp_rate 48e3 --hexdump 
gr::log :DEBUG: correlate_access_code_tag_ff0 - Access code: 4f5a34435542
gr::log :DEBUG: correlate_access_code_tag_ff0 - Mask: ffffffffffff
gr::log :DEBUG: correlate_access_code_tag_ff1 - Access code: 4f5a34435542
gr::log :DEBUG: correlate_access_code_tag_ff1 - Mask: ffffffffffff
gr::log :DEBUG: correlate_access_code_tag_ff0 - writing tag at sample 2804
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode corrected 0 bytes (interleaver path 0)
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 2k4 FSK downlink) (rs_errors . 0) (iterations . -1))
pdu_length = 92
contents = 
0000: 00 56 00 b1 92 48 27 00 03 00 00 54 14 57 1f 01 
0010: 26 6e 0e db c7 fe f8 7f 10 11 a3 00 04 00 3e 02 
0020: 38 ff a5 18 00 11 3b 03 43 f7 1a 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 01 57 00 00 00 00 ff ff 00 00 
0050: 00 00 00 00 00 00 00 00 00 00 3b 00 
***********************************
```

daniestevez, 2021-01-08:
That's the same I get, so good that this has fixed it for you.

There must be an odd bug with the AVX2 Volk Viterbi decoder kernel. My usual dev laptop doesn't have AVX2, and in some AVX2 machines I've tested I think it works fine.

I'll update the note regarding this in the documentation, to say that if the problems persist after running `volk_profile` then it's probably a good idea to fall back on `spiral` or `generic`.

daniestevez, 2021-01-08:
I have now modified the documentation, so regarding me this issue can be closed.

daniestevez, 2021-01-24:
I'm closing this, since all the problems seem solved now. If the problems persist, please re-open.

closed 2021-01-24 by daniestevez

sticky-tea, 2021-07-22:
I have the same problem with Reed-Solomon deconding. My volk_config looks like this:

```
volk_64u_popcnt a_sse4_2 generic
volk_64u_popcnt a_sse4_2 generic
volk_64u_popcnt generic generic
volk_16u_byteswap a_avx2 u_avx2
volk_32u_byteswap a_avx2 u_avx2
volk_32u_popcnt_32u a_sse4_2 generic
volk_64u_byteswap a_avx2 u_ssse3
volk_32fc_s32fc_x2_rotator_32fc a_avx_fma u_avx_fma
volk_8u_x4_conv_k7_r2_8u generic generic
volk_32f_s32f_32f_fm_detect_32f a_avx u_avx
volk_16ic_s32f_deinterleave_real_32f a_avx2 u_avx2
volk_16ic_deinterleave_real_8i a_avx2 u_avx2
volk_16ic_deinterleave_16i_x2 a_avx2 u_avx2
volk_16ic_s32f_deinterleave_32f_x2 a_avx2 u_avx2
volk_16ic_deinterleave_real_16i a_avx2 u_avx2
volk_16ic_magnitude_16i a_avx2 u_avx2
volk_16ic_s32f_magnitude_32f u_avx2 u_avx2
volk_16ic_convert_32fc a_avx2 u_avx2
volk_16ic_x2_multiply_16ic u_avx2 u_avx2
volk_16ic_x2_dot_prod_16ic a_axv2 u_axv2
volk_16i_s32f_convert_32f a_avx2 u_avx2
volk_16i_convert_8i a_avx2 u_avx2
volk_16i_32fc_dot_prod_32fc a_avx2_fma u_avx2_fma
volk_32f_accumulator_s32f a_avx u_avx
volk_32f_x2_add_32f a_avx u_avx
volk_32f_index_max_16u a_avx u_avx
volk_32f_index_max_32u a_avx u_avx
volk_32fc_32f_multiply_32fc a_avx generic
volk_32fc_32f_add_32fc a_avx u_avx
volk_32f_log2_32f u_avx2_fma u_avx2_fma
volk_32f_expfast_32f a_avx_fma u_avx_fma
volk_32f_x2_pow_32f a_avx2_fma u_avx2_fma
volk_32f_sin_32f a_avx2_fma u_avx2_fma
volk_32f_cos_32f a_avx2_fma u_avx2_fma
volk_32f_tan_32f a_avx2_fma u_avx2_fma
volk_32f_atan_32f a_avx2_fma u_avx2_fma
volk_32f_asin_32f a_avx2_fma u_avx2_fma
volk_32f_acos_32f a_avx2_fma u_avx2_fma
volk_32fc_s32f_power_32fc generic generic
volk_32f_s32f_calc_spectral_noise_floor_32f a_avx u_avx
volk_32fc_s32f_atan2_32f a_sse4_1 generic
volk_32fc_x2_conjugate_dot_prod_32fc a_avx u_avx
volk_32fc_deinterleave_32f_x2 a_avx u_avx
volk_32fc_deinterleave_64f_x2 a_avx u_avx
volk_32fc_s32f_deinterleave_real_16i a_avx2 u_avx2
volk_32fc_deinterleave_imag_32f a_avx u_avx
volk_32fc_deinterleave_real_32f a_avx2 u_avx2
volk_32fc_deinterleave_real_64f a_avx2 u_avx2
volk_32fc_x2_dot_prod_32fc a_avx_fma u_avx_fma
volk_32fc_32f_dot_prod_32fc a_avx2_fma u_avx2_fma
volk_32fc_index_max_16u a_avx2 u_avx2
volk_32fc_index_max_32u a_avx2 u_avx2
volk_32fc_s32f_magnitude_16i a_avx2 u_avx2
volk_32fc_magnitude_32f a_avx u_avx
volk_32fc_magnitude_squared_32f a_avx u_avx
volk_32fc_x2_add_32fc a_avx u_sse
volk_32fc_x2_multiply_32fc a_avx2_fma u_avx2_fma
volk_32fc_x2_multiply_conjugate_32fc a_avx u_avx
volk_32fc_x2_divide_32fc a_avx u_avx
volk_32fc_conjugate_32fc a_avx u_sse3
volk_32f_s32f_convert_16i a_avx2 u_avx2
volk_32f_s32f_convert_32i a_avx u_avx
volk_32f_convert_64f a_avx u_avx
volk_32f_s32f_convert_8i a_avx2 u_avx2
volk_32fc_convert_16ic a_avx2 u_avx2
volk_32fc_s32f_power_spectrum_32f a_sse3 generic
volk_32fc_x2_square_dist_32f a_avx2 u_avx2
volk_32fc_x2_s32f_square_dist_scalar_mult_32f a_avx2 u_avx2
volk_32f_x2_divide_32f a_avx u_avx
volk_32f_x2_dot_prod_32f a_avx u_sse3
volk_32f_x2_s32f_interleave_16ic a_avx2 u_avx2
volk_32f_x2_interleave_32fc a_avx u_avx
volk_32f_x2_max_32f a_avx u_avx
volk_32f_x2_min_32f a_avx u_avx
volk_32f_x2_multiply_32f a_avx u_avx
volk_32f_64f_multiply_64f a_avx u_avx
volk_32f_64f_add_64f a_avx u_avx
volk_32f_s32f_normalize a_avx u_avx
volk_32f_s32f_power_32f a_sse generic
volk_32f_sqrt_32f a_sse u_avx
volk_32f_s32f_stddev_32f a_avx u_avx
volk_32f_stddev_and_mean_32f_x2 a_avx u_avx
volk_32f_x2_subtract_32f u_avx u_avx
volk_32f_x3_sum_of_poly_32f a_avx2_fma u_avx_fma
volk_32i_x2_and_32i a_avx2 u_avx2
volk_32i_s32f_convert_32f a_avx2 u_avx2
volk_32i_x2_or_32i a_avx2 u_avx2
volk_32f_x2_dot_prod_16i a_avx u_avx2_fma
volk_64f_convert_32f a_avx u_avx
volk_64f_x2_max_64f a_avx u_avx
volk_64f_x2_min_64f a_avx u_avx
volk_64f_x2_multiply_64f a_avx u_sse2
volk_64f_x2_add_64f a_avx u_sse2
volk_8ic_deinterleave_16i_x2 a_avx2 u_avx2
volk_8ic_s32f_deinterleave_32f_x2 a_avx2 u_avx2
volk_8ic_deinterleave_real_16i a_avx2 u_avx2
volk_8ic_s32f_deinterleave_real_32f a_avx2 u_avx2
volk_8ic_deinterleave_real_8i a_avx2 u_avx2
volk_8ic_x2_multiply_conjugate_16ic a_avx2 u_avx2
volk_8ic_x2_s32f_multiply_conjugate_32fc a_avx2 u_avx2
volk_8i_convert_16i a_avx2 u_avx2
volk_8i_s32f_convert_32f a_avx2 u_avx2
volk_32fc_s32fc_multiply_32fc a_avx_fma u_avx_fma
volk_32f_s32f_multiply_32f a_avx u_avx
volk_32f_binary_slicer_32i a_avx u_avx
volk_32f_binary_slicer_8i a_avx2 u_avx2
volk_32u_reverse_32u lut lut
volk_32f_tanh_32f a_avx_fma u_avx_fma
volk_32f_s32f_mod_rangepuppet_32f a_avx u_avx
volk_8u_x3_encodepolar_8u_x2 a_avx2 u_ssse3
volk_32f_8u_polarbutterfly_32f u_avx2 u_avx2
```

It decodes only one packet, after that i have reed-solomon failed or corrected 1 byte. I tried to change generic to spiral, but no result. GNU Radio 3.8.3, Ubuntu 18.04 in VM, Intel i5-10400. maint-3.8 branch

daniestevez, 2021-07-25:
Apparently the problem is with the AVX2 volk kernel, which is broken and [has been disabled](https://github.com/gnuradio/volk/commit/61497d115aead301850453d418415f5ad346ee25). Since you're using the `generic` kernel, you shouldn't have this problem. Are you sure this is the `volk_config` file that GNU Radio is using? I don't know of a good way to check this.

daniestevez, 2021-07-25:
Setting the env variable `VOLK_GENERIC=1` as described in https://github.com/gnuradio/gnuradio/issues/3376 might be a good idea to debug this.

sticky-tea, 2021-07-25:
> Setting the env variable `VOLK_GENERIC=1` as described in [gnuradio/gnuradio#3376](https://github.com/gnuradio/gnuradio/issues/3376) might be a good idea to debug this.

I think the problem was in virtual machine. I installed Ubuntu on my PC and now it works. Thank you!