daniestevez/gr-satellites #219: LauncherOne satellites

daniestevez, 2021-01-20:
This issue is to centralize the information about adding supports to the satellites in the LauncherOne 2021-002 launch.

These are:

* CACTUS 1
* ExoCube 2
* MITEE 1
* PolarCube
* Q-PACE
* TechEdSat 7
* CAPE 3
* RadFxSat 2
* PICS 1, 2

I haven't searched information about these yet, but I hear that several of them use AX.25.

N6RFM, 2021-01-20:
Thanks tracking here!

MITEE-1 decoding as 9k6 AX.25 G3RUH

CAPE-3 decoding 1 of 3 reported TX types.   Quadrature demodulated signal to Direwolf as 1K2 AFSK.  Other frame types ("1K2 and 9K6 FSK") yet to be decoded. Expect team to release more information on formats in next few days. 

Reports of Exocube 2 reception.  Will edit further if successful.

daniestevez, 2021-01-24:
@N6RFM I've just pushed a commit that adds a SatYAML file for CAPE-3, describing only the 1k2 AFSK telemetry. I'll try to see if I can find the information I need to implement the other formats.

I tested this with your recording:
```
 $ gr_satellites CAPE-3 --rawfile Cape_3_50K_samp_rate_doppler_corr__20Jan21_153135Z.raw --iq --samp_rate 50e3
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x0b\xa4\x00\x00\x00\x02\x00\x01\x01\x13\n+\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fh\x00f\x00\x00\x00\xc4\x0f\xc0\x04\x00\x00\x0f$\x04\x01H\x0f\x08\x04\x00\x00\x05$\xff\xff\n\\\x00\x00\x01\x05(\x00L\n\x88\x00\x19\x05\x07P\x00q\n\xc4\x00\x82\x02\x05\x9c\x00\x00\t\xe4\x00\x00\x00\np\x00\x00\xfe\n\x04\x01\xc7\xf4\xbf/\xe0\xbc\xc1\x12qXA9\t\xc5\x01 \x06\x00\x00\x00C\xb2!lC\x9f\x7ffC\x84\xda`\xc2F\xcb\xb4\xc2\xa6"\xdfB\xa54X\xc2\x9f\x9c(\xc2\xa54X\xc2\x88#p\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x05\x00\x80\x00' (total 200)
```

In the recording there are two AFSK packets, two other packets that are 1k2 FSK, and several short bursts which don't carry any data (I'm not sure why they are transmitted).

daniestevez, 2021-01-24:
I've added SatYAML file for a few of the satellites.

I'll try to get more information about the non-AX.25 protocols used by CAPE-3.

I'm missing confirmation about whether these satellites are operational, and if so a sample recording:
* CACTUS 1
* PolarCube
* Q-PACE
* PICS 1, 2

N6RFM, 2021-01-26:
Thanks Dani!

daniestevez, 2021-01-31:
I've checked SatNOGS again and there is nothing new from CACTUS 1, PolarCube, Q-PACE and PICS 1, 2.

RadFxSat 2 seems to be partially working, but I'm not sure if I'm going to implement a decoder soon, since I guess that the protocol is based on the other FOX satellites, for which we don't have a decoder in gr-satellites yet (the 10b8b decoder is missing, for example).

I'm in contact with the CAPE-3 team to see how to implement a decoder for the FSK packets.

ExoCube 2, MiTEE-1 and CAPE-3 (AX.25) have correct NORADs in their SatYAML files, so I'm closing this issue for the release of v3.7 in a few days.

closed 2021-01-31 by daniestevez