daniestevez/gr-satellites #221: make generates "Warning: urllib3 (1.26.2) or chardet (4.0.0) doesn't match a supported version!"

janvgils, 2021-01-20:
With the latest `git pull` on an up to date Debian testing distribution I get the following warning when running:

`cmake ../ && make && sudo make install && sudo ldconfig && gr_satellites --version`

```
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_si4463_scrambler.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu_packed.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu_soft.py.block.yml
/usr/local/lib/python3.9/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.2) or chardet (4.0.0) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
```
The following distribution urllib version is installed:

`dpkg -l | grep urllib`

```
ii  python3-urllib3 1.26.2-1 all HTTP library with thread-safe connection pooling for Python3
```

I think the question is, do we need to update requirements or does this need a code change?

daniestevez, 2021-01-21:
This seems to be a mismatch between the urllib3 or chardet versions and requests. gr-satellites depends on requests (and any version will serve, I think). In turn, requests depends on urllib3 and chardet, aparently, and the version must follow some consistency rules.

I would say this is a problem with the versioning of dependencies in Debian testing. It's not rare that this happens, since packets are often not upgraded simultaneously.

Besides the warning, it's quite likely that everything works correctly with gr-satellites. If anything breaks, it will be the telemetry submission, which is what uses requests.

janvgils, 2021-01-21:
Thanks for the update and explanation.
I will do some further investigation and share the outcome.

janvgils, 2021-01-21:
By upgrading requests from 2.25.0 to 2.25.1 `sudo pip install --upgrade requests` the warning, problem was solved.

This issue can be closed.

closed 2021-01-22 by daniestevez