daniestevez/gr-satellites #224: Transporter-1 satellites

daniestevez, 2021-01-26:
The following Amateur satellites were launched on the SpaceX F9 Transporter-1 mission:

* [x] UVSQ-Sat. Support already added beforehand.
* [ ] SOMP 2b
* [x] IDEASSat. Support added shortly after launch
* [ ] YUSAT 1. Signal has been analysed. A decoder is in progress.
* [ ] [ARCE 1A, 1B, 1C](http://www.amsatuk.me.uk/iaru/finished_detail.php?serialnum=643). IARU coordination only states S-band, with a variety of modulations. It would be very interesting to get a recording of these.

jinsanko, 2021-01-29:
To: Daniel
Could you share the orbit info of YUSAT-1 and IDEASSat.
Thank.

daniestevez, 2021-01-29:
@jinsanko I don't have more information than what is publicly available: the TLEs at Celestrak and Space-Track and the TLE lottery done by several people on Twitter and the librespace foundation forums.

daniestevez, 2021-01-31:
The only thing that remains to do regarding this issue is to assign final NORAD numbers, since it doesn't seem likely that we'll get S-band recordings of the ARCE satellites soon.

daniestevez, 2021-02-06:
After 1f1703d8ea4597c78ea2e2966ddafae8f039f2c9, only UVSQ-Sat and YUSAT-1 are missing NORAD IDs.

daniestevez, 2021-02-13:
I just pushed ce157a1, which adds the NORAD ID for YUSAT-1.

Only USVQ-Sat is missing, and apparently it is [still too close to another object](https://twitter.com/ppapadeas/status/1359589350904848384) for a reliable Doppler identification.

daniestevez, 2021-02-19:
Apparently a reliable TLE identification for UVSQ-Sat is still missing, but I'm going to release v3.7 regardless.

daniestevez, 2021-05-09:
Closing, since we have already the final TLE for UVSQ-Sat and no data for the ARCE satellites.

closed 2021-05-09 by daniestevez