daniestevez/gr-satellites #225: GNU Radio 3.8.2.0 (Python 3.9.1) cannot run ax.25/bspk.grc due to `ModuleNotFoundError: No module named 'satellites_swig'`

daxsis, 2021-01-27:
I really tried to get everything working, however it doesn't matter what I the example from `examples/ax.25/bpsk.grc` does not start with this error
<details>
  <summary>Expand to see the output when I run the flow graph</summary>
  ```bash
Generating: '/tmp/bpsk.py'

Executing: /usr/bin/python3 -u /tmp/bpsk.py

Traceback (most recent call last):
  File "/usr/lib/python3.9/site-packages/satellites/__init__.py", line 31, in <module>
    from .satellites_swig import *
  File "/usr/lib/python3.9/site-packages/satellites/satellites_swig.py", line 13, in <module>
    from . import _satellites_swig
ImportError: libgnuradio-satellites.so.3.7.0git: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/tmp/bpsk.py", line 40, in <module>
    import satellites
  File "/usr/lib/python3.9/site-packages/satellites/__init__.py", line 33, in <module>
    from satellites_swig import *
ModuleNotFoundError: No module named 'satellites_swig'

>>> Done (return code 1)
 \ ```
</details>

GNU Radio 3.8.2.0 (Python 3.9.1)
System:
LSB Version:    n/a
Distributor ID: ManjaroLinux
Description:    Manjaro Linux
Release:        20.2.1
Codename:       Nibia

Logs from the installation process. Some of them are short because everything installed so I include them just to show that everything was built and installed
<details>
  <summary>cmake logs</summary>
```bash
[daxsis@daxsis-worksmachine build]$ cmake ..
-- Build type not specified: defaulting to release.
CMake Warning (dev) at /usr/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (GMP).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.19/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
  /usr/lib64/cmake/gnuradio/FindGMP.cmake:1 (include)
  /usr/lib64/cmake/gnuradio/FindMPLIB.cmake:1 (find_package)
  /usr/share/cmake-3.19/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:26 (find_dependency)
  CMakeLists.txt:97 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.19/Modules/FindPackageHandleStandardArgs.cmake:426 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (MPIR).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/share/cmake-3.19/Modules/FindPkgConfig.cmake:67 (find_package_handle_standard_args)
  /usr/lib64/cmake/gnuradio/FindMPIR.cmake:1 (include)
  /usr/lib64/cmake/gnuradio/FindMPLIB.cmake:2 (find_package)
  /usr/share/cmake-3.19/Modules/CMakeFindDependencyMacro.cmake:47 (find_package)
  /usr/lib64/cmake/gnuradio/GnuradioConfig.cmake:26 (find_dependency)
  CMakeLists.txt:97 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

[... 940 lines cut ...]
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_image_receiver.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_kiss_file_sink.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_kiss_server_sink.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_telemetry_parser.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_telemetry_submit.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_kiss_file_source.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_kiss_transport.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_satellite_decoder.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_ccsds_descrambler.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_ccsds_viterbi.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_pn9_scrambler.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_rms_agc.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_si4463_scrambler.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu_packed.py.block.yml
-- Up-to-date: /usr/local/share/gnuradio/grc/blocks/satellites_sync_to_pdu_soft.py.block.yml
\```
</details>

<details>
  <summary>~/.bashrc</summary>
  ```bash
export PYTHONPATH=/usr/lib/python3.9/site-packages/
\```
</details>

daniestevez, 2021-01-27:
The main error here is

> ImportError: libgnuradio-satellites.so.3.7.0git: cannot open shared object file: No such file or directory

Did you run `ldconfig` as root after running `make install`?

daniestevez, 2021-02-23:
@daxsis is there any update about this?

daxsis, 2021-03-02:
@daniestevez yes, I run everything.

daniestevez, 2021-03-02:
Thanks. Then I'm closing. Feel free to open again if you run into more problems.

closed 2021-03-02 by daniestevez