daniestevez/gr-satellites #226: need some helo with ccsds tansceiver

dagronplayer, 2021-01-30:
hello, i am trying to build a simple transceiver by using cccsds  block. but the connection between gmsk demod and ccsds 27 is not right. could someone tell me what i can do to solve the issue? 
[image]

daniestevez, 2021-01-30:
Decode CCSDS 27 expects soft symbols, while GMSK demod puts out hard symbols. It would be advisable to use a demodulator that puts out soft symbols, but if you want to keep GMSK demod, then you must convert from char to float and to some math for the unipolar to bipolar conversion (i.e., convert [0, 1], to [-1, 1]).

I think that this question is off-topic for gr-satellites, since it doesn't involve gr-satellites (everything is in-tree GNU Radio blocks). It would be better to ask this on the [GNU Radio mailing lists](https://wiki.gnuradio.org/index.php/MailingLists).

dagronplayer, 2021-01-30:
> Decode CCSDS 27 expects soft symbols, while GMSK demod puts out hard symbols. It would be advisable to use a demodulator that puts out soft symbols, but if you want to keep GMSK demod, then you must [...]

thank you!

closed 2021-01-30 by dagronplayer