daniestevez/gr-satellites #227: gr_satellites core dumped when enabling --kiss_server

janvgils, 2021-02-03:
This evening I was testing some CAPE-3 SatNOGS recordings and at the same time testing the integration with the DK3WN decoders that can communicate with `--kiss_server` port.

When the `--kiss_server` option is enabled on the cli, gr_satellites is generating a core dump.

The following commands are used:

`gr_satellites CAPE-3 --wavfile CAPE-3_satnogs_3513802_2021-01-22T03-31-15.wav --samp_rate 48e3 --kiss_server`

The last frame that was send when gr_satellite generated the core dump:

```
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x05p\x00\x00\x00\x03\x00\x01\x01\x089\x19\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fH\x00f\x00\x00\x00\xc4\x0f\x1c\x04\x00\xb5\x0f\x04\x04\x00\x00\x0f\x08\x04\x00\x00\x07h\x00\x00\nT\x00\x00\xf3\x07x\x00E\n\xa0\x00\x00\xee\x06\xf4\x00\x00\x08\xcc\x00\x00\xf0\x07\x10\x00\x9f\x03\xb0\x01\x97\xf0\x08\xd8\x00\x00\xec\x05\xec\x01\x90\xf6@\x86\x98m\xc1qw5?\xea\x80\xfa\x01 \x06\x00\x00\x00A\xa7\xe3{C\x9a\xa1\xa4C\x8f\xa3\t\xc2\x969zB\x9f\x1b\xae\x00\x00\x00\x00\xc2\xa6\xc1\xd2\xc2\xa7#\x99B\x9a\xb9h\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x00\x00\x80\x00' (total 200)
Aborted (core dumped)
```
The following command without the `--kiss_server` option:

```
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x05p\x00\x00\x00\x03\x00\x01\x01\x089\x19\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fH\x00f\x00\x00\x00\xc4\x0f\x1c\x04\x00\xb5\x0f\x04\x04\x00\x00\x0f\x08\x04\x00\x00\x07h\x00\x00\nT\x00\x00\xf3\x07x\x00E\n\xa0\x00\x00\xee\x06\xf4\x00\x00\x08\xcc\x00\x00\xf0\x07\x10\x00\x9f\x03\xb0\x01\x97\xf0\x08\xd8\x00\x00\xec\x05\xec\x01\x90\xf6@\x86\x98m\xc1qw5?\xea\x80\xfa\x01 \x06\x00\x00\x00A\xa7\xe3{C\x9a\xa1\xa4C\x8f\xa3\t\xc2\x969zB\x9f\x1b\xae\x00\x00\x00\x00\xc2\xa6\xc1\xd2\xc2\xa7#\x99B\x9a\xb9h\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x00\x00\x80\x00' (total 200)
```
It could be that gr_satellites isn't closing the tcp session in a correct way, but that is an assumption.

The recording that is used can be found at [following location](https://www.pe0sat.vgnet.nl/download/CAPE/CAPE-3_satnogs_3513802_2021-01-22T03-31-15.wav) This observation was received by [SA2KNG](https://network.satnogs.org/observations/3513802/).

daniestevez, 2021-02-03:
Thanks for reporting. Not sure what could be causing this.

When testing this, do you have the DK3WN decoder trying to connect with the gr_satellites TCP server or not? I tried to run the same thing as you but without having any client connecting to the gr_satellites TCP server and I didn't get the abort.

janvgils, 2021-02-03:
Good evening,

I start gr_satellites and then connect the decoder by enabling (check bock) the TCP Client, this will go from yellow to green (or from closed to connected) and this is indicating the decoder is connected. When gr_satellites decodes a frame the decoder receives it, and immediately visualizes the frame.

Then at the end of the run, or maybe somewhere halfway, I can't really tell the core dump accurse.

daniestevez, 2021-02-03:
Does the abort also happen if you don't run DK3WN's decoder?

janvgils, 2021-02-03:
Yes it does.

The command:

`gr_satellites CAPE-3 --wavfile CAPE-3_satnogs_3513802_2021-01-22T03-31-15.wav --samp_rate 48e3  --kiss_server`

The last part of the output:

```
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x05p\x00\x00\x00\x03\x00\x01\x01\x089\x19\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fH\x00f\x00\x00\x00\xc4\x0f\x1c\x04\x00\xb5\x0f\x04\x04\x00\x00\x0f\x08\x04\x00\x00\x07h\x00\x00\nT\x00\x00\xf3\x07x\x00E\n\xa0\x00\x00\xee\x06\xf4\x00\x00\x08\xcc\x00\x00\xf0\x07\x10\x00\x9f\x03\xb0\x01\x97\xf0\x08\xd8\x00\x00\xec\x05\xec\x01\x90\xf6@\x86\x98m\xc1qw5?\xea\x80\xfa\x01 \x06\x00\x00\x00A\xa7\xe3{C\x9a\xa1\xa4C\x8f\xa3\t\xc2\x969zB\x9f\x1b\xae\x00\x00\x00\x00\xc2\xa6\xc1\xd2\xc2\xa7#\x99B\x9a\xb9h\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x00\x00\x80\x00' (total 200)
Aborted (core dumped)
```

daniestevez, 2021-02-03:
Then it seems this is something I can't replicate.

Can you try to paste a backtrace here so we can see where the abort happens?

Roughly, you'll need to run `ulimit -c unlimited`, then force the abort again to get a core file, then `gdb python3 core`, and use `bt` in `gdb`.

janvgils, 2021-02-04:
Good morning,

I tried as you suggested but it seems there is a issue with generating core dumps on a Linux system running within a WSL environment. Therefor I switched to a laptop running the same Linux distribution to see if I was able to generate the core file.

And as you might already have guessed, `gr_satellites` runs fine on that system without generating a core dump.

OS and hardware information from the system that runs fine.
```
cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux bullseye/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
```
cat /proc/cpuinfo 
processor	: 0
vendor_id	: GenuineIntel
cpu family	: 6
model		: 42
model name	: Intel(R) Core(TM) i5-2540M CPU @ 2.60GHz
stepping	: 7
microcode	: 0x2f
cpu MHz		: 1169.845
cache size	: 3072 KB
physical id	: 0
siblings	: 4
core id		: 0
cpu cores	: 2
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 13
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
vmx flags	: vnmi preemption_timer invvpid ept_x_only flexpriority tsc_offset vtpr mtf vapic ept vpid unrestricted_guest
bugs		: cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips	: 5188.19
clflush size	: 64
cache_alignment	: 64
address sizes	: 36 bits physical, 48 bits virtual
power management:
```

`gr_satellites CAPE-3 --wavfile CAPE-3_satnogs_3513802_2021-01-22T03-31-15.wav --samp_rate 48e3 --kiss_server`

```
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x05p\x00\x00\x00\x03\x00\x01\x01\x089\x19\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fH\x00f\x00\x00\x00\xc4\x0f\x1c\x04\x00\xb5\x0f\x04\x04\x00\x00\x0f\x08\x04\x00\x00\x07h\x00\x00\nT\x00\x00\xf3\x07x\x00E\n\xa0\x00\x00\xee\x06\xf4\x00\x00\x08\xcc\x00\x00\xf0\x07\x10\x00\x9f\x03\xb0\x01\x97\xf0\x08\xd8\x00\x00\xec\x05\xec\x01\x90\xf6@\x86\x98m\xc1qw5?\xea\x80\xfa\x01 \x06\x00\x00\x00A\xa7\xe3{C\x9a\xa1\xa4C\x8f\xa3\t\xc2\x969zB\x9f\x1b\xae\x00\x00\x00\x00\xc2\xa6\xc1\xd2\xc2\xa7#\x99B\x9a\xb9h\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x00\x00\x80\x00' (total 200)
```

Well from a `gr_satellites` point of view this is great news, but from running `gr_satellites` and the Windows based decoders from DK3WN on the same system, this is lets say, generating an new challenge.

Thanks for the support and if there are no further options we can investigate this issue can be closed.

janvgils, 2021-02-04:
And things get even stranger, the outcome wasn't very satisfying so I had another look at WSL.

As you might know, currently there are two version of WSL, version 1 and 2, I went back in the past to WSL1 because of X11 session termination issues but to see if WSL2 would be a solution to this problem I did the following:

From a Windows cli:

`wsl --list` (show the list distributions that are registered)
`wsl --export <Distro> <FileName>`
`wsl --import <New Distro Name> <InstallLocation> <FileName> --version 2` (this will import the previous export and convert it to WSL2)

Then one needs to change some X11 configuration because now the subsystem will have it own IP address and will no longer communicate via the localhost connection with the host system. The following was changed in `${HOME}/.bashrc` and because the X11 session will not use localhost the X-server needs to allow connections to its interface ip.

```
HOST_IP=$(host `hostname` | grep -oP '(\s)\d+(\.\d+){3}' | tail -1 | awk '{ print $NF }' | tr -d '\r')
export LIBGL_ALWAYS_INDIRECT=1
export DISPLAY=$HOST_IP:0.0
export NO_AT_BRIDGE=1
export PULSE_SERVER=tcp:$HOST_IP
```

When this was finally up and running we tried the gr_satellites command again and this time without a problem.

`gr_satellites CAPE-3 --wavfile CAPE-3_satnogs_3513802_2021-01-22T03-31-15.wav --samp_rate 48e3 --kiss_server`

```
-> Packet from 1k2 AFSK downlink
Container: 
    header = Container: 
        addresses = ListContainer: 
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = False
                    ssid = 1
                    extension = False
            Container: 
                callsign = u'W5UL' (total 4)
                ssid = Container: 
                    ch = True
                    ssid = 11
                    extension = True
        control = 0x03
        pid = 0xF0
    info = b'\x02CAPE-3 W5UL-1100010001B0000000\x00\x00\x00\x00\x00\x00\x05p\x00\x00\x00\x03\x00\x01\x01\x089\x19\x00\x00:\x98\x01\x00\x00\x01\x00\x00\x00\x00\x0fH\x00f\x00\x00\x00\xc4\x0f\x1c\x04\x00\xb5\x0f\x04\x04\x00\x00\x0f\x08\x04\x00\x00\x07h\x00\x00\nT\x00\x00\xf3\x07x\x00E\n\xa0\x00\x00\xee\x06\xf4\x00\x00\x08\xcc\x00\x00\xf0\x07\x10\x00\x9f\x03\xb0\x01\x97\xf0\x08\xd8\x00\x00\xec\x05\xec\x01\x90\xf6@\x86\x98m\xc1qw5?\xea\x80\xfa\x01 \x06\x00\x00\x00A\xa7\xe3{C\x9a\xa1\xa4C\x8f\xa3\t\xc2\x969zB\x9f\x1b\xae\x00\x00\x00\x00\xc2\xa6\xc1\xd2\xc2\xa7#\x99B\x9a\xb9h\x00\x00\n\x06\x00\x00\x00\x00Qh\x03\x08\x00\x00\x80\x00' (total 200)
```

So switching to WSL2 also solves the problem

daniestevez, 2021-02-04:
Hi Jan,

Awesome that you found a workaround to this issue. In fact the most of the functionality of the KISS server in gr_satellites comes from GNU Radio blocks, so I guess that the problem would be found in those blocks rather than in gr_satellites itself. Debugging that can be quite difficult.

For other users that want to run gr_satellites in Windows and those kinds of problems are getting in their way, another possibility is to use [conda](https://gr-satellites.readthedocs.io/en/latest/installation_conda.html) to install a native GNU Radio and gr-satellites for Windows.

I will now close this, since you have found and described the workaround. I will first add "WSL" in the issue title to make it easier for other people having the same problem to find this.

closed 2021-02-04 by daniestevez