daniestevez/gr-satellites #230: UnicodeDecodeError: 'cp950'

KOUKILEO, 2021-02-09:
I tried to decode IDEASat.wav, 
but I got UnicodeDecodeError:'cp950' codec can't decode byte 0xc3 in position 37: illegal multibyte sequence.
error code
Did I use the wrong method?
Thanks for help
[image]

daniestevez, 2021-02-09:
That points out to a problem with your [IDEASSat.yml](https://github.com/daniestevez/gr-satellites/blob/master/python/satyaml/IDEASSat.yml) file. This should contain only ASCII bytes, and it seems that the error you get says that it has an invalid sequence that can't be parsed as Unicode. I just checked that the file in the master branch is ASCII, so it seems that your copy of that file is different and corrupted.

KOUKILEO, 2021-02-09:
> That points out to a problem with your [IDEASSat.yml](https://github.com/daniestevez/gr-satellites/blob/master/python/satyaml/IDEASSat.yml) file. This should contain only ASCII bytes, and it seems [...]

Hello, I installed it through the teaching [file](https://gr-satellites.readthedocs.io/en/latest/installation_conda.html#installing-using-conda). I found that the IDEEASSat.yml file is missing for the teaching installation using the file.
So after I downloaded IDEEASSat.yml, I still got an error, this error is try_add_options(cls._deframer_hooks[transmitter['framing']], deframe_options)
KeyError:'IDEASSat'.
Attach the operation video and my information (C:\Users\user\miniconda4\Lib\site-packages) for you to see~
[video](https://youtu.be/Ytorlj2AwuUl)  [file](https://drive.google.com/file/d/1RIsE5yh3ndazwkH1DVo2gwvLyXCUd2D9/view?usp=sharing)
PS.IDEASSat.wav file is from satellite-recordings

daniestevez, 2021-02-11:
Thanks for all the information. I've now taken a look in more detail and understood your problem.

It turns out that there are some SatYAML files (not IDEASSat's) which have UTF-8 characters. By default, Python will open these files using the system's encoding, which in your case is [CP 950](https://en.wikipedia.org/wiki/Code_page_950), which is used for Traditional Chinese in Windows (perhaps this is the first time anyone is trying to use gr-satellites on a Windows machine set in Traditional Chinese, so it's good to debug these issues). I have just pushed a commit that forces Python to open the SatYAML files as UTF-8, so this should fix the problem.

Seeing that you're using gr-satellites from conda on Windows, it might not be so easy for you to test this fix in the master branch, so as a work around I think that you can delete `IRAZU.yml` and `NuSat_1.yml`, which are the files that have UTF-8 characters (probably you don't care about those satellites). After deleting this two files, the IDEASSat decoder should work.

KOUKILEO, 2021-02-28:
> 感謝您所提供的所有信息。現在，我已經更詳細地了解了您的問題。
> 
> 事實證明，有些SatYAML文件（不是IDEASSat的文件）具有UTF-8字符。默認情況下，Python將使用系統的編碼（在您的情況下為[CP 950）](https://en.wikipedia.org/wiki/Code_page_950)打開這些文件，該編碼用於Windows中的繁體中文（也許這是第一次有人嘗試在Windows機器上使用gr-satellite設置）。繁體中文，因此最好調試這些問題）。我剛剛推送了一個提交，該提交強制Python以UTF-8格式打開SatYAML文件，因此這應該可以解決問題。
> 
> 看到您在Windows上使用的是conda的gr-satellites，在master分支中測試此修復程序可能並不容易，因此，作為一種變通辦法，我認為您可以刪除`IRAZU.yml`和`NuSat_1.yml`，這些文件具有UTF-8字符（可能您並不在乎那些衛星）。刪除這兩個文件後，IDEASSat解碼器應該可以工作。

Sorry for the late reply. After I downloaded the new repaired version, it can be used normally. Thank you for your help~

daniestevez, 2021-02-28:
Thanks! Closing.

closed 2021-02-28 by daniestevez