daniestevez/gr-satellites #232: Bobcat flowgraph

ik8ozv, 2021-02-11:
Please is there anybody that can give me an hand to solve the problem?
I am not so clever on Linux. I am trying to make the Flowgraph for the Bobcat-1 
but when i run it i get this error. Anybody please can tell me where am i doing wrong and how to correct it?
Thanks in advance.
73's de Enzo IK8OZV

[image]

daniestevez, 2021-02-11:
Hi Enzo,

Nothing is wrong there. There are warnings telling you that demodulators for the higher baudrates that Bobcat-1 can use have been disabled due to the sample rate being too low (this is normal). There is also a warning saying that the performance of one of the demodulators (the one for 38k4) will not be optimal due to the sample rate. Not much else. Your flowgraph should work well for baudrates smaller or equal than 38k4.

daniestevez, 2021-02-28:
i'm closing this due to inactivity. Feel free to reopen.

closed 2021-02-28 by daniestevez