daniestevez/gr-satellites #234: Issue decoding Aztechsat-1 telemetry

XE3ADC, 2021-02-23:
Hello, good day,
I am trying to decode the telemetry of Aztechsat-1 satellite using Gnuradio 3,8 with gr-satellites under ubuntu 18.04 lts. 
As a source WAV file I’m using the sample telemetry beacon of the following link:

https://upaep.mx/templates/aztechsat/files/gqrx_20200131_192013_437300000.wav

As per page 16 of document https://upaep.mx/templates/aztechsat/files/AZT_ORR_BL_GS_ARI_v2.pdf, the expected PDU output for that particular telemetry beacon sample file  is:

0000: c3 c2 80 00 68 74 74 70 73 3a 2f 2f 75 70 61 65 
0010: 70 2e 6d 78 2f 61 7a 74 65 63 68 73 61 74 20 3c 
0020: 48 4b 3e 01 65 01 5c 01 66 1c f6 00 c8 00 00 00 
0030: 00 00 15 00 78 00 6f 00 39 00 00 00 05 01 01 01 
0040: 00 00 00 01 29 00 00 06 d6 00 00 00 de 00 17 00 
0050: fc ff 72 ff 74 01 00 00 00 00 00 00 01 bf a4 1a 
0060: 42 c1 b8 9d 8a c4 01 2a ab bf c7 8f 78 3f a3 84 
0070: 38 3d b6 db 6d 02 03 05 0a 04 00 0a 00 00 00 00 
0080: 00 00 00 00 00 00 00 00 17 00 00 b9 fc ce e3 04 
0090: 00 04 00 04 00 04 00 04 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3f 89 
00b0: cf 24 40 0f 13 af 00 3c 2f 48 4b 3e 00 00 00 00 
00c0: 00 00 00 00 00 00 00 00 00 00 00 00

Instead, when using the satellite decoder module with gnuradio companion I obtain the following messages on the console:

_**Generating: '/home/abesr/gr-satellites/examples/satellite_decoder/satellite_decoder.py'
Warning: This flow graph may not have flow control: no audio or RF hardware blocks found. Add a Misc->Throttle block to your flow graph to avoid CPU congestion.

Executing: /usr/bin/python3.6 -u /home/abesr/gr-satellites/examples/satellite_decoder/satellite_decoder.py

gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 45789
Golay decode OK. Bit errors: 0, Frame length: 240, Viterbi: 0, Scrambler 0, Reed-Solomon: 0.
RS decode failed**._

A similar output is obtained when using the CLI command: 
**gr_satellites AztechSat-1 --wavfile ~/satellite-recordings/aztechsat1.wav --samp_rate 48e3 –verbose_fec**
(please note that the  sample telemetry wav file was renamed to aztechsat1,wav for easier handling)

I tried different values for the CLI parameter options on the GRC module with no success, could you please help me to solve this issue?

Thanks in advance and best regards,
Abraham Diaz de Leon Camou – XE3ADC
Villahermosa – Tabasco - Mexico

daniestevez, 2021-02-23:
Hi,

The `Frame length: 240` doesn't match what we would expect, since the frame you show as an example is much shorter. Therefore, that seem to be a false sync where there are zero Golay errors just by chance.

First, ensure that the signal polarity is correct. Maybe try `--gain -1` just in case the polarity needs to be swapped.

Next, you can use `--dump_path` to ensure that the FSK demodulation is being done correctly (without bit errors) and use parameters such as `--clk_bw` if not. There is some description in the documentation about how to do this, but #196 has more detailed information.

5ch4um1, 2021-02-23:
"Add a Misc->Throttle block to your flow graph to avoid CPU congestion."

Just an idea, but have you tried the --throttle option on the cli?

daniestevez, 2021-02-23:
The presence of the throttle block shouldn't matter. The flowgraph should give exactly the same results (and I reckon that sometimes it doesn't, due to subtle bugs with scheduling and blocks that change behaviour depending on how their input is chunked).

XE3ADC, 2021-02-24:
> "Add a Misc->Throttle block to your flow graph to avoid CPU congestion."
> 
> Just an idea, but have you tried the --throttle option on the cli?

Hello 5ch4um, yes I tried to place a Throttle block on the flow-graph but it could not be done as the WAV source signal is not compatible with the input of the throttle block .
As for the CLI: the Throttle option is available for some satellites but not available for the Aztechsat-1 satellite.
Thanks for your suggestion, Abraham

XE3ADC, 2021-02-24:
> Hi,
>
> The `Frame length: 240` doesn't match what we would expect, since the frame you show as an example is much shorter. Therefore, that seem to be a false sync where there are zero Golay errors [...]

Hello, 
Thanks for your quick response, since the gain and dump parameters were not accepted by the satellite demodulator block in the grc flowgraph I did the following on the CLI:

For the non-inverted beacon wav signal:

**abesr@abraham-5520:~$ gr_satellites AztechSat-1 --wavfile ~/satellite-recordings/aztechsat1.wav --samp_rate 48e3 --verbose_fec --dump_path /tmp/fsk-norm**
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 17673
Golay decode OK. Bit errors: 1, Frame length: 198, Viterbi: 0, Scrambler 1, Reed-Solomon: 0.
RS decode failed.
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 45789
Golay decode OK. Bit errors: 0, Frame length: 240, Viterbi: 0, Scrambler 0, Reed-Solomon: 0.
RS decode failed.

the following zip file contains a set of .f32 files that were created on the /tmp/fsk-norm directory:

[fsk-norm.zip](https://github.com/daniestevez/gr-satellites/files/6037497/fsk-norm.zip)

For the non-inverted beacon wav signal:

**abesr@abraham-5520:~$ gr_satellites AztechSat-1 --wavfile ~/satellite-recordings/aztechsat1.wav --samp_rate 48e3 --verbose_fec --dump_path /tmp/fsk-inv --input_gain -1**
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb0 - writing tag at sample 42787
Golay decode OK. Bit errors: 3, Frame length: 48, Viterbi: 1, Scrambler 1, Reed-Solomon: 1.
RS decode failed.
abesr@abraham-5520:~$ 

the following zip file contains a set of .f32 files that were created on the /tmp/fsk-inv directory.

[fsk-inv.zip](https://github.com/daniestevez/gr-satellites/files/6037513/fsk-inv.zip)

It can be noted that when using input signal inversion:
-  There appear 3 bit errors, against 0 obtained with the non inverted signal
-  Only 48 pdus against 240. 
-  Only outputs Scrambler 1 whereas Scrambler 1 and 2 have outputs with the non inverted signal
-  Reed Solomon flag reported as 1 compared to 0 obtained with the non inverted signal.

I will follow up with the graphical analysis of the f32 files and report back.
Thanks again, Abraham

XE3ADC, 2021-02-25:
> Hello 5ch4um, yes I tried to place a Throttle block on the flow-graph but it could not be done as the WAV source signal is not compatible with the input of the throttle block .
> As for the CLI: the [...]
Hello 5ch4um, I was wrong..... in fact I was able to put a throttle block between the wav source block and the satellite decoder block, but, although the warning message was not displayed  on the console and the processing was slowed down, the results were the same as before....cheers, Abraham.

XE3ADC, 2021-02-26:
hello, good day,
The following zip file contains the de-codification plots obtained with both, the non-inverted (norm) and inverted(inv) options of the CLI decoder command, all other decoder parameters were default:

[Aztechsat-1-plots.zip](https://github.com/daniestevez/gr-satellites/files/6047016/Aztechsat-1-plots.zip)

I could note differences on the clk_recovery_out plots:
Non inverted:
[image]
Inverted:
[image]

Also, differences can be noted on the waveform plots, 
Non inverted:
[image]
Inverted:
[image]

All other pair of plots do not show evident differences between the norm and inv input, is this to be expected?
The noted differences on the plots are as if the inverted plots are the upside down versions of the non-inverted, is this also to be expected too?
I suppose the next steps are to run the decoder with different decoding parameters until obtaining a successful decode, however, my expertise on this is very limited and require your help.
Could you please provide feedback on the above?, Any suggestions on the procedure to follow?, what input should I keep using to carry on?, what decoding parameters are to try first?, what change of values are recommended?

Thanks in advance,

Abraham

daniestevez, 2021-02-26:
Thanks for the plots. The clock recovery output looks pretty bad. You're looking for the symbols to "stand still" close to the +1 and -1 levels. All those wavy oscillations are bad and can cause bit errors. Can you try different values for `--clk_bw` and `--clk_limit` to see if it improves

> The noted differences on the plots are as if the inverted plots are the upside down versions of the non-inverted, is this also to be expected too?

Yes. This is expected. I suggest you first use non-inverted and adjust clock parameters to get a clean demodulation (clock recovery output). If with that there are still no decodes, you can try to invert.

If I have some time this weekend I'll try this myself and tell you if I manage to make this work.

daniestevez, 2021-03-23:
Hi Abraham, sorry for the delay. I finally took a look at the recording and was able to decode it after some effort, obtaining the frame:

```
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length = 208 bytes
pdu vector contents = 
0000: 00 80 c2 c3 68 74 74 70 73 3a 2f 2f 75 70 61 65 
0010: 70 2e 6d 78 2f 61 7a 74 65 63 68 73 61 74 20 3c 
0020: 48 4b 3e 01 65 01 5c 01 66 1c f6 00 c8 00 00 00 
0030: 00 00 15 00 78 00 6f 00 39 00 00 00 05 01 01 01 
0040: 00 00 00 01 29 00 00 06 d6 00 00 00 de 00 17 00 
0050: fc ff 72 ff 74 01 00 00 00 00 00 00 01 bf a4 1a 
0060: 42 c1 b8 9d 8a c4 01 2a ab bf c7 8f 78 3f a3 84 
0070: 38 3d b6 db 6d 02 03 05 0a 04 00 0a 00 00 00 00 
0080: 00 00 00 00 00 00 00 00 17 00 00 b9 fc ce e3 04 
0090: 00 04 00 04 00 04 00 04 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3f 89 
00b0: cf 24 40 0f 13 af 00 3c 2f 48 4b 3e 00 00 00 00 
00c0: 00 00 00 00 00 00 00 00 00 00 00 00 3b 29 f5 c0 
************************************
```

Now, there are a number of related problems I needed to sort out:
* It turns out that Aztechsat-1 is using the AX100 in ASM+Golay mode, but has scrambling disabled. This is apparent from the long strings of zeros that can be seen in the recording waveform. I have confirmed that this is intentional by looking at the GNU Radio 3.7 flowgraph supplied in the [mission operations page](https://upaep.mx/aztechsat/missionoperations). This flowgraph uses the deframer from [mndza/gr-sattools](https://github.com/mndza/gr-sattools) with the CCSDS Randomization parameter set to disabled. It's the first AX100 in ASM+Golay that I've seen to do this. I even thought it was impossible. So gr-satellites assumes that scrambling is used. I'll probably think of a way to indicate this in the SatYAML file.
* The strings of zeros are so long that the DC block in the FSK demodulator in gr-satellites makes a mess of them. I had to disable it. I'll add a command line parameter so that this can be done without touching the code.
* The recording has a significant DC offset. I had to correct this by hand, by saving the file in raw float32, adding a suitable constant, and using that with gr-satellites. For this I can't give a solution. The DC offset was caused by a frequency error when the recording was done.

So to summarise, two tasks for me:
* [ ] Add a way to indicate no scrambler for AX100 in ASM+Golay mode in SatYAML
* [ ] Add a command line parameter to disable the DC block in the FSK demodulator

daniestevez, 2021-03-24:
It turns out that the DC offset, even though large, doesn't impede decoding, since the SNR of this recording is pretty good. After adding the two commits mentioned above, I can now decode the recording with
```
gr_satellites AztechSat-1 --wavfile gqrx_20200131_192013_437300000.wav --disable_dc_block
```

I think I'm done. You can close this issue if you're happy with the result.

XE3ADC, 2021-03-24:
Hello Daniel, 
Many thanks for your efforts, I have been doing tests with different values for parameter clk_limit, as the tests were not successful I started to suspect that perhaps there was something wrong with the source wav file, the following file contains a brief report of the results of the tests done with parameter clk_limit: 

[Aztechsat-clk_limit-tests.pdf](https://github.com/daniestevez/gr-satellites/files/6201007/Aztechsat-clk_limit-tests.pdf)

I opened the source wav file with Audacity and was expecting to have a non return to zero binary signal varying between two values, one negative and the other positive. I encountered that the signal was not swinging symmetrically between  the positive and the negative values, but in fact the negative swing was comparatively less than the positive, in other words if the positive swing has a peak value of A the negative has a peak value of A/3 approximately. I was trying to figure out how to compensate for this, perhaps using a Low pass filet block to remove the DC offset of the signal or maybe using a sound application like audacity.

After reading your two last reports with your findings and amends I now comprehend that the issue required your magic, would you please give me the procedure to have my system parched with your commits?
As per the wav source file and to avoid the FM demodulation errors, Would it be better to use a captured IQ file containing the samples of the Beacon?

Tons of Thanks again for your efforts, please keep me in the loop on this, I’ll happily help with further testing and follow ups.

Best regards,

Abraham Diaz de Leon - XE3ADC

daniestevez, 2021-03-25:
The commits are in the master (GNU Radio 3.9) and maint-3.8 (GNU Radio 3.8) branches. You can pull any of these branches, build from source and install in your system.

XE3ADC, 2021-03-25:
Thanks Daniel, 
Yes, I located the commits, 
Sorry for my inexperience, I'm not too sure how to proceed for the installation, do I need to uninstall gr-satellites?, will the following be enough?:
cd /gr-satellites/build
sudo make unistall
make clear 

Do I need to remove the gr-satellites directory and do the git clone again in my home directory? 

Thanks for bearing with me, your efforts and patience are greatly appreciated
Best regards,

Abraham Diaz de Leon - XE3ADC

daniestevez, 2021-03-25:
You can re-install on top of your existing installation. No need to uninstall.

> Do I need to remove the gr-satellites directory and do the git clone again in my home directory?

No. This is not at all necessary. You just need to pull the branches from the remote. But if you don't have any experience with git, removing the repository and cloning again might be more simple.

XE3ADC, 2021-03-26:
Thanks a lot Daniel..!!

closed 2021-03-26 by XE3ADC