daniestevez/gr-satellites #236: DELFI-n3xt.yml isn't being pulled in

janvgils, 2021-03-02:
There seems something wrong when executing a git clone or pull regarding the DELFI-n3xt.yml file.

I started with a git pull in the maint-3.8 branch and the file wasn't being pulled in.

```
cd gr-satellites
git branch
* maint-3.8
  master

git pull
Already up to date.

find . -name DEL\* -print
./python/satyaml/DELFI-C3.yml
```

Then started with a new `git clone https://github.com/daniestevez/gr-satellites.git` and the file was also not part of the tree.

Any idea how to solve this

daniestevez, 2021-03-02:
Sorry. Silly me, I cherry picked the commits from master into maint-3.8 the other day, but I forgot to push my branch to the remote. It should be fixed now.

cpbridges, 2021-03-02:
OK - This explains my issue too. Using Conda for my installation btw with the following versions:

```
gnuradio                  3.8.1.0         py37hc0a61fd_10    conda-forge
gnuradio-satellites       3.3.0            py37he3a1320_0    conda-forge
gnuradio-soapy            2.1.3            py37ha1519f6_0    conda-forge
```

daniestevez, 2021-03-02:
Hi Chris,

The DELFI-n3xt SatYAML isn't available yet in any stable version of gr-satellites, but you can download the file and place it anywhere you want.

Note that the version of your `gnuradio-satellites` is somewhat old. We're already at 3.7.0.

janvgils, 2021-03-03:
Good morning,

Thanks for the update and I can share that this is now fixed:

`git pull`

```
remote: Enumerating objects: 18, done.
remote: Counting objects: 100% (18/18), done.
remote: Compressing objects: 100% (5/5), done.
remote: Total 18 (delta 13), reused 18 (delta 13), pack-reused 0
Unpacking objects: 100% (18/18), 2.12 KiB | 51.00 KiB/s, done.
From https://github.com/daniestevez/gr-satellites
   1f9a7bf..9fe85e2  maint-3.8  -> origin/maint-3.8
Updating 1f9a7bf..9fe85e2
Fast-forward
 CHANGELOG.md                  |  6 ++++++
 CMakeLists.txt                |  3 ++-
 MANIFEST.md                   |  2 +-
 README.md                     | 10 ++++++++--
 docs/source/conf.py           |  2 +-
 python/__init__.py            |  2 +-
 python/satyaml/DELFI-n3xt.yml | 13 +++++++++++++
 python/satyaml/UVSQ-SAT.yml   |  2 +-
 8 files changed, 33 insertions(+), 7 deletions(-)
 create mode 100644 python/satyaml/DELFI-n3xt.yml
 ```

So from my point of view this issue can be closed.

daniestevez, 2021-03-03:
To distinguish the stable version (in this case 3.7.0), from the development branch (which will become the next stable version in the future), I'm always versioning the development branch as the next sable version followed by `-git`, so the development branch in `maint-3.8` is called 3.8.0-git.

daniestevez, 2021-03-19:
Hi, I'm closing this, since I think all the problems raised here are solved now.

closed 2021-03-19 by daniestevez