daniestevez/gr-satellites #238: SOKRAT and BEESAT-9 runtime errors

N6RFM, 2021-03-06:
Hi Dani,

Just upgrading to gr-satellites 4.0 and GNU3.9.  Fortunately, gr-osmosdr now compiles using GNU 3.9.  Looks great with a few glitches. :-)
[error.txt](https://github.com/daniestevez/gr-satellites/files/6096067/error.txt)

However, when using ./test.sh, seeing errors related to SMOG-P (which I recall has decayed), SOKRAT and Beesat-9.  For the latter two, these appear to be related to the OOT Mobitex decoder.  Is there a port of gr-tnc_nx which will build under GNU3.9?  

Thanks as always!
Bob

daniestevez, 2021-03-07:
Hi Bob,

gr-tnc_nx isn't ported yet to GNU Radio 3.9. This project is originally from TU Berlin, but it seems the no longer maintain it (in fact I did the port to GNU Radio 3.8). I'll have to see if I want to port this to GNU Radio 3.9 or if I want to take the relevant parts of the code and integrate it with the rest of gr-satellites so that it fits better with the current deframer organization.

Let's keep this issue open to serve as a reminder.

N6RFM, 2021-03-07:
Sounds like a plan.  Thanks Dani.  Will QRX.

N6RFM, 2022-06-03:
Hi Dani,

As I've mostly been using gr-satellites under GNU radio 3.9, I was surfing around to see if there had been any updates to gr-tnx_nx.  It appears that one of the developers at Tubingen has been making some revisions but in the 3.8 environment.  @benp. 

https://git.tu-berlin.de/rft/com/mobitub-2/-/merge_requests/1

Since this is a gitlab repo, I don't have access and am unable to contact the person and ask about porting to 3.9.  Perhaps you two are connected?

Hoping we can add support back for satellites needing this OOT module.   Ideally a solution will work for 3.9 and 3.10.  :-)

Thanks,

Bob

benpnz, 2022-06-10:
Hi Bob,

The person doing the modifications to the repo is me :) So far the aim of my project is only to add support for GNU radio 3.8 to the repository (thanks to @daniestevez for his work in this previously). If time permits I can try and look at this though and will let you know.

Thanks,
Ben.

closed 2025-04-29 by daniestevez