daniestevez/gr-satellites #244: commit/7dd1cb15...  install error

N6RFM, 2021-03-19:
Hi Dani,

Thanks adding new protocols and support, especially for upcoming Soyuz-2 launch!  Did git pull here into Mastter branch.  compiles OK, but sudo make install gives error at end -
CMake Error at grc/components/deframers/cmake_install.cmake:41 (file):
  file INSTALL cannot find
  "/home/bob/gr-satellites/grc/components/deframers/satellites_usp_deframer.block.yml":
  No such file or directory.
Call Stack (most recent call first):
  grc/components/cmake_install.cmake:42 (include)
  grc/cmake_install.cmake:115 (include)
  cmake_install.cmake:51 (include)

make: *** [Makefile:86: install] Error 1

Any suggestions?  Thanks.  Bob

daniestevez, 2021-03-19:
Hi Bob,

Sorry. I forgot to add this file to the Git repo. I've just pushed this in a new commit.

I thought that I had added everything correctly, since the CI tests were passing. However, the CI doesn't try to `make install` gr-satellites. It only runs the tests. So this problem went undetected. Thanks for reporting quickly!

N6RFM, 2021-03-19:
Thanks Dani!  So fast!  All set here.  Looking forward to trying the new type of deframer.  Great weekend.  Bob

closed 2021-03-19 by daniestevez