daniestevez/gr-satellites #248: Crop first 4 bytes when AX.25 from USP going to KISS

bychek, 2021-03-19:
So the payload will start with the 16-byte AX.25 header. Should remove EtherType and BPQ length field. Discovered by PE0SAT on SatNOGS matrix/element IM. Not really sure how to implement it yet.

daniestevez, 2021-03-19:
There is a `pdu_head_tail` block that can be used to crop the first 4 bytes. This could be placed as the last block in the `usp_deframer`. However, if doing so, we lose the information about the AX.25 frame length, so perhaps we would need a custom block that not only crops out these 4 bytes, but also the padding after the AX.25 frame.

daniestevez, 2021-03-19:
I'm implementing a custom Python block which does what I said in my previous comment. However, I'm having some trouble understanding the BPQ length field. I have these two test frames:

```
08ff5a00a464829c8c4060a4a66060a6406f00f016420200010042
000000000000000000000000000000000000000000000000000000
00000000000000000000000000000000000000001b1bff671f2025
0eaab14060f43c01002400f01cde511c3c900e66e158aa0c5b0000
000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000
000000000000000000000000000000000000000000000000000000
00000000000000

08ff1b00a464829c8c4060a4a66060a6406f00f0e1ff0200010003
000026063d568f178c6854e277c4ffbf0000000000
```

I think that the first AX.25 packet should end at `0c5b` and the second at `ffbf`, and the remaining zeros are padding. The length fields are 90 and 27 respectively. However, if I take this number of bytes starting at `a464...`, I find that I miss the 12 last bytes of the packet.

Is there a reason for this? Should I just add 12 to the length field?

daniestevez, 2021-03-19:
For reference, I'm working in the [usp_ax25_crop](https://github.com/daniestevez/gr-satellites/tree/usp_ax25_crop) branch.

bychek, 2021-03-19:
No need to add 12 bytes. Short packet ends with 002606, then it looks like some garbage.

daniestevez, 2021-03-19:
Thanks! I've just force-pushed new commits to the usp_ax25_crop. Please check if the output of the USP deframer now looks like what you would like to get in the KISS files and in SatNOGS DB (see, for instance, the [QA test](https://github.com/daniestevez/gr-satellites/blob/6f82918ef11e6e5485f736474827a9d0e52ca082/python/components/deframers/qa_usp_deframer.py#L108)). If this is correct, I'll merge the branch into master, closing this issue.

bychek, 2021-03-19:
Thanks a lot! Looks ok. @janvgils can check too I think. Anyway, it looks good to merge, so you can just go ahead, and if we've missed something, then we'll just reopen.

janvgils, 2021-03-19:
Good evening,

I just did a new `git pull` and build on a maint-3.8 branch.

```
remote: Enumerating objects: 70, done.
remote: Counting objects: 100% (56/56), done.
remote: Compressing objects: 100% (13/13), done.
remote: Total 34 (delta 25), reused 29 (delta 21), pack-reused 0
Unpacking objects: 100% (34/34), 5.37 KiB | 423.00 KiB/s, done.
From https://github.com/daniestevez/gr-satellites
   f021c99..838456a  maint-3.8     -> origin/maint-3.8
   4f22668..c050be6  master        -> origin/master
 * [new branch]      usp_ax25_crop -> origin/usp_ax25_crop
Updating f021c99..838456a
Fast-forward
 python/satyaml/CubeSX-Sirius-HSE.yml | 2 +-
 python/satyaml/KSU_CubeSat.yml       | 2 +-
 2 files changed, 2 insertions(+), 2 deletions(-)
 ```

The old `--hexdump` based on the following command `gr_satellites 99724 --rawfile Orbicraft-Zorkiy_SignalSample_float.iq --samp_rate 480e3 --iq --hexdump`

```
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) (rs_errors . 0) (iterations . -1))
pdu_length = 223
contents = 
0000: 08 ff 5a 00 a4 64 82 9c 8c 40 60 a4 a6 60 60 a6 
0010: 40 6f 00 f0 16 42 02 00 01 00 42 00 00 00 00 00 
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 00 00 1b 1b ff 67 1f 20 
0050: 25 0e aa b1 40 60 f4 3c 01 00 24 00 f0 1c de 51 
0060: 1c 3c 90 0e 66 e1 58 aa 0c 5b 00 00 00 00 00 00 
0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
And here the new `hexdump` based on the same command:

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) (rs_errors . 0) (iterations . -1))
pdu_length = 223
contents = 
0000: 08 ff 5a 00 a4 64 82 9c 8c 40 60 a4 a6 60 60 a6 
0010: 40 6f 00 f0 16 42 02 00 01 00 42 00 00 00 00 00 
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 00 00 1b 1b ff 67 1f 20 
0050: 25 0e aa b1 40 60 f4 3c 01 00 24 00 f0 1c de 51 
0060: 1c 3c 90 0e 66 e1 58 aa 0c 5b 00 00 00 00 00 00 
0070: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00d0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

The seem to look the same, maybe the branch that I am using is the reason for no difference in the output?

janvgils, 2021-03-19:
Can you please have a look at my last update just after you closed the issue.

daniestevez, 2021-03-19:
Hi Jan, I forgot to cherry-pick these commits in `maint-3.8`. I've just done so. Please test now.

janvgils, 2021-03-20:
I will and be back to the both of you to share the results.

janvgils, 2021-03-20:
With the last build I get the following output:

`gr_satellites 99724 --rawfile Orbicraft-Zorkiy_SignalSample_float.iq --samp_rate 480e3 --iq --hexdump`

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) (rs_errors . 0) (iterations . -1))
pdu_length = 90
contents = 
0000: a4 64 82 9c 8c 40 60 a4 a6 60 60 a6 40 6f 00 f0 
0010: 16 42 02 00 01 00 42 00 00 00 00 00 00 00 00 00 
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 1b 1b ff 67 1f 20 25 0e aa b1 
0050: 40 60 f4 3c 01 00 24 00 f0 1c 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) (rs_errors . 0) (syncword . 0) (iterations . -1))
pdu_length = 27
contents = 
0000: a4 64 82 9c 8c 40 60 a4 a6 60 60 a6 40 6f 00 f0 
0010: e1 ff 02 00 01 00 03 00 00 26 06
```
Is this what would expect?

bychek, 2021-03-20:
Looks ok.

janvgils, 2021-03-20:
Thanks, then lets see if it will decode the satellite telemetry when it is launched tomorrow.

commit 6f82918e (2021-03-19): Crop AX.25 frame in USP deframer

This adds a block that reads the lenght field in a USP packet
and crops out the encapsulated AX.25 frame. This block is used
in the USP deframer so that it outputs AX.25 frames.

This closes #248
commit e5c5b592 (2021-03-19): Crop AX.25 frame in USP deframer

This adds a block that reads the lenght field in a USP packet
and crops out the encapsulated AX.25 frame. This block is used
in the USP deframer so that it outputs AX.25 frames.

This closes #248

(cherry picked from commit 6f82918ef11e6e5485f736474827a9d0e52ca082)