daniestevez/gr-satellites #252: SMOG-1 packets missing last two bytes(CRC)

nepeee, 2021-03-26:
I tested the SMOG-1 decoder and the BME submitter rejected all the pockets with checksum error.
I done some comparison with a known good pocket from the BME site recent packets lists and it seems like the packets decoded by gr-satellites is 2 byte shorter.

Valid beacon packet from the BME site:
`0459B85D6048656C6C6F20776F726C642C207468697320697320534D4F473120484135424D45000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000010D00B1A35D60000AB80000280100183439323437316183415084416885416D9BB6D20E57F415131089`

Invalid beacon packet from gr-satnogs:
`049198519648656C6C6F20776F726C642C207468697320697320534D4F473120484135424D450000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000043C000000010038343932343731610000000000000000005E4845C0936EE56F`

Thanks as always!

daniestevez, 2021-03-26:
When there is a CRC, gr-satellites usually strips the CRC after checking its validity. However, in some special cases, the CRC is not stripped. In this case, I think it makes sense not to strip it, since that's what the BME server is expecting.

I'll push a modification to avoid stripping the CRC.

daniestevez, 2021-03-26:
I just pushed a commit that avoids stripping the CRC-16. Please test if this works.

nepeee, 2021-03-26:
Yes it's working. Thanks!

I forgot to mention in the previous comment that i also modified some files to get it going:
SMOG-1.yml line 2 to norad: 99736
telemetry_submit.py line 54 to satellites = {44830 : 'atl1', 44832 : 'smogp', 99736 : 'smog1'}

daniestevez, 2021-03-26:
Thanks! I've just added that modification as well.

closed 2021-03-26 by nepeee