daniestevez/gr-satellites #253: UPMSat-2 no longer being decoded

janvgils, 2021-03-28:
I am not sure where this issue started but it seem gr-satellites is no longer able to decode UPMSat-2.

I have used the following observation `https://network.satnogs.org/observations/3702218/`

`gr_satellites 46276 --wavfile UPMSAT-2_satnogs_3702218_2021-02-28T10-09-27.wav --samp_rate 48e3 --hexdump`

daniestevez, 2021-03-28:
Hi Jan,

I've checked that particular SatNOGS recording, and it's very saturated (open it with Audacity and zoom into a packet). I wonder if we can get a better recording?

janvgils, 2021-03-28:
Hi Dany,

I will have a look if we can find a better one.

janvgils, 2021-03-28:
Is this a better candidate to do some testing?

`https://network.satnogs.org/observations/3853893/`

daniestevez, 2021-03-28:
No. Same problem. I think the problem is with all the SatNOGS OGG recordings. I have [an issue](https://gitlab.com/librespacefoundation/satnogs/satnogs-flowgraphs/-/issues/17) open since quite a while ago regarding this problem (basically the wrong gain is set in the Quadrature Demod block), but they haven't fixed it yet.

janvgils, 2021-03-28:
Thanks for the support Daniel,

Then we leave it for now, maybe I can ask @kng for an IQ recording or create one myself and see how that will perform.

kng, 2021-03-28:
I have one, I'm guessing 48k. That one seems to be demodulated fine with gr_satellites.
https://network.satnogs.org/observations/3702218/
https://drive.google.com/open?id=1Jd4s9IxCyYRhmZ8HhB-rZ2h8wDOr0I9U

janvgils, 2021-03-28:
Thanks for the update.

I gave it a try with the following cli parameters, but it didn't decode.

`gr_satellites 46276 --rawfile UPMSAT-2_3702218_2021-02-28T10-09-27_0.raw --samp_rate 48e3 --iq --hexdump`

janvgils, 2021-04-01:
Daniel,

Is it possible to have a look if this IQ file, I am not able to decode UPMSat 2 and this was possible in earlier versions.

daniestevez, 2021-04-02:
Hi Jan,

I'm getting good results with
```
gr_satellites UPMSAT2 --rawint16 3702218_2021-02-28T10-09-27_0_UPMSAT-2.raw --samp_rate 48e3 \
    --iq --hexdump --disable_dc_block  --deviation 500 --clk_bw 0.15
```

The `--rawint16` is important because the file is int16 rather than complex64. The `--disable_dc_block` is a good idea because of the long preambles and postambles of 0x7e flags, which have significant DC offset. The `--deviation` parameter is important, because with the default deviation we will get a small amplitude into the clock recovery, which won't work properly. Finally, I needed to up the clock recovery bandwidth to get good results.

janvgils, 2021-04-02:
Hi Daniel,

Thanks for the update and sorry for using the wrong file input methode. I will do some further testing and see if we can add these options to the gr-satellites SatNOGS integration.

@kng We can of course create a case structure in `grsat_wrapper.sh` but that will introduce a lot of manual configuration when more satellites will need extra command line parameters.

Food for thought

janvgils, 2021-04-02:
Hi Daniel,

Would it be possible to add the satellite specific options to the corresponding Satellite yaml file?

daniestevez, 2021-04-03:
Hi Jan,

Regarding the deviation, yes, this can be added to the SatYAML file. Maybe it's a good idea to add it for UPMSat-2, although we don't have good values for most of the satellites, either from documentation or from measurements.

Regarding the rest of the options, my opinion is that it would be very difficult to maintain different demodulator options for the large number of satellites that gr-satellites supports. Optimal demodulator options for a particular recording may depend on the receiver, then SNR, the Doppler, and other factors. Also, it's difficult to argue and prove why a particular choice of demodulator options gives better results that another (unless you compare performance over a large set of recordings). Moreover, with demodulator improvements or changes, the best demodulator options might change slightly. gr-satellites ships with default demodulator options that tend to work good across a large range of situations, and these can be overridden via the command line.

It might make sense to have a config file so that users can override the defaults for particular satellites, but I wonder how useful would people really find this feature.

janvgils, 2021-04-03:
Hi Daniel,

Thank you for the elaborate explanation, much appreciated.

So we are again back to the question, "information" for the specific satellite. Then it would be good to know what options are available in the yaml files so that volunteers can suggest options and distribute working yaml files via the gr-satellites repository.

Are all the yaml file options, variables somewhere available?

This is what I could find `https://gr-satellites.readthedocs.io/en/latest/satyaml.html?highlight=yaml`

Regarding the config file, maybe it is an idea to support an option where one can call gr_satellites with an option that uses something like:

`gr_satellites <satellites identifier> --cfg ` and then will use `${HOME}/.gr_satellites/<satellites identifier>.cfg` as commandline options.

Just an idea, and maybe I am missing something very basic that would make this difficult.

I hope @kng and others can also share there thoughts.

daniestevez, 2021-04-04:
Hi Jan,

The [documentation you found about SatYAML](https://gr-satellites.readthedocs.io/en/latest/satyaml.html) is all there is. I think this is a good starting point, but it could see some improvement. For instance, options regarding the modulation, like deviation, are not explained, and maybe the whole section of this document could be organized a bit better.

Your suggestion about satellite-specific config fies sounds good. Let's see what others suggest.

kng, 2022-10-05:
I'm doing some testing with this branch https://github.com/kng/gr-satellites/tree/cfgfile

daniestevez, 2022-10-05:
Hi, thanks for sharing this. I think that the idea is good. However, I would suggest using only NORAD IDs in the `sat.cfg` file as keys for the satellites. Since `gr_satellites` can be run using either satellite names (allowing some differences in case, whitespace and symbols), NORAD IDs, or paths to a YAML file, using `sys.argv[1]` to search in `sat.cfg` would be potentially too error prone. I think that `gr_satellites` should open the SatYAML file corresponding to the satellite that the user has specified as usual, search the NORAD in the SatYAML file it has just opened, and then use that to search in `sat.cfg`.

janvgils, 2022-10-05:
Does python have the option `DefineOptinal` ${HOME}/.gr_satellites/`${norad}.cfg`

kng, 2022-10-05:
I was trying to find the norad id parsed out, but only available inside the flowgraph. Was hesitant to do that computation again. In either case, should I use or copy [this](https://github.com/daniestevez/gr-satellites/blob/main/python/core/gr_satellites_flowgraph.py#L363) ?
Other solution is to add a get_norad() in flowgraph and read it after [add_options](https://github.com/daniestevez/gr-satellites/blob/main/apps/gr_satellites#L154) is run ?

daniestevez, 2022-10-05:
> Does python have the option `DefineOptional` ${HOME}/.gr_satellites/`${norad}.cfg`

What do you mean?

daniestevez, 2022-10-05:
I'm thinking that the static method `gr_satellites_flowgraph.open_satyaml()` should better be a method of [`SatYAML`](https://github.com/daniestevez/gr-satellites/blob/main/python/satyaml/satyaml.py#L24), and the prototype should be `open_satyaml(file=None, name=None, norad=None)`.

Then, at the beginning of `gr_satellites`'s [`main()`](https://github.com/daniestevez/gr-satellites/blob/main/apps/gr_satellites#L298), we could do
```
norad = yamlfiles.open_satyaml(**parse_satellite(argv[1]))['norad']
```
to get the NORAD ID.

kng, 2022-10-08:
Hopefully made some useful changes (: tried to implement what you suggested, the [branch](https://github.com/kng/gr-satellites/tree/cfgfile) is updated.

janvgils, 2022-10-11:
I use this define on Apache, if the file exists, use it.