daniestevez/gr-satellites #255: Error trying to decode S-NET A (handler caught exception: name 'l' is not defined)

janvgils, 2021-04-14:
When trying to decode one if the S-NET satellites with the following command I receive the below error message.

`gr_satellites 43187 --wavfile S-NET_A-satnogs_3944435_2021-04-14T11-07-15.wav --samp_rate 48e3 --hexdump`

To make sure it wasn't a build error, I started with a `git pull && cd build && rm -rf * && cmake ../ && make && sudo make install && sudo ldconfig && gr_satellites --version`

I believe that all the extra dependencies are met:

```
swig                                   4.0.2-1
swig4.0                                4.0.2-1
python3-requests                       2.25.1+dfsg-2
liborc-0.4-0:amd64                     1:0.4.32-1
liborc-0.4-dev:amd64                   1:0.4.32-1
liborc-0.4-dev-bin                     1:0.4.32-1
construct                              2.10.56
```

```
gr_satellites v3.8.0-git
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Error message:

```
handler caught exception: name 'l' is not defined
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/gr/gateway.py", line 78, in eval
    try: self._callback(arg)
  File "/usr/local/lib/python3/dist-packages/satellites/snet_deframer.py", line 44, in handle_msg
    if not all((decode_bch15(ltu[j, :]) for j in range(14))):
  File "/usr/local/lib/python3/dist-packages/satellites/snet_deframer.py", line 44, in <genexpr>
    if not all((decode_bch15(ltu[j, :]) for j in range(14))):
  File "/usr/local/lib/python3/dist-packages/satellites/bch15.py", line 121, in decode_bch15
    errors = compute_error_locations(syndromes)
  File "/usr/local/lib/python3/dist-packages/satellites/bch15.py", line 79, in compute_error_locations
    L[3] = gf_mult(l[3], inv_det_S)
NameError: name 'l' is not defined
thread[thread-per-block[1]: <block snet_deframer(20)>]: SWIG director method error. Error detected when calling 'feval_p.eval'
```
The gr_satellites command will not come back to the OS prompt, so this give me the idea that it is waiting for another function to complete. I need to stop the command with a `CTRL-C`

I have used the same SatNOGS observation with the SNET decoder from UZ7HO and it is producing frames.

daniestevez, 2021-04-14:
Thanks for reporting!

It turns out that this is a bug that I caused when renaming the variable `l` as `L`, and I inadvertently left a couple of occurences without renaming. I have pushed a commit that should fix this.

janvgils, 2021-04-14:
I will update and build and see if this solved the `bug`

janvgils, 2021-04-14:
The fix solved the error, I don't have a decode with this observation but I'll first look to see if I find a better observation and try some command line option to tune the behavior.

This issue can be closed,

janvgils, 2021-04-14:
Found another obs.

`gr_satellites 43187 --wavfile S-NET_B-satnogs_3911741_2021-04-08T10-46-28.wav --samp_rate 48e3 --hexdump`

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 AFSK downlink) (SNET SrcId . 0))
pdu_length = 69
contents = 
0000: f3 50 0d f8 00 00 2c 39 4c 46 dd 4f 01 01 0f 02 
0010: 22 00 00 04 04 03 00 46 00 f3 ff f9 ff 54 f5 e3 
0020: f7 9d 07 b2 56 dc 1e 98 54 00 00 00 13 18 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 74 8e a9 ba 92 
0040: 0c 45 eb 08 00 
***********************************
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 AFSK downlink) (SNET SrcId . 0))
pdu_length = 51
contents = 
0000: f3 50 34 40 00 01 2c 27 4c 42 dd 4f 52 2f 55 27 
0010: 8a 1b 58 09 a5 c5 42 a5 b9 3a 58 58 5b 0a 4d c0 
0020: b2 b4 1c 9e 42 6a eb 28 fb 13 04 b6 f1 98 7c 3d 
0030: 09 01 04 
```

daniestevez, 2021-04-14:
Perfect! Closing.

closed 2021-04-14 by daniestevez