daniestevez/gr-satellites #265: Can not decode a recorded signal from nanocom ax100 transciever

mkamel27, 2021-04-29:
Hello,
I am trying to use gr-satellites to decode a recorded signal of a nanocom ax100 transceiver with no luck. I think the problem may be in the sync word, how can i change the syncword ? can i send you the file to decode ?

daniestevez, 2021-04-30:
Dear Mohamed,

Could you please give us more information about how you're configuring the AX100 (it supports several modes of operation) and how you are are using gr-satellites to decode the signal?

The standard syncwords for the ASM+Golay and Reed-Solomon are hardcoded somewhere in the gr-satellites deframers, but I've never seen a satellite using a non-standard syncword, so perhaps this isn't the cause of your problem.

daniestevez, 2021-04-30:
Those look like completely standard settings of the AX100 transceiver, and your flowgraph should be able to decode the frames.

There is something about the sync word that perhaps is worth mentioning (taken from [this post](https://destevez.net/2018/01/a-first-look-at-ty-2/) I wrote several years ago):

> [...] the manual gives the syncword as [0xC9D08A7B] MSB, but this is misleading. This is only correct if each byte is sent starting by the least significant bit, which is not the case for the length field and data.

So 0xC9D08A7B is the sync word in LSB first notation, which is counter-intuitive, since the AX100 sends all the data MSB first.

I can't see anything wrong with your flowgraph. I would typically decimate before going into the FSK demodulator, but it should work nevertheless. So this leaves signal quality aspects as possible causes of the problem.

mkamel27, 2021-05-01:
Thank you for your reply. I will try again with a better recording.

mkamel27, 2021-05-01:
Hello @daniestevez,
I tried a better recording and added bit slicer and a file sink after the fsk demodulator. I can find the 100 byte preamble of 0XAA but can not find the syncword. Any recommendation ?

daniestevez, 2021-05-02:
Are the demodulated symbols clean and with no bit errors? See the [gr-satellites documentation](https://gr-satellites.readthedocs.io/en/latest/command_line.html#dump-internal-signals) for how to plot these with the command line tool, or just use a QT Time Sink as in `examples/components/fsk_demodulator.grc`.

mkamel27, 2021-05-02:
[image]
Here is the output of the fsk_demodulator.

daniestevez, 2021-05-02:
That looks good, other than the fact that the frame seems quite short (should it be so short?). I haven't visually checked if the syncword is correct.

mkamel27, 2021-05-03:
I think it should be short because the data is only 4 bytes. Is it possible to find the syncword ? Isn't the syncword scrambled ?

daniestevez, 2021-05-03:
The syncword is not scrambled in ASM+Golay mode.

mkamel27, 2021-05-03:
I am out of ideas, whateo you recommend ?

daniestevez, 2021-05-03:
I recommend you look carefully at your Audacity screenshot, which is something that I haven't done. Try to find if the syncword is there. It should be
```
_syncword = '10010011000010110101000111011110'
```

Look at the Golay-encoded header following the syncword. Is it correct? Look at the rest of the data. Does it match the payload you intend to transmit?

daniestevez, 2021-05-23:
Hi @mkamel27, any progress with this? Did you manage to check if anything in the waveform in Audacity resembles the data you intended to transmit?

mkamel27, 2021-05-23:
Hello,
I could not extract the data using audacity so I dumped the binary data using gnuradio. I found the preamble but I could not find the payload or the syncword at all.

daniestevez, 2021-05-23:
Then I don't know what your problem is. In the ASM+Golay mode the syncword is sent unencoded, so other than the weird endianness choice in the user manual that I mentioned above, you should be able to see it in plain sight. Maybe your clock recovery is wrong? Have you tried to do the same with the FSK demodulated waveform (before clock recovery)? The other possibility is that you AX100 is not correctly configured.

mkamel27, 2021-05-23:
I have tried a custom fsk demodulation using gnuradio blocks with no luck. I also suspect the ax100 configuration. What do you think I should check in the configuration ?

daniestevez, 2021-05-23:
No idea. I don't have any first hand experience with the configuration of the AX100. Maybe you can try to get GOMspace in the loop for some support?

mkamel27, 2021-05-23:
I will try. Thanks for your support

daniestevez, 2021-07-01:
@mkamel27 since this might be a configuration problem with the AX100, can we close this issue?

mkamel27, 2021-07-05:
Yeah sure. Thanks though.

closed 2021-07-05 by mkamel27