daniestevez/gr-satellites #267: Physical layer operations with software

kaushiksv, 2021-05-12:
I'm trying to implement a source block for plop-2 as defined in the [231.0-B-3 Blue Book](https://public.ccsds.org/Pubs/231x0b3.pdf) for my master's thesis. I'm a Comp Sci student actually. It will output 10.. sequence at a predefined rate, and send out CLTU frames when available (with a minor delay). If this will be useful in real life, I wish to create PR and merge it here after suitable modifications.

Basically this is it:
[image]

Would love to have more inputs.

Thanks

daniestevez, 2021-05-12:
Hi! I think that anything having to do with CCSDS can fit in gr-satellites. In fact we have some CCSDS block regarding the TM and TC Space Data Link protocols and the Space Packet protocol that Athanasios Theocharis did during ESA summer of code in space, so your contribution would be more than welcome.

A PLOP-2 use case is rather typical, where the use wants to send frames only when available but wants to keep an modulated carrier constantly. It is not so straightforward to implement this in a satisfactory manner in GNU Radio (mainly regarding latency) because of the buffers between blocks. For instance, see [this GRCon talk by Dan CaJacob](https://www.youtube.com/watch?v=gVCCQ6rX3nk) for some of the possible pitfalls and solutions. How do you plan to tackle these in your PLOP-2 block?

Regarding interface, from your mock up I see that you plan on getting the frames by TCP/UDP. Since this is GNU Radio, wouldn't it make more sense to get the frames from a PDU input in the CCSDS PLOP-2 block. If the user needs TCP/UDP, then they can use any of the available blocks, and this also gives the possibility of producing PDUs within GNU Radio (for example via the Message Strobe, which would be useful for basic testing).

Can you give more details about how you plan to implement this or what exactly would the PLOP-2 block do?

kaushiksv, 2021-05-14:
Hi!

Nice to know. Yes, I will be adding PDU support. The block basically implements what is mentioned in 231.0-B-3 Blue Book.

1. Begins in CMM-1 state (continuous zeros).
2. When a client connects (or a command is received), it switches to CMM-2 state (acquisition sequence), then shortly after that, to CMM-4 (idle sequence), .
3. A seperate thread works on the incoming data to produce "CLTU"s and queue them in a circular buffer. Eventually it gets consumed during next call to `work()` (briefly moves to CMM-3).
4. Goes to back to CMM-4 (idle sequence) and stays there for a minimum duration (configurable). Then sends out another CLTU if available (CMM-3 again).
5. Finally when client disconnects or when a command is received, switches back to CMM-1 (carrier only). At this point, the bit transition generator and many internal variables are reset.

There is a working prototype written in python. I've tested with following setting so far:
> Latency: 0.1 sec (output_rate/10)
> Burst size 0.05 sec
> Output rate 28.8 kbps (one bit in each output byte)
The will always strive to have sent `[<time_since _beginning> + 0.1] * samp_rate` . There are internal counters to keep track of time and number of output elements sent out. Number of output elements to be sent out at any given time has to be greater than burst size, otherwise `work()` will return 0.

Seems to work pretty well so far. I also did on-air test with sps=4 and two usrp equipments (tx and rx). Works smoothly. Of course, I will do more tests in different settings. Especially with regard to burst size.

Perhaps I should make the code more elegant, and then make a PR.

kaushiksv, 2021-05-14:
I like to introduce few more optimizations (like use of `popcnt` intrinsic), and vmcircbuf. if you happen to have some inputs regarding the usage of vmcircbuf and also benchmarking (both python/cpp), that would be great. Currently I am to get gr-perf-monitorx to work.

daniestevez, 2021-05-14:
So you control the number of samples you emit from your PLOP-2 block by counting time? What happens in the long run if the SDR transmitter clock is slightly faster than the computer clock? I think you would starve the SDR from samples.

kaushiksv, 2021-05-15:
Yes. The block keeps track of number of output elements sent, and will try to maintain the  number near (t*samp_rate + L), where L is the number of extra samples that ought to have been sent at any given moment. This number can be configured by the user in block options. This will introduce a small latency. Also, the burst size is always comparable to the samp_rate. So if the consumption rate is approaching the production rate, I presume work() will be called way more often, indicating an impending underrun. Then we can set L=2L. This needs to be tested of course. I am fairly sure though, because I noticed enormous CPU before switching to burst mode. The idea of `--nperiods` and `--periods` options at [https://linux.die.net/man/1/jackd](https://linux.die.net/man/1/jackd) sounds similar.

daniestevez, 2021-05-15:
I'm not completely sure, but I suspect that you'll eventually run into problems if you try to control how many samples are present downstream of a particular block by producing more or less samples with that block depending on how your work function is called. Here is a [talk by Matt Ettus](https://www.youtube.com/watch?v=jq0RewceCwc) where he uses some feedback to solve this kinds of problems.

kaushiksv, 2021-05-15:
Hi. Thank you so much for that. I was thinking about some sort of feedback but wasn't sure how to go about it. This latency manager idea looks great. Will try to incorporate it.

daniestevez, 2022-05-22:
Hi @kaushiksv did you make any progress about this?

kaushiksv, 2022-05-22:
Hi. No. Well my thesis is complete, but I haven't been able to work on incorporating it here yet. Also, I guess the 231.0-B-4 supersedes 231.0-B-3. I guess we can close the issue for now. I'll reopen when I start to work on it and create a pr when it is ready :)

closed 2022-05-22 by kaushiksv

daniestevez, 2022-05-23:
Thanks! Feel free to reopen in the future.

I think the main change in B-4 is the addition of USLP, so the update doesn't affect this issue.