daniestevez/gr-satellites #268: GOMspace AX100 Deframer

luisguarnizo1234, 2021-05-20:
Hola Dani, espero estes muy bien.

Dani, actualmente estoy utilizando el bloque GOMsapce AX100 Deframer de gr-satellites, pero tengo una duda, revisando el manual del radio AX100 que está disponible en la página de GOMspace

daniestevez, 2021-05-23:
Hola Luis. El bloque AX100 deframer maneja las capas de randomization y Reed Solomon FEC. El CRC supongo que se trata del CRC-32C de CSP. Hay algún bloque en gr-satellites para encargarse de ello, aunque la forma en la que distintos satélites usan este CRC varía (en aspectos como el endianness o si la cabecera CSP está incluida en el cálculo del CRC). Actualmente gr-satellites no incluye soporte para el HMAC.

luisguarnizo1234, 2021-05-28:
1. Dani, gracias por tu respuesta. entonces con este bloque de GOMSpace AX100 Deframer en el modo ASM+GOLAY ya estaría realizando la parte de Randomization y Reed Solom FEC, quisiera saber que polinomio usa ese bloque para la randomization, según el manual del radio usan CCSDS h(x) = x8 + x7 + x5 + x3 + 1, es esté el mismo que usa el bloque?

2. cuando agrego el bloque de visualización de print CSP Header ya estaría ejecutando el CRC32 o debo implementar un bloque adicional para esto, porque cuando coloco el bloque de CSP Header ya estoy obteniendo datos en Hexadecimal y coinciden con el formato de la telemetría.

gracias Dani

daniestevez, 2021-05-28:
1. Sí, es el mismo scrambler.
2. Hay un bloque llamado "Check CRC" para comprobar el CRC-32C de un paquete CSP.

luisguarnizo1234, 2021-05-28:
Gracias Dani por la respuesta, Dani ahora que yo tengo mis tramas CSP decodificadas quisiera hacer una aplicación que me permita  recibir estos paquetes CSP en un cliente de la librería LIBCSP y enviarlos a un servidor con LIBCSP que los pueda entender. Tú tienes alguna recomendación para hacer esto?

gracias

daniestevez, 2021-05-28:
Quizá es buena idea sacar los paquetes CSP de GNU Radio usando ZMQ.

luisguarnizo1234, 2021-05-28:
Gracias Dani,

entiendo, yo ya tengo un ZMQ proxy corriendo entre un cliente y un servidor CSP, podría utilizar ese mismo ZQM por medio del proble ZQM PUB SINK
[image]

daniestevez, 2021-05-30:
Para enviar PDUs debes usar los "Message Sink". El endpoint que uses debe coincidir entre GNU Radio y el servidor CSP.

luisguarnizo1234, 2021-06-01:
Gracias Dani, intentaré algo con ese bloque. Dani tú sabes que método de FEC se implementa en el bloque ASM + GOLAY, yo conozco que el AX100 cuenta con 12 bit para FEC, sin embargo, el manual no dice que método utiliza para este propósito

daniestevez, 2021-06-01:
Es un código Golay(24,12).

luisguarnizo1234, 2021-06-01:
Entiendo Dani, ya revisé la documentación que tienes sobre este método de FEC (https://destevez.net/2018/05/algebraic-decoding-of-golay2412/) y revisé el libro de the art of error correcting coding, pero me queda la duda de cómo obtuviste los parámetros de la matriz H, y que en el libro son diferentes a los que utilizas para implementar el código GOLAY
https://destevez.net/2018/05/algebraic-decoding-of-golay2412/

[image]

[image]

daniestevez, 2021-06-01:
Puedes leer la nota al final de este post https://destevez.net/2016/09/decoding-gomx-1-telemetry/

luisguarnizo1234, 2021-06-01:
Según veo en la nota, el orden de los bits de la matriz del libro debe invertirse, sin embargo no entiendo por qué la matriz del código tiene más bits que la matriz que la del libro

daniestevez, 2021-07-31:
@luisguarnizo1234 podemos cerrar este issue?

luisguarnizo1234, 2021-07-31:
Hola Daniel, muchas gracias. Si, creo que ya s3 puede cerrar

closed 2021-07-31 by daniestevez