daniestevez/gr-satellites #269: AALTO-1.yml - NameError: name 'syncword' is not defined

janvgils, 2021-05-23:
Good morning Daniel,

After tuning the AALTO-1 SatNOGS transmitter downlink frequency I was trying to get it to produce positive decodes but I am running into the following error when running gr_satellites.

Observation used: `https://network.satnogs.org/observations/4148774/`
CLI command used: `gr_satellites 42775 --wavfile AALTO-1_satnogs_4148774_2021-05-23T04-35-14.wav --samp_rate 48e3 --hexdump`

Satyaml file used: `cat $PYTHONPATH/satellites/satyaml/AALTO-1.yml`

```
name: AALTO-1
norad: 42775
data:
  &tlm_ax25 AX.25 Telemetry:
    telemetry: ax25
  &tlm_cc1125 CC1125 Telemetry:
    unknown
transmitters:
  9k6 FSK AX.25 downlink:
    frequency: 437.216e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm_ax25
  9k6 FSK CC1125 downlink:
    frequency: 437.216e+6
    modulation: FSK
    baudrate: 9600
    framing: AALTO-1
    data:
    - *tlm_cc1125
```

Error output:

```
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 316, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 301, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 189, in __init__
    self.flowgraph = satellites.core.gr_satellites_flowgraph(
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 160, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowgraph.py", line 271, in _init_demodulator_deframer
    deframer = self.get_deframer(transmitter['framing'])(
  File "/usr/local/lib/python3/dist-packages/satellites/components/deframers/aalto1_deframer.py", line 51, in __init__
    packlen=255, sync=syncword, threshold=syncword_threshold)
NameError: name 'syncword' is not defined
```

If the time allows, could you have a look at this error.

daniestevez, 2021-05-23:
Hi Jan, thanks for detecting this bug. It was a silly mistake. I deleted an underscore preceding the `syncword` variable name when reformatting the code. I've just pushed a commit that should fix this.

janvgils, 2021-05-23:
Hi Daniel,

Thanks for the update.

After pulling the update and performing a build everything look fine.

CLI command: `gr_satellites 42775 --wavfile AALTO-1_satnogs_4148774_2021-05-23T04-35-14.wav --samp_rate 48e3 --hexdump`

Output:

```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK AX.25 downlink))
pdu_length = 148
contents = 
0000: 9e 90 64 82 8e a6 00 9e 90 64 82 62 a6 17 03 f0 
0010: d6 db a9 60 e0 4a 0d 00 03 19 4e ff 01 ff ff 2e 
0020: d3 14 00 4f 12 6d 0d 01 01 00 00 01 10 4d 10 00 
0030: 00 00 00 00 eb 00 00 00 00 00 00 02 f4 00 13 00 
0040: 39 00 04 00 02 03 06 03 59 03 5d 03 75 01 4b 00 
0050: 01 00 03 02 40 00 02 00 06 03 d1 00 02 00 be 03 
0060: 00 00 00 00 00 00 00 00 00 00 20 70 00 00 00 00 
0070: 00 00 00 00 00 2f ff ff 00 0d 59 35 01 ab 00 00 
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0090: 00 00 00 00 
***********************************

```

If you agree that the output is ok, then the issue can be closed.

Enjoy the weekend, Jan

daniestevez, 2021-05-23:
Thanks for testing. Everything seems OK, so I'm closing this.

closed 2021-05-23 by daniestevez