daniestevez/gr-satellites #270: BOBCAT-1 1k2 decoding RuntimeError

janvgils, 2021-06-01:
Lately all BOBCAT-1 1200 baud gr_satellites SatNOGS decodes throw in a error and cause gr_satellites to crash indicated by the below error message `No such process`.

```
Jun 01 18:05:46 canberra satnogs-client[29946]: gr-satellites: Observation: 4199063, Norad: 46922, Name: BOBCAT-1, Script: satnogs_fsk.py
Jun 01 18:05:46 canberra satnogs-client[29946]: gr-satellites: Starting observation 4199063
Jun 01 18:05:46 canberra satnogs-client[29946]: gr-satellites: running at 48000 sps
Jun 01 18:05:47 canberra satnogs-client[29946]: [INFO] Using format CF32.
Jun 01 18:10:13 canberra satnogs-client[29946]: gr-satellites: Observation: 4199063, Norad: 46922, Name: BOBCAT-1, Script: satnogs_fsk.py
Jun 01 18:10:13 canberra satnogs-client[29946]: gr-satellites: Stopping observation 4199063
Jun 01 18:10:13 canberra satnogs-client[29946]: /usr/local/bin/grsat-wrapper.sh: line 74: kill: (6212) - No such process
Jun 01 18:10:13 canberra satnogs-client[29946]: gr-satellites: Keeping logs, you need to purge them manually.
```

An example observation is `https://network.satnogs.org/observations/4199063/` the decoder in this examples isn't using the audio out but the doppler corrected IQ via localhost.

Command line used:

`gr_satellites $NORAD --samp_rate 48000 --iq --throttle --udp --udp_port $UDP_DUMP_PORT --udp_raw --start_time $DATEF --kiss_out $KSS --ignore_unknown_args --use_agc"`

gr_satellites error: `RuntimeError: firdes check failed: 0 < fa <= sampling_freq / 2` details below.

```
::::::::::::::
grsat_4193936.log
::::::::::::::
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb2 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb2 - Mask: ffffffff
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 316, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 301, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 192, in __init__
    **sat)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowg
raph.py", line 162, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "/usr/local/lib/python3/dist-packages/satellites/core/gr_satellites_flowg
raph.py", line 262, in _init_demodulator_deframer
    **demodulator_additional_options)
  File "/usr/local/lib/python3/dist-packages/satellites/components/demodulators/
fsk_demodulator.py", line 79, in __init__
    1, samp_rate, carson_cutoff, 0.1 * carson_cutoff)
  File "/usr/lib/python3/dist-packages/gnuradio/filter/filter_swig.py", line 143
, in low_pass
    return _filter_swig.firdes_low_pass(*args, **kwargs)
RuntimeError: firdes check failed: 0 < fa <= sampling_freq / 2
```

The 1k2 BOBCAT-1 satyaml definition is as follows:

```
name: BOBCAT-1
norad: 46922
data:
  &tlm Telemetry:
    unknown
transmitters:
  1k2 FSK downlink:
    frequency: 436.600e+6
    modulation: FSK
    baudrate: 1200
    framing: AX100 ASM+Golay
    data:
    - *tlm
```

daniestevez, 2021-06-01:
Hi Jan,

Thanks for reporting. This is related to #207. I've checked the code and the IQ case was not handled properly, so I'll submit a fix right away.

daniestevez, 2021-06-01:
The fix is now in `master` and `maint-3.8`. Please test this and let me know if it works.

janvgils, 2021-06-01:
Hi Daniel,

Thanks for the quick response and fix.

Both stations (https://network.satnogs.org/stations/431/ and 1696/) running this solution are now updated with this new code, the first BOBCAT-1 passes are at 2021-06-02 13:59:23 UTC, lets see how they will perform and if the issue is solved.

janvgils, 2021-06-02:
Hi Daniel,

Here an update:

This is the observation that was used: `https://network.satnogs.org/observations/4200125/`

The gr_satellites process no longer crashes as can be seen here:

```
Jun 02 13:59:23 canberra satnogs-client[29946]: gr-satellites: Observation: 4200125, Norad: 46922, Name: 0_BOBCAT-1, Script: satnogs_fsk.py
Jun 02 13:59:23 canberra satnogs-client[29946]: gr-satellites: Starting observation 4200125
Jun 02 13:59:23 canberra satnogs-client[29946]: gr-satellites: running at 48000 sps
Jun 02 13:59:24 canberra satnogs-client[29946]: [INFO] Using format CF32.
Jun 02 14:04:05 canberra satnogs-client[29946]: gr-satellites: Observation: 4200125, Norad: 46922, Name: 0_BOBCAT-1, Script: satnogs_fsk.py
Jun 02 14:04:05 canberra satnogs-client[29946]: gr-satellites: Stopping observation 4200125
Jun 02 14:04:05 canberra satnogs-client[29946]: gr-satellites: processing KISS data to network
Jun 02 14:04:05 canberra satnogs-client[29946]: gr-satellites: Keeping logs, you need to purge them manually.
```

There was maybe one log message that would need some further investigation:

`symbol_sync_ff3 - block performing more interopolations per symbol (2.000000) than input samples per symbol(1.250000). Consider reducing osps or increasing sps`

```
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb1 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb1 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb2 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb2 - Mask: ffffffff
gr::log :WARN: symbol_sync_ff3 - block performing more interopolations per symbol (2.000000) than input samples per symbol(1.250000). Consider reducing osps or increasing sps
gr::log :DEBUG: correlate_access_code_tag_bb3 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb3 - Mask: ffffffff
gr::log :DEBUG: correlate_access_code_tag_bb4 - Access code: 930b51de

<snip>
```

```
gr::log :DEBUG: correlate_access_code_tag_bb7 - writing tag at sample 100339
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 FSK downlink))
pdu_length = 160
contents =
0000: 01 80 b7 89 01 01 02 07 0c c2 2c 60 b7 91 b0 00
0010: 04 57 38 50 5a 53 00 42 4f 42 43 41 54 2d 31 00
0020: 40 dc 00 85 00 45 01 99 06 97 00 ad 15 6a 00 ef
0030: 15 6a 00 01 00 02 00 00 01 5f 01 03 4d 02 00 00
0040: 02 59 16 00 23 00 00 00 00 00 32 00 14 00 3e 00
0050: 19 00 46 00 23 00 41 00 14 00 b1 00 04 00 05 00
0060: 04 00 03 ff ff ff ff 03 e0 00 00 02 ad e0 00 00
0070: f2 07 60 b7 91 b0 00 05 00 36 00 3b ff 9a 00 6d
0080: 1f 2a 00 1a 2a e8 4d a0 a7 f5 03 dc b9 2e 03 16
0090: 00 00 01 00 43 32 68 18 09 dd 43 25 58 aa dd d7
***********************************
gr::log :DEBUG: correlate_access_code_tag_bb4 - writing tag at sample 1631711
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 FSK downlink))
pdu_length = 154
contents =
0000: 01 80 b7 89 01 03 02 07 0c c2 2c 60 b7 91 b0 00
0010: 04 57 38 50 5a 53 00 42 4f 42 43 41 54 2d 31 00
0020: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 32
0030: 1e 60 b7 91 b0 00 04 2f 66 6c 61 73 68 2f 63 66
0040: 67 73 2f 4e 47 5f 32 31 30 33 31 39 2e 6a 73 6f
0050: 6e 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
0060: 00 00 00 00 00 00 00 00 00 00 38 ac 58 00 00 00
0070: 58 00 c2 2c 60 b7 91 b0 00 04 00 00 00 00 00 00
0080: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 60 0f
0090: aa 9d a0 d8 b6 7e cf 0b 3f a4
***********************************
gr::log :DEBUG: correlate_access_code_tag_bb1 - writing tag at sample 2060330
```

daniestevez, 2021-06-02:
The warning is innocuous. It just means that we are using a rather low sample rate 48000 for the high baud rate of 38400, so things will not be so optimal.

It seems that this is working correctly now, so I'll close this.

closed 2021-06-02 by daniestevez