daniestevez/gr-satellites #271: One satellite multiple downlink frequencies

janvgils, 2021-06-02:
In this specific case I am using UWE-3 as an example, this satellites and also others that are build by the [Uni Würzburg](https://www.informatik.uni-wuerzburg.de/space/forschung/space-exploration/projects/) use FSK AX.25 G3RUH but have multiple downlink frequencies.

Would it make sense to configure them all in the `UWE-3.yml` file?

```
name: UWE-3
norad: 39446
data:
  &tlm Telemetry:
    telemetry: ax25
transmitters:
  1k2 AFSK downlink:
    frequency: 437.385e+6
    modulation: AFSK
    baudrate: 1200
    af_carrier: 1700
    deviation: 500
    framing: AX.25
    data:
    - *tlm
  9k6 FSK downlink 1:
    frequency: 435.000e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm
  9k6 FSK downlink 2:
    frequency: 436.395e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm
  9k6 FSK downlink 3:
    frequency: 437.385e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm
```

daniestevez, 2021-06-02:
Hi Jan,

The problem with adding several transmitters using exactly the same mode and different frequencies is that people will get duplicated decodes. At the moment the frequency field is not used by gr-satellites. I decided to put it there because it's good documentation. gr-satellites instances a decoder for all the transmitters listed in the SatYAML file, and all of them work in parallel. So if you have multiple transmitters with the same mode then you will get several identical decoder which produce exactly the same output (and so repeated messages are displayed to the user).

This already happened with the OrbiCraft satellites and what we did was just to include the transmitter for one of the frequencies. 

I'm open to suggestions about what to do in the future. Maybe gr-satellites should be smarter and notice that some transmitters are identical and instance only one copy for each type.

janvgils, 2021-06-02:
Hi Daniel,

Thank you for the explanation and I will give it some thought.

janvgils, 2021-06-17:
Good evening Daniel,

Let us close this issue and see if there will be more reason to investigate this further in the future.
The only thing that I can think of currently is some intelligence to do some retuning based on a the frequency configured.

closed 2021-06-17 by janvgils