daniestevez/gr-satellites #272: MIR-SAT1 missing in maint-3.8 branch

janvgils, 2021-06-13:
Good evening Daniel,

This evening after reading an email on the AMSAT-BB I saw that MIR-SAT1 should be supported by gr-satellites.
After pulling the last updates (I am on branch 3.8, because I have no GNURadio 3.9 support yet on Debian bullseye) I saw that the MIR-SAT yaml file wasn't being pulled in.

So I did some searches and saw that it was part of the master branch and created the file manually, this went fine and `gr_satellites --list_satellites` did show MIR-SAT1.

```
* MIR-SAT1 (NORAD 90001)
    9k6 FSK downlink 436.925 MHz FSK AX.25 G3RUH
```

I do wonder, what is the function of the `&file` option?

```
ame: MIR-SAT1
norad: 90001
data:
  &tlm Telemetry:
    telemetry: mirsat1
  &file Files:
    files: mirsat1
transmitters:
  9k6 FSK downlink:
    frequency: 436.925e+6
    modulation: FSK
    baudrate: 9600
    framing: AX.25 G3RUH
    data:
    - *tlm
    - *file
```

I hope it is possible to make MIR-SAT1 available in maint-3.8.

daniestevez, 2021-06-13:
Hi Jan,

Support for MIR-SAT1 is currently in the `mirsat` branch only. At some point soon it will be merged into master and maint-3.8.

The `&file` option is a reference to be used below in the `data: - *file`. MIR-SAT1 is able to transmit image files using the PACSAT protocol, and gr-satellites will include a decoder to receive these files.

janvgils, 2021-06-13:
Thanks for the update, will be waiting patiently.

daniestevez, 2021-06-16:
I merged today the mirsat branch into master and maint-3.8.

janvgils, 2021-06-17:
Thanks for the update, the pull and build worked fine. This issue can be closed.

closed 2021-06-17 by janvgils