daniestevez/gr-satellites #273: AttributeError: module 'satellites.telemetry' has no attribute

vnnspace, 2021-06-15:
Hello.
I have a question I put several different satellites like
KSU CubeSat or LightSail-2 and all-time I has this answer
[image]

but when I put au03 in attribute everything is ok.

What I am doing wrong?
I am already looking at the list of satellites by using 
[image]

I thought maybe i put some wrong name , but i found KSU CubeSat in the list and same issue is appear 
Please explain to me what I should do?

Thank you

daniestevez, 2021-06-15:
Hi!

The list of satellites that can be used _for the telemetry parser_ is in [this file](https://github.com/daniestevez/gr-satellites/blob/master/python/telemetry/__init__.py). There is a [section in the documentation](https://gr-satellites.readthedocs.io/en/latest/components.html#telemetry-parser) that explains how to get this list in another manner.

As you can see, gr-satellites doesn't have (yet) specific telemetry parsers for KSU Sat or LightSail-2.

vnnspace, 2021-06-15:
Thank you very much for that answer. 
In that, I have another question.
I have this flowchart 
[image]

Please look only at the part with the Satellite decoder.
This part gives me an error as I write before, or does it mean that KSU CubeSat is not decoding yet and I will have this error all the time until you add a decoding scheme into your code?
In that case, I have one more question. You write in your description that KSU CubeSat is FSK 
[image]

But on another information, I found that KSU CubeSat is GMSK
[image]

As you see in my scheme I also have GMSK demodulation which information is correct ?

Thank you for your answer.

daniestevez, 2021-06-16:
The error you pasted includes a `telemetry_parser()` block in the Python flowgraph generated by GRC, but I can't see that block in the flowgraph image you've pasted, so I don't understand. Is there some inconsistency? Can you share the .grc file?

Regarding GMSK, it's a special type of FSK, so it is treated by gr-satellites as FSK.

vnnspace, 2021-06-16:
Thank you for your kind answer.
Of course, I can share my file with you. I think maybe I missed some block during the installation of GNURadio or GR-Satellite...
Please look at my file and give me advice.

[BladeRF_GQRX_UDP_KSU_CubeSat.zip](https://github.com/daniestevez/gr-satellites/files/6662653/BladeRF_GQRX_UDP_KSU_CubeSat.zip)

Best Regards

daniestevez, 2021-06-16:
I don't see how that flowgraph can possibly give you the problem with the `telemetry_parser()`. There is no telemetry parser at all.

vnnspace, 2021-06-17:
Thank you for your answer. That is why it looks very very strange.
I am searching for suggestion now, maybe you will find them.
After I will find a solution I will inform you.

vnnspace, 2021-06-17:
I am also thinking about maybe I missed something during the installation gr-satellite?
Maybe the same blocks were not installed? 
How I can change it? or on your PC you have the same result as me?

vnnspace, 2021-06-18:
Good morning 
I launch gnuradio under the sudo and now have another error, Maybe this error will give more clear answer on my question.
[image]

best regards

vnnspace, 2021-06-18:
Good morning I try to do such kind of work with samples from the sample folder.
I found some satellite_decoder - sample and try to use them.
1) it immediately gives to me an error because it was not Throttle on it ( now working )
[image]

It is working if generate option is no GUI
[image]

2) If I made generate option as QT GUI
[image]

immediately happen error
[image]

I am sorry for so many questions but I have one more in case of option 1) I not see any answer any messages no information ... Where I should find decoding data from wav file? Where I can look at it?

Best Regards

vnnspace, 2021-06-18:
Good morning I just deeper investigate option 1) from the previous file and found that it is also an error
[image]

error related to the real-time scheduling ( no problem i can switch of it ) but other parts i do not understand again...

daniestevez, 2021-06-18:
> Good morning
> I launch gnuradio under the sudo and now have another error, Maybe this error will give more clear answer on my question.
> [image]
> 
> best regards

This error points to a problem with your GNU Radio installation. This is a GNU Radio .so file that is not properly linked. Nothing to do with gr-satellites, so given that the other problems you're finding are quite strange, I suggest trying to fix your GNU Radio installation first.

daniestevez, 2021-06-18:
Out of interest: what version of gr-satellites and GNU Radio are you trying to run, by the way?

vnnspace, 2021-06-21:
Thank you for your note, about GNU radio problem, i think so also. But how to install it right i do not know , I am already reinstall it maybe 15 times... and same result . 
I am use GNURadio:
[image]

And GR-Satellite v 4.1

daniestevez, 2021-06-21:
Have you tried using GNU Radio 3.9?

GNU Radio 3.10 (or rather what is currently in the master branch of GNU Radio, since 3.10 hasn't been released yet) is not guaranteed to work with gr-satellites.

vnnspace, 2021-06-21:
I see thank you.
I will try to reinstall today

vnnspace, 2021-06-29:
Good day again.
I am re-install everything on my PC. 
I tried to compile and make gr-satellite (without GnuRadio installed ) and it failed all tests. I do not do make installation yet, just test
[image]

after that, I install GNURadio 3.9 same result 
after that, I reinstall GNURadio 3.10 same result...

I think I am doing something wrong

daniestevez, 2021-06-29:
Can you run `ctest -V` to see more detailed information?

daniestevez, 2021-07-31:
@vnnspace were you able to run `ctest -V` to try to get more information about the problem?