daniestevez/gr-satellites #278: maint-3.8 build warnings (incompatible function types from ‘PyObject*)

janvgils, 2021-06-20:
Good evening Daniel,

After todays merges, I did a pull and build, and was confronted with a lot of warnings.

Below a part of the warnings, an extended log file can be found at: [output.log](https://www.pe0sat.vgnet.nl/download/gr-satellites/build-error-20-06-2021_22:29.log)
```
[ 96%] Building CXX object swig/CMakeFiles/satellites_swig.dir/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx.o
/home/janvg/source/git/daniestevez/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:70355:39: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*, PyObject*, PyObject*)’ {aka ‘_object* (*)(_object*, _object*, _object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
   { "ax100_decode_make", (PyCFunction)_wrap_ax100_decode_make, METH_VARARGS|METH_KEYWORDS, (char *)"ax100_decode_make(bool verbose) -> ax100_decode_sptr"},
                                       ^~~~~~~~~~~~~~~~~~~~~~~
```

At the end the build was completed and `gr_satellites --version` showed an expected version output.

```
gr_satellites v3.9.0-git
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Is this a warning that needs further attention?

daniestevez, 2021-06-21:
Hi Jan,

These warnings come from SWIG, which is used in GNU Radio 3.8 to generate Python bindings for C++ code. SWIG has a reputation for being like black magic and difficult to debug. In GNU Radio 3.9 it has been replaced by Pybind11. With this, I want to say that while these are warnings rather errors and if the software works correctly, I wouldn't worry about it, since debugging this problem can be quite difficult and not worth it.

janvgils, 2021-06-21:
Thanks for the update Daniel,

And I will close this issue.

closed 2021-06-21 by janvgils