daniestevez/gr-satellites #280: Error when using "mirsat1" in Image Receiver block

K4KDR, 2021-06-21:
Platform:  Maint3.8 branch in GNU Radio 3.8.1.0 on Ubuntu 18.04 computer.

---------------------------------------------------------------------------------------------------------

The **image_receiver** example flowgraph runs perfectly using the default ImageReceiver class of "**by70_1**".

However, using the list of available classes from the python3 command 

**`import satellites.filereceiver; help(satellites.filereceiver)`**

```
NAME
    satellites.filereceiver - gr-satellites filereceiver

DESCRIPTION
    This module contains the File Receiver class and child classes
    This class is used to reassemble files transmitted in chunks

PACKAGE CONTENTS
    by70_1
    dsat
    filereceiver
    imagereceiver
    k2sat
    lucky7
    mirsat1
    sat_1kuns_pf
    smogp
    swiatowid

FILE
    /usr/local/lib/python3/dist-packages/satellites/filereceiver/__init__.py
```

... when attempting to use File Receiver class of **mirsat1**, the following error is generated:

```
Executing: /usr/bin/python3.6 -u /home/k4kdr/gr-satellites/examples/components/image_receiver.py

Traceback (most recent call last):
  File "/home/k4kdr/gr-satellites/examples/components/image_receiver.py", line 144, in <module>
    main()
  File "/home/k4kdr/gr-satellites/examples/components/image_receiver.py", line 120, in main
    tb = top_block_cls()
  File "/home/k4kdr/gr-satellites/examples/components/image_receiver.py", line 82, in __init__
    self.satellites_image_receiver_0 = satellites.components.datasinks.file_receiver('mirsat1', '/tmp/', True, display=True, fullscreen=False, options="")
  File "/usr/local/lib/python3/dist-packages/satellites/components/datasinks/file_receiver.py", line 56, in __init__
    verbose, **kwargs)
TypeError: __init__() got an unexpected keyword argument 'display'

>>> Done (return code 1)
```

daniestevez, 2021-06-21:
Hi Scott, you need to use the "File Receiver" block for this. The only difference between "File Receiver" and "Image Receiver" is that Image Receiver uses feh to display the images in real-time as they are being received. File Receiver just saves the files to some folder.

MIR-SAT1 will transmit JPEG2000 and raw files, none of which can be opened with feh. For the JPEG2000 there are other software tools to open it. For the raw format, a few months ago I was working on a simple Python script that converts these to DNG. The DNG files can be opened with a raw image file processor such as Rawtherapee. If people find this useful, I can publish this Python script once the satellite launches (but receiving raw files will be a bit tricky due to their large size; some collaboration between different Amateurs can be useful to assemble all the pieces).

K4KDR, 2021-06-21:
Oh!  Very good - thank you so much for the info.

I suggest that the proper config for receiving MIR-Sat1 be outlined in a prominent place; perhaps I am not the only one who might make an incorrect assumption about the image downlink format.

Thanks!

K4KDR, 2021-06-21:
Related:

By chance is a sample I/Q or Audio file available containing an image downlink transmission from MIR-Sat1?

N6RFM, 2021-06-21:
Thanks also Dani.  Was having the issue.. :-)  Scott and I will work on this for sure!

daniestevez, 2021-06-21:
> I suggest that the proper config for receiving MIR-Sat1 be outlined in a prominent place; perhaps I am not the only one who might make an incorrect assumption about the image downlink format.

With the command line tool no settings need to be done. It will show the telemetry in the console and store received files in a folder (which is `/tmp/` by default but can be changed with a command line parameter).

Those using a flowgraph might need to do more work and use the File Receiver block (what is basically what you've already done).

In any case, any suggestion to improve the documentation is welcome.

> By chance is a sample I/Q or Audio file available containing an image downlink transmission from MIR-Sat1?

Not really, but you can use the simulation that I've made for [this video](https://www.unoosa.org/documents/video/psa/hsti/AccSpace4All/KiboCUBE/Mauritius/mirsat.mp4). The file is [here](https://drive.google.com/file/d/1b-RWyxmSIrx0NGmwzAjO0hFFT3y9QSsR/view?usp=sharing). It is 480ksps IQ complex64, so you can open it with GQRX or with GNU Radio.

K4KDR, 2021-06-21:
Watched the video from your earlier Tweet - much appreciated!

Will download that sample file immediately - thanks so much!

daniestevez, 2021-07-31:
Hi @K4KDR, can we close this issue?

K4KDR, 2021-07-31:
Yes sir - many thanks!

closed 2021-07-31 by daniestevez