daniestevez/gr-satellites #283: Strange TUBIN Mobitex-NX decodes

janvgils, 2021-07-01:
Good evening,

I tried to decode the signals from TUBIN on both gr-satellites master and maint-3.8.
Master isn't working because the Mobitex-NX OOT module source isn't migrated to GNURadio 3.9.

When I try to decode the signals with a maint-3.8 installation and using the following Yaml file I also ran in to issues.

```
name: TUBIN
norad: 99678
data:
  &tlm Telemetry:
    unknown
transmitters:
  1k2 FSK downlink:
    frequency: 435.950e+6
    modulation: FSK
    baudrate: 1200
    framing: Mobitex-NX
    data:
    - *tlm
  2k4 FSK downlink:
    frequency: 435.950e+6
    modulation: FSK
    baudrate: 2400
    framing: Mobitex-NX
    data:
    - *tlm
  4k8 FSK downlink:
    frequency: 435.950e+6
    modulation: FSK
    baudrate: 4800
    framing: Mobitex-NX
    data:
    - *tlm
  9k6 FSK downlink:
    frequency: 435.950e+6
    modulation: FSK
    baudrate: 4800
    framing: Mobitex-NX
    data:
    - *tlm
```

Decode example:

```
# Decoding Satellite 99678.
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 9k6 FSK downlink) . 5)
pdu_length = 0
contents =
```

```
***********************************
Error while submitting telemetry: HTTP Error 400: Bad Request
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 1k2 FSK downlink) . 1)
pdu_length = 520
contents = 
0000: 3b f1 58 64 2e 0c 3f f7 f6 80 39 c8 b0 d6 d5 f3 
0010: 84 93 91 07 20 b9 fd 68 32 16 ec 6a b8 86 2b 1b 
0020: b1 9f 17 43 10 e2 56 6e 2c be d7 70 af 11 9c e6 
0030: 02 c5 28 d7 18 bc 6a d9 57 97 32 c7 c4 ce 61 37 
0040: 86 12 9f 2b 20 73 07 78 51 b8 68 14 50 58 85 9b 
0050: c4 90 9a 65 b7 79 75 d4 85 58 4d ec 49 ee 99 e2 
0060: fe d0 75 d8 1c 6a 69 bc 4b 64 76 88 7a 8f 0b 38 
0070: a9 66 34 3d 42 33 8b c7 6c 51 27 1e 39 b0 53 21 
0080: bb 0a 04 5f 67 a8 2e 6d cb 98 82 28 eb b2 94 df 
0090: 77 02 2b 50 b3 b4 dc 34 be 07 f3 5d 4e 55 25 2f 
00a0: aa 61 ac 01 3c 60 2d c3 64 12 e4 c6 e6 ba f2 d4 
00b0: 7e 93 7b 17 01 2a 7e 4e 94 c6 4e 8f 0a f7 ed 08 
00c0: c4 dc cf 0c ef 2c 3c 38 3d 5a 3d f3 d2 5e 28 f0 
00d0: a9 29 09 39 12 75 70 06 0c 63 a1 d8 f4 15 da 96 
00e0: 29 96 f5 7d f6 77 05 33 93 05 e7 0e 2d 72 7a d4 
00f0: de e5 82 ab 6d 95 b9 d8 30 11 9a 1f c0 39 b5 dc 
0100: 52 ac 71 ab b3 9a 29 8a 94 6c 69 4e 14 6b a8 0b 
0110: 4c 4e d5 46 f3 11 15 73 0b be 4a b4 05 60 3a 39 
0120: 24 c3 21 6b 2f 8c 4c 15 0a 25 13 3b 99 07 f8 71 
0130: a5 01 84 a8 14 80 a8 6c 15 be 33 cc cb a2 8b 02 
0140: f8 5a 8c dd 3a bd 8b 42 18 4c 60 f3 13 29 d7 a1 
0150: da 5b b3 ae 92 e2 f2 41 89 79 e3 46 1c c8 8f a4 
0160: d4 e7 ae 39 4c 49 8f 62 1c c8 ea 44 45 4a df 9d 
0170: c2 59 44 74 1a 43 39 e0 2d f8 3b b3 60 66 fd 3b 
0180: 4b 18 62 88 2c a2 96 96 45 bc b5 ad 82 fb f0 4b 
0190: da 6b ed 6f 3d 38 2f 94 28 e0 69 ea 9e 9b 98 3a 
01a0: 90 a2 a6 d8 66 8a 65 71 db da a3 e5 4a 2f 79 91 
01b0: d4 fa 6e f9 5a bd af 26 b6 c6 65 75 6b 9b 47 62 
01c0: 67 9f 98 bc f0 01 3d fb 2c 9b 54 0e f0 5b 78 dd 
01d0: ea 0e 51 a2 27 fd 70 75 56 83 87 1e 34 42 0f f5 
01e0: 7e 14 45 a9 07 c5 29 68 cb 45 6f 9f ff 30 c2 57 
01f0: 43 38 26 7e 1c 43 cc 8a 03 82 de f3 07 4e a8 87 
0200: 3b 2a aa ff ff ff 0f bb
```

Maybe if time allows it, you can have a look.

Observation used: https://network.satnogs.org/observations/4351767/

daniestevez, 2021-07-01:
Hi Jan,

These don't seem good decodes. The errorcode `aa ff ff ff 0f bb` indicates that most of the FEC blocks couldn't be decoded. Probably the framing is not correct. gr-satellites supports the `Mobitex` and `Mobitex-NX` variants that are implemented in gr-tnc_nx. It might happen that they've changed something with TUBIN, so I'll try to ask some folks and take a look at the documentation.

janvgils, 2021-07-01:
Thanks Daniel.

daniestevez, 2021-07-03:
Hi Jan, I've been taking a more careful look at this, and I've found out that the quality of most of the SatNOGS recordings of TUBIN is not so good. Also, using the `--disable_dc_block` parameter with `gr_satellites` seems to help.

I found [this observation](https://network.satnogs.org/observations/4357219/), which is a bit better than others, and produces a valid decode by running
```
gr_satellites TECHNOSAT --wavfile satnogs_4351795_2021-07-01T14-25-30.wav --disable_dc_block
```

The decode looks like this:
```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 1)
pdu_length = 592
contents = 
0000: 3f 0e 44 50 30 54 42 4e 15 c0 05 11 01 0f 00 1c 
0010: 03 28 70 8e 5a 05 15 c4 0c 00 00 00 54 00 5f 28 
0020: 70 8d cd 05 11 11 03 00 00 00 02 ff f9 28 70 7b 
0030: 8b 1e 00 e5 de 00 05 03 28 70 8e 5a 02 28 44 00 
0040: 6e 59 5c c2 60 00 11 00 28 70 8e 5a 06 3c 44 17 
0050: 00 0a 3c e1 31 f6 a4 13 99 fa fe 00 29 02 36 f7 
0060: d8 00 41 00 28 70 8e 5a 06 83 00 03 ff a0 ff f3 
0070: f6 e8 ff fc 57 c1 00 07 a7 ea fa aa fe 32 ff 8d 
0080: ff fe ff fd 00 01 00 80 02 b8 fe 76 00 b0 fc 46 
0090: 00 ee 02 0e fe 59 fe 79 ff 54 03 87 01 87 f3 40 
00a0: 0a 31 15 77 ff fd 00 04 ff fc 00 58 f2 00 03 03 
00b0: 28 70 8e 5a 0a f9 00 00 03 06 d3 00 0e 00 28 70 
00c0: 8e 59 1e 5a 44 01 7f ff 7f ff 7f ff ff ff ff ff 
00d0: 00 00 fb 83 00 10 05 28 70 8e 5a 00 10 46 00 01 
00e0: 10 0d 00 00 0a 00 04 14 40 00 08 a6 a7 fe 53 00 
00f0: 41 15 28 67 c4 3f 06 83 00 07 2a ad 00 08 f2 e1 
0100: 00 06 39 c2 00 07 e9 f7 01 17 00 7d 04 d2 ff f2 
0110: 00 04 00 02 02 19 fe 85 fe 5f 03 bf 00 11 fe e8 
0120: fe 9d 02 52 fe 97 ff c3 03 93 01 8d fa e8 10 96 
0130: eb 54 00 03 ff fc ff fc 00 a2 3e 00 2a 12 28 67 
0140: c4 40 00 a9 00 41 00 20 00 24 4e 00 00 00 00 34 
0150: 00 00 0f 0e 08 09 0a 0c 1d 1a 3e 1c 18 00 00 00 
0160: 00 00 00 0a 16 18 14 10 0d 0e 10 13 3b e5 a8 96 
0170: 00 41 15 28 67 c4 7b 06 83 00 01 d9 95 ff f7 78 
0180: 8b ff ff 10 37 ff f3 84 c5 00 c6 00 a7 04 74 00 
0190: 05 ff fe 00 03 ff 11 fd 22 ff 39 00 37 fc 1f 00 
01a0: 63 fe aa 02 42 fe 39 ff c3 03 93 01 8d fb a7 0f 
01b0: 62 ea 41 00 03 ff fb ff fc 00 57 fe 00 41 15 28 
01c0: 67 c4 b7 06 83 00 06 b1 9f ff f8 b8 10 00 02 19 
01d0: d9 ff f4 92 78 00 52 00 da 04 87 00 00 00 01 00 
01e0: 05 fd 4f ff be fe 18 fe 89 fc cd 01 b1 fe bc 02 
01f0: 25 fd de ff c3 03 93 01 8d fc 6c 0e 1c e9 46 00 
0200: 03 ff fb ff fd 00 ab c8 00 2a 12 28 67 c4 b8 00 
0210: a9 00 41 00 1f 00 22 00 00 26 0c 00 00 00 00 0f 
0220: 09 07 09 0c 0f 22 1e 3e 9b 00 00 00 10 21 13 00 
0230: 00 17 15 10 0d 0c 0f 13 15 3d b6 99 10 ff ff ff 
0240: ff ff ff ff ff ff ff ff ff ff aa 00 00 00 00 bb 
***********************************
```

Note that between the `aa` and `bb` at the end of the frame we have zeros. This indicates that all the FEC codewords were successfully decoded. Also the callsign `DP0TBN` appears somewhere in the beginning of the frame.

The decoder also produces some empty frames marked with the flag `5` by the `gr-tnc_nx` decoder (I don't remember what this means):
```
* MESSAGE DEBUG PRINT PDU VERBOSE *
((transmitter . 4k8 FSK downlink) . 5)
pdu_length = 0
contents = 
***********************************
```

So apparently TUBIN uses exactly the same protocol as TECHNOSAT, which gr-satellites calls `Mobitex-NX`. This matches the online documentation and the reports of people on Twitter using UZ7HO's soundmodem for TECHNOSAT to decode TUBIN.

I think this solves the matter (besides #238). I'll now add a SatYAML file for TUBIN. For the time being I'll only include the 4k8 mode unless we have an indication that other baudrates are used.

janvgils, 2021-07-03:
Thanks for taking the time to investigate this, I will use the suggestion and see if I can get better decodes.

closed 2021-07-03 by janvgils