daniestevez/gr-satellites #285: ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"

Wint3rmute, 2021-07-05:
Operating system: `fedora 34`
Python version: `3.9.5`
GNURadio version: `3.9.0.0` (from `gnuradio-companion -h`)
GR-Satellites branch: `maint-3.9`

#

Hello!

I'm, trying to set up `gr-satellites` on my PC, but I'm running into `ImportError`s when importing the `satellites` python module. I get the same error in GnuRadio and in the Python interpreter:

```
Traceback (most recent call last):
  File "/usr/lib64/python3.9/site-packages/satellites/__init__.py", line 42, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib64/python3.9/site-packages/satellites/__init__.py", line 44, in <module>
    from .satellites_python import *
ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"
```

Here's a minimal `Dockerfile` that allows to reproduce this error:

```dockerfile
FROM fedora:34

RUN dnf install git g++ cmake gnuradio-devel -y
RUN dnf install log4cpp-devel gmp-devel pybind11-devel libunwind-devel bzip2 python3-devel -y

RUN git clone https://github.com/daniestevez/gr-satellites.git  --single-branch --branch maint-3.9 --depth 1

RUN mkdir /gr-satellites/build
WORKDIR /gr-satellites/build

RUN cmake .. -DCMAKE_INSTALL_PREFIX=/usr
RUN make
RUN make install

RUN ldconfig
```  

Running instructions, in case you're unfamiliar with Docker:
```bash
# Save the snippet from above to a file called `Dockerfile` that's in your working directory first!

docker build . -t gr_satellites_ax100_issue
docker run -it minimal_gr python -c 'import satellites'
```

Please let me know if there's any way to fix this. Thanks in advance!

daniestevez, 2021-07-05:
Hi @Wint3rmute, thanks for the very detailed report! I don't have a Docker at hand to test this at the moment (but could easily do it if needed). Is this perhaps a [problem with the PYTHONPATH](https://gr-satellites.readthedocs.io/en/latest/installation.html#pythonpath), so that using `ENV` to set the `PYTHONPATH` to wherever gr-satellites gets installed in Fedora?

Wint3rmute, 2021-07-09:
Hello @daniestevez, thanks for a quick response!

I have verified where the `satellites` module is installed on Fedora 34. Unfortunately, adding it to `PYTHONPATH` does not solve the problem:

```bash
$ docker run -it gr_satellites_ax100_issue bash -c "PYTHONPATH=/usr/lib64/python3.9/site-packages python -c 'import satellites'"

Traceback (most recent call last):
  File "/usr/lib64/python3.9/site-packages/satellites/__init__.py", line 42, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/lib64/python3.9/site-packages/satellites/__init__.py", line 44, in <module>
    from .satellites_python import *
ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"

```

daniestevez, 2021-07-09:
I'm testing your Dockerfile, since I think that this should work. However, there seems to be something wrong regarding the way that GNU Radio 3.9 OOT modules are built/installed.

I've tried building gr-symbolmapping (another OOT that came to mind) in a container built with your Dockerfile and I get a similar error:
```
[root@984b34c4fda4 ~]# LD_LIBRARY_PATH=/usr/local/lib64 python
Python 3.9.6 (default, Jun 29 2021, 00:00:00) 
[GCC 11.1.1 20210531 (Red Hat 11.1.1-3)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import symbolmapping
Traceback (most recent call last):
  File "/usr/local/lib64/python3.9/site-packages/symbolmapping/__init__.py", line 18, in <module>
    from .symbolmapping_python import *
ImportError: generic_type: type "interleaver_bb" referenced unknown base type "gr::sync_block"

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib64/python3.9/site-packages/symbolmapping/__init__.py", line 22, in <module>
    from .symbolmapping_python import *
ImportError: generic_type: type "Interleaver" is already registered!
```

In fact I was wrong that the problem was the PYTHONPATH, since the Python module gets found and the problem only happens when importing the pybind11 bindings.

janvgils, 2021-11-05:
This evening I ran into the same problem on a Debian testing distro where I use the GNURadio 3.9.2 packages from experimental.

```
gr_satellites --version
Traceback (most recent call last):
  File "/usr/local/lib/python3/dist-packages/satellites/__init__.py", line 42, in <module>
    from .bindings.satellites_python import *
ModuleNotFoundError: No module named 'satellites.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 16, in <module>
    import satellites.core
  File "/usr/local/lib/python3/dist-packages/satellites/__init__.py", line 44, in <module>
    from .satellites_python import *
ImportError: generic_type: type "ax100_decode" referenced unknown base type "gr::block"
```

If there are some hints or tips to debug this further, please let me know.

daniestevez, 2021-11-06:
Now this strikes me as the same problem as #303. There, the cause was that the GNU Radio package in the PPA had been built against a version of pybind11-dev that was different from what was installed in the system.

Could this also be what happens in your case @janvgils? If you're pulling GNU Radio from experimental, maybe you should pull pybind11-dev from there too.

I still don't know what to say about @Wint3rmute's original problem, since in that case he was building GNU Radio from source, so the version incompatibility of pybind11 is not possible, I think.

janvgils, 2021-11-06:
Good Morning Daniel,

Thanks for the reply and pointing me to this [#303](https://github.com/daniestevez/gr-satellites/issues/303) problem.
Looking at my Debian testing (bookword/sid) distro I have the options install pybind11-dev 2.6.2-1 or pybind11-dev 2.7.1-1.

The current version installed is `pybind11-dev 2.7.1-1` and I downgraded to `apt-get install pybind11-dev=2.6.2-1` then did a new build of gr-satellites, this in my case didn't solve the problem.

Lets see if I can find some more detailed information on how gnuradio 3.9.2.0 is build in the Debian experimental repository.

To be continued, have a great weekend.

daniestevez, 2021-11-06:
Another option would be to build GNU Radio from source and check if that works. Even if you want to end up using the .deb package, it might be interesting to know whether building from source works or not.

janvgils, 2021-11-06:
Currently investigation the debian build [logs](https://buildd.debian.org/status/fetch.php?pkg=gnuradio&arch=amd64&ver=3.9.2.0-2&stamp=1624125250&raw=0) and there I see that indeed ` pybind11-dev all 2.6.2-1` was used during the build process. I will have another try by downgrading to this version.

janvgils, 2021-11-06:
Well it took a lot of time, but the problem is solved.

Rebuilding the GNURadio 3.9 debian source with pybind11-dev 2.7.1-1 installed.

```
# add to /etc/apt/sources.list

# Debian Experimental
# -----------------------------------------------------------------------------
#
deb http://deb.debian.org/debian experimental main contrib non-free
deb-src http://deb.debian.org/debian experimental main contrib non-free

# update package list information
sudo apt-get update

# create and use a build directory 
mkdir -p ${HOME}/source/src/debian/experimental/gnuradio
cd ${HOME}/source/src/debian/experimental/gnuradio

# get source and build package from source
apt source gnuradio/experimental

cd gnuradio-3.9.2.0

dpkg-buildpackage -uc -us -nc -j4

cd ..

for i in *.deb; do sudo dpkg -i $i; done

cd ${HOME}/source/git/daniestevez/gr-satellites/build/

cmake -Wno-dev ../ && make -j4 && sudo make install && sudo ldconfig && gr_satellites --version

gr_satellites v4.5.0-git
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

```

daniestevez, 2023-01-29:
I'm closing this issue, since several problems were treated here and the issue has been inactive for long. The original problem about running on Fedora 34 probably persists, but that Fedora release was end of life half a year ago. People finding similar problems can open an new issue.

closed 2023-01-29 by daniestevez