daniestevez/gr-satellites #288: Error Running gnradio.

ik8ozv, 2021-07-14:
Please apologise if i ma posting in the wrong thread.
As my Ubuntu has done update every time i run a grc file in the gnuradio i get:

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/enzo/GRC/Sat_TW-1A_IK8OZV.grc"
>>> Done

Generating: '/home/enzo/GRC/TW_1A.py'

Executing: /usr/bin/python3 -u /home/enzo/GRC/TW_1A.py

Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff

I remeber before the update that i got only:

Block paths:
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/enzo/GRC/Sat_TW-1A_IK8OZV.grc"
>>> Done

Generating: '/home/enzo/GRC/TW_1A.py'

Executing: /usr/bin/python3 -u /home/enzo/GRC/TW_1A.py

and not even the laste three files.
This happens on whichever grc file i run.
Please is it normal?
If no how top correct it?
Thanks
73's de Enzo IK8OZV

daniestevez, 2021-07-14:
Hi Enzo,

These lines are harmless:
```
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff
```
The DEBUG lines can be disabled as described [here](https://wiki.gnuradio.org/index.php/Logging).

ik8ozv, 2021-07-15:
Dani,
thanks for the reply but I did not understood if I need to give a command in terminal or I have to edit a file.
Will you please be so kind to tell me exactly what to do?

ik8ozv, 2021-07-18:
> Hi Enzo,
>
> These lines are harmless:
>
> ```
> Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
> gr::log :DEBUG: correlate_access_code_tag_bb0 - [...]

Dani Apologise you've said that These lines are harmless but i am noticing that i cannot decode anymore.
Please any help?
Thanks

daniestevez, 2021-07-19:
Are you able to obtain decodes with the examples in [test.sh](https://github.com/daniestevez/gr-satellites/blob/master/test.sh)? Can you give more information as to what version of gr-satellites you're using, etc?

ik8ozv, 2021-07-19:
Dani,
i have tried to give the command:

gr_satellites 1KUNS-PF --wavfile satellite-recordings/1kuns_pf.wav

while i am in the gr-satellites and i got:

-> Packet from 1k2 FSK downlink
Container: 
    csp_header = Container: 
        priority = 2
        source = 1
        destination = 9
        destination_port = 10
        source_port = 37
        reserved = 0
        fragmentation = False
        hmac = False
        xtea = False
        rdp = False
        crc = False
    beacon_counter = 4274
    solar_panel_voltage = ListContainer: 
        2448.0
        2448.0
        2432.0
    eps_temp = ListContainer: 
        1.0
        3.0
        2.0
        2.0
    eps_boot_cause = 7
    eps_batt_mode = 3
    solar_panel_current = 0.0
    system_input_current = 80.0
    battery_voltage = 8262.0
    radio_PA_temp = 4.0
    tx_count = 45584
    rx_count = 0
    obc_temp = ListContainer: 
        1.0
        1.0
    ang_velocity_mag = 10
    magnetometer = ListContainer: 
        288.0
        0.0
        0.0
    main_axis_of_rot = 89
enzo@ubuntu:~/gr-satellites$ 

so it seems i got the decodes .

I am not sure if the correct command to see the gr-satellites version is:

python3 -c "import satellites; print(satellites.__version__)"

but giving this command i get:

v3.9.0-git

Thanks

daniestevez, 2021-07-19:
So it seems that gr-satellites is working correctly for you.

ik8ozv, 2021-07-19:
Ok Thanks. Please can you tell me which is the command to disable The DEBUG lines ?
Or i need to edit a file?
Thanks

daniestevez, 2021-07-19:
As I said:
> The DEBUG lines can be disabled as described [here](https://wiki.gnuradio.org/index.php/Logging).

ik8ozv, 2021-07-19:
Thanks but ad i said before_

Please Dani,
thanks for the reply but I did not understood if I need to give a command in terminal or I have to edit a file.
Will you please be so kind to tell me exactly what to do?
Thanks

ik8ozv, 2021-07-21:
Dani,
please apologize if i insist but i cannot understand which parameter to change in the gnuradio-runtime.conf.
I have tried to change the parameter debug_level under the Log section changing with all the sets:

DEBUG, INFO, WARN, TRACE, ERROR, ALERT, CRIT, FATAL, EMERG, changing them one at time saving the file and trying running the grc file but i still get the error:

Warning: failed to XInitThreads()
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

Please where am i missing?
Thanks

daniestevez, 2021-07-21:
The
> gr::log :DEBUG: correlate_access_code_tag_bb0 - Access code: 930b51de
gr::log :DEBUG: correlate_access_code_tag_bb0 - Mask: ffffffff

messages can be disabled by editing `~/.gnuradio/config.conf`. However, the other warnings
>Warning: failed to XInitThreads()
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.

Do not come directly from GNU Radio, so they settings of `config.conf` won't affect these messages. I don't know exactly what library is producing these messages (the second seems to come from QT), but as I said they seem to be harmless, so my suggestion is that you ignore them.

ik8ozv, 2021-07-22:
Dani i have tried to go in the folder you mentioned but in that folder there is no config.conf file.
How to do please?
Thanks

daniestevez, 2021-10-15:
Hi, can this issue be closed?

daniestevez, 2021-10-31:
Closing. Feel free to re-open if you have more information.

closed 2021-10-31 by daniestevez