daniestevez/gr-satellites #290: Error while Receiving Data

ik8ozv, 2021-08-09:
Hello All,
please i am noticing that when i receive something, at the end of the data i can read:

Error while submitting telemetry: <urlopen error [Errno -3] Errore temporaneo nella risoluzione del nome>

Please how to avoid it?
Thanks
Enzo

daniestevez, 2021-08-09:
Hi Enzo,

That error means that the URL for the telemetry server could not be resolved by DNS. For what satellites does this happen? Depending on that, the error could refer to the SatNOGS server or to another specific server.

This may be just a combination of telemetry submission being enable in the config file (see [the docs](https://gr-satellites.readthedocs.io/en/latest/command_line.html#telemetry-submission)) and your computer not being connected to the internet.

daniestevez, 2021-10-15:
Hi, can this issue be closed?

daniestevez, 2021-10-31:
Closing. Feel free to re-open.

closed 2021-10-31 by daniestevez