daniestevez/gr-satellites #291: Setting fixedlen_tagger packet length while flowgraph is running

LucasBriggs154, 2021-08-09:
While using the "Sync and Create Packed PDU" block I was trying to implement setting the packet length while the flowgraph was actively running. I noticed that the "Packet length (bytes)" parameter was underlined (implying it is mutable at runtime), but it didn't seem to be working properly.

The problem lies with the Fixed Length Packet Tagger and its own packet length parameter, which is immutable at runtime. In order to enable this behavior I made some small changes to 3 files in my local gr-satellites build that I will lay out below. It amounts to creating a single callback setter for the packet_len in fixedlen_tagger.py.

I'm submitting this as an issue rather than a PR because I'm not well enough versed with gr-satellites (or gr-modules in general) to know if there's additional places that these changes need to propagate to.

Here are my changes as code blocks:
./python/fixedlen_tagger.py
```
class fixedlen_tagger(gr.basic_block):
   ...
   def set_packet_len(self, packet_len):
      self.packet_len = packet_len
   ...
```
./grc/satellites_fixedlen_tagger.block.yml
```
templates:
   ...
   callbacks:
   - set_packet_len(${packet_len})
   ...
```
./python/hier/sync_to_pdu_packed.py
```
class sync_to_pdu_packed(gr.hier_block2):
   ...
   def set_packlen(self, packlen):
      self.packlen = packlen
      self.satellites_fixedlen_tagger_0_0_0.set_packet_len(self.packlen)
```
^ I believe this last change would have been automatically added by GRC if the module was generated again from GRC with the new fixedlen_tagger callback. Should also work for the other hierarchical blocks that use the fixedlen_tagger.

I imagine the sync_to_pdu_packed.py hier block was generated with GRC, which defaults parameters to be mutable, so the fact this didn't work before makes sense. However, setting the packet length at runtime is quite useful, especially for my application, and I think that its worth including.

daniestevez, 2021-08-10:
Hi Lucas,

Thanks! I think that having the callback to set the packet length in the `fixedlen_tagger` would be a very good addition. I greatly encourage you to submit this as a PR.

I think that just the three files that you've changed are sufficient. Once thing I've spotted is that it will also be necessary to change the `maxlen` of [`self.stream`](https://github.com/daniestevez/gr-satellites/blob/master/python/fixedlen_tagger.py#L31), specially when going to a larger packet size.

Regarding the `sync_to_pdu_packed.py` block, the story of this and the other blocks in `python/hier` is that they come from .grc hierarchical flowgraphs. However, maintaining these is somewhat cumbersome, since the `.py` and `.yml` files need to be generated when gr-satellites is compiled, and installed correctly. Therefore, I ended up including the generated `.py` and `.yml` files. I've been meaning to rewrite the `.py` files to simplify and make them more like manually generated code (like the things in `python/components/`), but I haven't done so yet.

The reason why `packlen` has a callback in `sync_to_pdu_packed` is that every parameter in a GRC hierarchical flowgraph gets a callback automatically. The callback might not do anything if fields affected by the update don't have a callback (which is what happened here).

LucasBriggs154, 2021-08-17:
Hello!
Sorry its taken me so long to respond. Had a busy project week last week.

Would love to submit this as a separate branch and PR. I think I might need to be added as a collaborator? Pushing my local branch is giving me a permission-denied error. Not sure if there's another process for submitting a PR as a non-collaborator.

Thank you!
Lucas

daniestevez, 2021-08-17:
Hi Lucas,

You need to fork the repository, push your branch to your own repository, and then submit a PR from there. The steps are listed [here](https://jarv.is/notes/how-to-pull-request-fork-github/).

LucasBriggs154, 2021-08-17:
Aha that makes a lot of sense. This will actually be my first contribution to an open-source project, so thank you for the help! Submitting PR now.

closed 2021-08-17 by LucasBriggs154