daniestevez/gr-satellites #295: CSP header swap for AX100 transceiver

daniestevez, 2021-08-29:
Following some messaging with Mark Jessop, I think it's probably time to eliminate the [swap of the CSP header](https://github.com/daniestevez/gr-satellites/blob/717c7540e266957aba43f9665da3accc2fe5bad9/lib/ax100_decode_impl.cc#L80) that exists for historical reasons. There is a lot of background about why this is in place [in this comment in the librespace forums](https://community.libre.space/t/standardization-of-frames-formats-for-satnogs-db/7057/2?u=ea4gpz).

It would be good to adopt a consistent solution between all the applications that submit frames to SatNOGS DB, including gr-satellites, gr-satnogs (which does not swap the header), and the UZ7HO soundmodem (which I'm not sure what it currently does).

darksidelemm, 2021-08-30:
According to Patrick Dohmen, the UZ7HO modem doesn't do any header swapping.

daniestevez, 2021-08-30:
Just say Patrick's message. So apparently after UZ7HO dropped the swap earlier this year, gr-satellites is the last application that is doing the swap. I'll remove the swap now.

commit 07b01cda (2021-08-30): Remove swap in CSP header in AX100 and U482C deframers

The swap comes from historical reasons, as explained in #295.
Other applications such as gr-satnogs and UZ7HO soundmodem are
not performing this swap. This removes the swap so that the
frames sent to SatNOGS DB and other SiDS servers by gr-satellites
match those sent by other applications.

This closes #295

(cherry picked from commit ccc38e77961ebc3ca50d341b3b6b7d40734dc72c)
commit ccc38e77 (2021-09-11): Remove swap in CSP header in AX100 and U482C deframers

The swap comes from historical reasons, as explained in #295.
Other applications such as gr-satnogs and UZ7HO soundmodem are
not performing this swap. This removes the swap so that the
frames sent to SatNOGS DB and other SiDS servers by gr-satellites
match those sent by other applications.

This closes #295