daniestevez/gr-satellites #296: Missing attribute 'byte_t'

gvanem, 2021-09-02:
I think I've built Gnuradio 3.9 fine (on Windows-10).
But testing some `satellite-recordings/*.wav`files with `apps/gr_satelittes`, I'm not sure.
E.g. when running this command (under my build-root `f:\gv\dx-radio\gnuradio\gv-build`):
```
python gr-satellites\apps\gr_satellites 1KUNS-PF --wavfile gr-satellites\satellite-recordings\1kuns_pf.wav
```

gives me:
```
Traceback (most recent call last):
  File "gr_satellites", line 316, in <module>
    main()
  File "gr_satellites", line 301, in main
    tb = gr_satellites_top_block(parser)
  File "gr_satellites", line 192, in __init__
    **sat)
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\satellites\core\gr_satellites_flowgraph.py", line 162, in __init__
    self._init_demodulator_deframer(key, transmitter)
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\satellites\core\gr_satellites_flowgraph.py", line 274, in _init_demodulator_deframer
    options=self.options, **deframer_additional_options)
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\satellites\components\deframers\ax100_deframer.py", line 60, in __init__
    sync=_syncword, threshold=syncword_threshold)
  File "f:\gv\dx-radio\gnuradio\gv-build\lib\site-packages\satellites\hier\sync_to_pdu_packed.py", line 50, in __init__
    blocks.tagged_stream_to_pdu(blocks.byte_t, 'packet_len'))
AttributeError: module 'gnuradio.blocks' has no attribute 'byte_t'
```

Lots of `.wav` files gives similar errors. I fail to see how `byte_t` is part of `gnuradio.blocks`.
Am I missing some vital files in my build?

gvanem, 2021-09-02:
*Update:* Replacing `blocks.byte_t` in these files:
```
  python/hier/sync_to_pdu.py
  python/hier/sync_to_pdu_packed.py
```

with `gr.types.byte_t`, it seems to work better. But with some warnings:
```
gr::log 2021-09-02 11:59:16,663 :WARN: `tagged_stream_to_pdu` has moved to gr-pdu and will be removed from gr-blocks soo
n. Please update to use pdu.tagged_stream_to_pdu()
gr::log 2021-09-02 11:59:16,671 :WARN: `tagged_stream_to_pdu` has moved to gr-pdu and will be removed from gr-blocks soo
n. Please update to use pdu.tagged_stream_to_pdu()
gr::log :ERROR: pagesize - no info; setting pagesize = 4096
-> Packet from 1k2 FSK downlink
Container:
    csp_header = Container:
        priority = 2
        source = 1
        destination = 9
        destination_port = 10
        source_port = 37
        reserved = 0
        fragmentation = False
        hmac = False
        xtea = False
        rdp = False
        crc = False
    beacon_counter = 4274
    solar_panel_voltage = ListContainer:
        2448.0
        2448.0
        2432.0
    eps_temp = ListContainer:
        1.0
        3.0
        2.0
        2.0
    eps_boot_cause = 7
    eps_batt_mode = 3
    solar_panel_current = 0.0
    system_input_current = 80.0
    battery_voltage = 8262.0
    radio_PA_temp = 4.0
    tx_count = 45584
    rx_count = 0
    obc_temp = ListContainer:
        1.0
        1.0
    ang_velocity_mag = 10
    magnetometer = ListContainer:
        288.0
        0.0
        0.0
    main_axis_of_rot = 89
-> Packet from 1k2 FSK downlink
Container:
    csp_header = Container:
        priority = 2
        source = 1
        destination = 9
        destination_port = 10
        source_port = 37
        reserved = 0
        fragmentation = False
        hmac = False
        xtea = False
        rdp = False
        crc = False
    beacon_counter = 4275
    solar_panel_voltage = ListContainer:
        2256.0
        2256.0
        2240.0
    eps_temp = ListContainer:
        0.0
        3.0
        2.0
        2.0
    eps_boot_cause = 7
    eps_batt_mode = 4
    solar_panel_current = 0.0
    system_input_current = 80.0
    battery_voltage = 8296.0
    radio_PA_temp = 4.0
    tx_count = 45840
    rx_count = 0
    obc_temp = ListContainer:
        1.0
        1.0
    ang_velocity_mag = 10
    magnetometer = ListContainer:
        318.0
        0.0
        0.0
    main_axis_of_rot = 89
```

daniestevez, 2021-09-02:
Hi,

This works for me under GNU Radio 3.9.1.0. Both running
```
gr_satellites 1KUNS-PF --wavfile satellite-recordings/1kuns_pf.wav
```
and
```
$ python
Python 3.9.5 (default, May  9 2021, 00:55:09) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gnuradio.blocks
>>> gnuradio.blocks.byte_t
<vector_type.byte_t: 0>
```

gvanem, 2021-09-02:
> This works for me under GNU Radio 3.9.1.0. Both running

So I have a newer version than yours. From `CHANGELOG.md`:
```
## [3.10.0.0] - Unreleased

### Changed

- Moved PDU blocks from gr-blocks to gr-network and gr-pdu
   - Compatibility shim included to allow access to these blocks from gr-blocks
     but these are deprecated from the gr-blocks namespace and the shim is
     scheduled for removal in 3.11.
```
I assume this is the cause?
But I do no know what this `Compatibility shim` is. Doesn't seem to add any `gnuradio.blocks.byte_t` into the namespace for me:
```
python 
>>> import gnuradio.blocks
>>> gnuradio.blocks.byte_t
AttributeError: module 'gnuradio.blocks' has no attribute 'byte_t'
```

daniestevez, 2021-09-02:
GNU Radio 3.10 hasn't been released yet. What you've installed is probably the git master branch, which contains the development branch that will eventually become 3.10.

Typically, OOT modules like gr-satellites need to be modified to run with new releases of GNU Radio. So here you've spotted two changes that will need to be done in gr-satellites to make it compatible with 3.10. Since these changes typically are not backwards compatible, a new branch specific for GNU Radio 3.10 will be created.

For the time being, gr-satellites only supports GNU Radio 3.9 (and also 3.8, in a separate branch), so my suggestion is that you install 3.9 instead of the git master branch. If you're using Windows, you have GNU Radio 3.9 and gr-satellites already built in conda, so maybe that's an easy and quick way to install both.

gvanem, 2021-09-02:
So 3.7 broke 3.8. 3.8 broke 3.9 and now 3.10 breaks 3.9. When will this insanity end?

daniestevez, 2021-09-02:
My understanding is that the changes required when going from 3.9 to 3.10 are very minor.

daniestevez, 2021-10-15:
I've checked that it doesn't hurt to remove `test_satellites.cc` and have removed it in 2df75369b4a47cd32b67091809bba7c4c661ee2a

daniestevez, 2022-01-15:
I have taken care of this in #322

closed 2022-01-15 by daniestevez

gvanem, 2022-01-15:
> I have taken care of this in #322

Works fine now. Although still lots of these annoying:
```
message_debug :info: The `print_pdu` port is deprecated and will be removed; forwarding to `print`.
```

daniestevez, 2022-01-15:
> message_debug :info: The `print_pdu` port is deprecated and will be removed; forwarding to `print`.

This is known. See #323.