daniestevez/gr-satellites #297: Add support for the Colorado Ultraviolet Transit Experiment (CUTE)

nsdecicco, 2021-09-09:
# Introduction

Our team at the Laboratory for Atmospheric and Space Physics (LASP) at the University of Colorado Boulder (CU) recently completed the integration of the Colorado Ultraviolet Transit Experiment (CUTE) into a PSC dispenser on LandSat-9, which is scheduled for launch no sooner than September 23rd of 2021. We would like to add support to gr-satellites for CUTE telemetry.

# Mission overview

CUTE is a 6U CubeSat containing a Cassegrain telescope and near-ultraviolet (NUV, 250-330 nm) spectrograph intended for study of atmospheres of short period exoplanets. The spectrograph uses a Teledyne E2V CCD42-10 charged-coupled device (CCD) detector for imaging spectra.

# Spacecraft radios

CUTE contains two radios:

Band | 70cm | 13cm (S-band)
-----|------|-------
Mfg./Model | SpaceQuest TRX-U | BCT TX only
Frequency | 437.250 MHz | 2.402 GHz
Modulation | GFSK | BPSK
Baudrate | 9.6k or 19.2k | 500M
Framing | AX.25/G3RUH | CCSDS
Antenna | Low-gain λ/4 monopole | High-gain patch
Transmit power | 1.75W (32.43 dBm) | ~2W

## S-band

Details of the modulation, encoding, and packetization of data on the S-band link will be discussed in more detail in a separate issue report.

## 70cm (UHF)

### Framing

All data sent by the CUTE spacecraft over the 70cm (437.250 MHz) link are contained within AX.25 unnumbered information (UI) frames:

<table>
<tr><td></td><th colspan="5" style="text-align: center">Field</th></tr>
<tr><td></td><th>Source callsign</th><th>Destination callsign</th><th>Control Field</th><th>PID</th><th>Data</th></tr>
<tr><td>Size</td><td>7 bytes</td><td>7 bytes</td><td>1 byte</td><td>1 byte</td><td><i>variable</i></td></tr>
<tr><td>Value</td><td><code>'BCT&nbsp;&nbsp;&nbsp;0'</code></td><td><code>'CUTE&nbsp;&nbsp;p'</code></td><td><code>0x03</code></td><td><code>0xF0</code></td><td><i>variable</i></td></tr>
</table>

The control field is always set to `0x03`, which identifies the AX.25 frames as unnumbered frames. The PID field is always set to `0xF0`, which indicates that no layer 3 protocol is used.

### Source/destination callsigns

All AX.25 frames sent by the CUTE spacecraft contain the following source and destination addresses (callsigns):

<table>
<tr><th></th><th colspan="2">Source</td><th colspan="2">Destination</th></tr>
<tr><td>Byte</td><td>Char</td><td>Hex</td><td>Char</td><td>Hex</td></tr>
<tr><td>1</td><td><code>'C'</code></td><td><code>0x86</code></td><td><code>'B'</code></td><td><code>0x84</code></td></tr>
<tr><td>2</td><td><code>'U'</code></td><td><code>0xAA</code></td><td><code>'C'</code></td><td><code>0x86</code></td></tr>
<tr><td>3</td><td><code>'T'</code></td><td><code>0xA8</code></td><td><code>'T'</code></td><td><code>0xA8</code></td></tr>
<tr><td>4</td><td><code>'E'</code></td><td><code>0x8A</code></td><td><code>' '</code></td><td><code>0x40</code></td></tr>
<tr><td>5</td><td><code>' '</code></td><td><code>0x40</code></td><td><code>' '</code></td><td><code>0x40</code></td></tr>
<tr><td>6</td><td><code>' '</code></td><td><code>0x40</code></td><td><code>' '</code></td><td><code>0x40</code></td></tr>
<tr><td>7</td><td><code>'p'</code></td><td><code>0xE1</code></td><td><code>'0'</code></td><td><code>0x60</code></td></tr>
</table>

Note that the 'hex' column shows the numeric value corresponding to the ASCII character displayed in the 'char' column *before* it is bit shifted to obtain said character.

It is not clear why the last byte of the source callsign is the lowercase letter 'p', nor why the last byte of the destination callsign is the numeric character '0': these appear to be errors in the CUTE C&DH flight software provided by BCT. (Note that the AX.25 specification allows only uppercase alpha and numeric ASCII characters, so neither callsign is compliant.)

### Info segment

The info segment of every AX.25 frame transmitted by the CUTE spacecraft begins with a CCSDS header:

<table border="1" style="text-align: center">
<tr><th rowspan="2"></th><th rowspan="2">Byte</th><th colspan="8">Description</th></tr>
<tr><th>7</th><th>6</th><th>5</th><th>4</th><th>3</th><th>2</th><th>1</th><th>0</th></tr>
<tr><td rowspan="6"><span style="writing-mode: vertical-rl; transform: rotate(180deg)">CCSDS Primary Header</span></td><td>0</td><td colspan="3">CCSDS Version (always 0)</td><td>Packet Type (always 0)</td><td>Secondary Header Flag</td><td>Stored Data Flag</td><td colspan="2">APID MSBits</td></tr>
<tr><td>1</td><td colspan="8">APID LSByte</td></tr>
[... 58 lines cut ...]
# Progress thus far

I have succeeded in implementing and testing the former, and I have pushed my changes to a fork of this repository hosted here:

https://github.com/lasp/gr-satellites

I pushed the changes to the maint-3.8 branch.

I have attached a "minimal working example" showing use of both the satellite decoder and a combination of the FSK demodulator and AX.25 deframer blocks to decode a recorded CUTE state of health packet.

In order to run the MWE, you will need to edit the 'file' parameter of the File Source block to contain an absolute path to the location of the `cute_soh_beacon.bin` file that is included in the .zip archive. You may also need to change the path to the CUTE YAML file in the Satellite decoder block. (Of course, building and installing the forked version of gr-satellites is also a prerequisite to use the telemetry decoder.)

The MWE can be found here: [gr-satellites-cute-soh-mwe.zip](https://github.com/daniestevez/gr-satellites/files/7139920/gr-satellites-cute-soh-mwe.zip)

# To do

1. I assume that the desired path to getting this code into gr-satellites is to open a pull request. If there a different way to proceed would be preferred, please let me know.

2. I am not certain how to use the telemetry decoder framework as it currently exists in gr-satellites to decode packets that span multiple AX.25 frames. If the sequence count reset with each new FSW packet, it should be possible to use the construct library to check the sequence count and grouping flag to determine how to decode each packet. But because the sequence count does not reset, while the construct library can look at the grouping flags for the first and last frames of a FSW packet, it would get "lost" in the middle of a FSW packet where the grouping flag = `0`. It seems that either the telemetry decoder would need to retain state (i.e., it would need to remember the last packet that it saw), or a custom block may need to be written to string the multiple PDUs containing a single longer packet together before these get handed to the telemetry decoder.

3. As mentioned earlier in this issue report, details of the S-band protocol etc. will be provided in another issue report (or as a comment on this issue report if that would be preferred). (My intention was to establish a line of communication with the project as soon as possible rather than postpone communicating until details of the S-band link could be written out.)

4. In addition to details of the S-band link, I will also need to provide details of the formatting of science data and payload (instrument) telemetry. There is a not insignificant amount of telemetry to document and write decoders for, so this could similarly be the subject of an additional issue report (if desired).

5. A minor "bug": the sub-seconds field is currently ignored when computing the time stamp.

daniestevez, 2021-09-10:
Thank you for this very detailed documentation!

A couple minor things I spotted when reading it:
* Is the baudrate for the S-band downlink really 500M?
* I think (but I'm speaking from memory, and checking the documentation would be appropriate) that the last character of the AX.25 address is the SID. This is typically a numerical value used to identify different stations or services using the same callsign, and `0` is commonly used when there is just one station or service. So the `0` at the end of `BCT` looks alright to me. The `p` at the end of `CUTE` doesn't look alright.

I will take a look at your repository soon and comment here.

Regarding your question in item 2 of the To do, I see several possibilities:
1. Processing each of the AX.25 that conform an FSW packet independently. I do this for some satellites, and I like this approach because of its simplicity and because if you lose some of the frames in the packet you still get some data out. However, this approach gives problems with fields that straddle two frames, and I think that perhaps it will not be applicable to your FSW, because as you point out from looking at a single frame the parser won't be able to tell which frame inside the whole FSW packet it is, unless it can be ensured that by looking at the Space Packet sequence count modulo 7 we can tell which frame within the packet this is.
2. Implementing a fancy stateful parser that handles defragmentation as needed. Construct can still be used to parse each of the fragments, but defragmentation is done in a class that acts as the "top-level" telemetry parser. I do this in the [FUNcube parser](https://github.com/daniestevez/gr-satellites/blob/dfc4db64e345f41d11134194af50010ed5ccbf8c/python/telemetry/funcube.py#L511) to handle whole orbit data, which is fragmented between several consecutive frames. This has the advantage that the behaviour can be customized as desired (for instance, we can choose what to print when there are missing frames). It has the disadvantage that it requires implementing some code which is somewhat complex and not very reusable.
3. Implement a "[transport](https://gr-satellites.readthedocs.io/en/latest/components.html#transports)" (this is how the component in gr-satellites is called) that sits between the AX.25 deframer and the telemetry parser. Transports are specifically intended to handle passage from the data link layer to an upper protocol layer, including defragmentation. For instance, the existing KISS transport handles defragmentation of a packetized KISS stream. This still requires a somewhat complex implementation of the defragmentation code, but is more reusable (the output of the transport could be sent into a file, for instance). However, it doesn't give so much flexibility. For instance, if frames are missing the transport should most likely drop the whole packet. Implementation of a telemetry parser if using a transport should be straightforward, since the telemetry parser would be receiving complete FSW packets.

daniestevez, 2021-09-11:
Hi, I took a look at the maint-3.8 branch of lasp/gr-satellites and the changes look well organized and reasonable. A pull request when this code is ready would be the appropriate way to get this integrated. If possible, please test your code against the master branch of gr-satellites (which uses GNU Radio 3.9), and submit the PR against master. Since you're not doing C++ blocks then most likely your code will carry over to GNU Radio 3.9 without any changes.

nsdecicco, 2021-09-15:
Hi Daniel,

Thanks for your responses.

> Is the baudrate for the S-band downlink really 500M?

Yep. [LASP has a 4.2m S-band dish in a radome on its roof](https://lasp.colorado.edu/home/smallsats/capabilities/) that was built to support a previous LASP CubeSat mission, the Compact Spectral Irradiance Monitor (CSIM), which also has a 500M S-band link. There are several other upcoming LASP CubeSat missions with similarly high speed S-band transmitters. CSIM has had issues with extremely high PER, so a FEC (Reed-Solomon) has been implemented on newer LASP CubeSats that haven't launched yet (namely, the Compact Total Irradiance Monitor, CTIM). However, while CUTE is launching years after CSIM, a lack of engineering support (basically, personnel and time) prevented implementing a FEC on CUTE.

> I think (but I'm speaking from memory, and checking the documentation would be appropriate) that the last character of the AX.25 address is the SID. This is typically a numerical value used to [...]

Yes, now that you say that, I do recall reading about the SID in the AX.25 documentation--I think you are correct. I'm really not sure why the `p` is there, hence my suspicion that this is a bug. But, the gr-satellites AX.25 telemetry decoder seems to have no issue parsing this.

> Regarding your question in item 2 of the To do, I see several possibilities:
>
> 1. Processing each of the AX.25 that conform an FSW packet independently. I do this for some satellites, and I like this [...]

I would agree that I don't think that this would work, except for the first and last frames. I don't think that it will be reliable to look at the space packet sequence count mod 7, as the sequence count would become "shifted" if shorter packets (e.g., SOH) were sent between FSW packets.

> 2. Implementing a fancy stateful parser that handles defragmentation as needed. Construct can still be used to parse each of the fragments, but defragmentation is done in a class that acts as the [...]

Thanks for pointing out the FUNcube parser as an example of a stateful parser—I don't think I would have found this myself. I see how you've accomplished retention of state, and I didn't realize this was possible with the existing software framework prior. (I am not primarily a Python programmer, so the way you've exploited Python's loosely typed nature—i.e., that the Funcube class just needs to implement a `parse` method and doesn't need to derive from a `construct` class—was a non-obvious option to me, but is obvious in hindsight.)

> 3. Implement a "transport" (this is how the component in gr-satellites is called) that sits between the AX.25 deframer and the telemetry parser. Transports are specifically intended to handle [...]

I haven't had a chance to study the transport components yet. I ended up going with option 2 and implementing a stateful telemetry parser for FSW packets (which also works for SOH packets, and I think should work for any APID that spans multiple AX.25 frames). I pushed this to the `master` branch of `lasp/gr-satellites` as commit [`9be615cb`](https://github.com/lasp/gr-satellites/commit/9be615cb7959110b5cb379c3dc30d0d0e159dd1d).

My stateful parser works a bit differently from the FUNCube one, in that it either returns `None` (if either an invalid frame was encountered or if the last frame received did not complete a packet), or a `Container` with the re-assembled and parsed packet.

By the way, in working on this, it occured to me that a useful feature might be if the telemetry parser block were capable of not just writing the `Container` to `stdout` or a file, but if this could be written e.g., as a CSV file or NetCDF or some such. I could create a separate issue to track this idea if you think that's worth pursuing. (I am tempted to hack a NetCDF output into the telemetry parser block in particular.)

> [...] I took a look at the maint-3.8 branch of lasp/gr-satellites and the changes look well organized and reasonable. A pull request when this code is ready would be the appropriate way to get this [...]

I went ahead and cherry-picked the commits I made to `maint-3.8` on `lasp/gr-satellites` over to master (both the commit you looked at previously and the new stateful parser that I implemented), and tested these against GNU Radio 3.9, and my testing made it appear that the code worked just fine without modification.

Along these lines, I have uploaded a pair of much larger I&Q files for testing to Google Drive, here:

* [cute_fsw_beacons.bin](https://drive.google.com/file/d/1w8kMXS4oTv1h4W50WPeAMXeR7bkxImm8/view?usp=sharing) (135.8 MB)
* [cute_soh_beacons.bin](https://drive.google.com/file/d/1rxt8bYg_W4gtjP9NCuFGeD0q3ABIKUXI/view?usp=sharing) (192.7 MB)

Both of these files are floating point and recorded at 307.2 ksps.

If you would like, I could squash the two commits that I made ([`9be615cb`](https://github.com/lasp/gr-satellites/commit/9be615cb7959110b5cb379c3dc30d0d0e159dd1d) and [`47bd10dd`](https://github.com/lasp/gr-satellites/commit/47bd10dd5ec8be15e7668d5b4e2b7eb61b6d0de9)) into one before making a pull request.

Anyway, let me know if you have any feedback to give and before I go ahead and make the pull request.

Thanks,

Nick

daniestevez, 2021-09-16:
Regarding the S-band link, that's quite interesting. How much bandwidth do you use? I had the idea that there is not that much bandwidth available at S-band (maybe with an experimental licence?).

> as the sequence count would become "shifted" if shorter packets (e.g., SOH) were sent between FSW packets.

I think the sequence count is per APID, so since you use an APID for FSW packets only (and put SOH in another APID) this shouldn't happen. But still, I agree that relying on this doesn't seem so good of a solution.

> I pushed this to the master branch of lasp/gr-satellites as commit 9be615cb.

On a first glance this looks quite good. It gives you the best of two worlds by using construct for parsing each piece and using a custom state machine in a Python class.

> By the way, in working on this, it occured to me that a useful feature might be if the telemetry parser block were capable of not just writing the Container to stdout or a file, but if this could be [...]

That seems a great idea. I wouldn't do this probably in the telemetry parser block itself, but just create a different block that does the output to file in the required format. This could be hooked up via an optional command line parameter as we already have `--kiss_out` and others, and also with its own block in GNU Radio Companion.

I'm not too savvy about NetCDF, but I'm thinking that the extremely flexible format of `Container`s could give you problems to convert this data into NetCDF. I'm thinking that this "telemetry file output" block could get `Containers` whose data is almost anything one can imagine, depending on the satellite. Perhaps a more flexible format such as JSON could be easier to handle this. But I guess that it depends a lot on what you want to do with the data afterwards. For many things NetCDF is better than JSON, because it is much more compact.

> If you would like, I could squash the two commits that I made (9be615cb and 47bd10dd) into one before making a pull request.

I think you can keep these separate if you want, for the first commit will work correctly without the second, so they are somehow independent.

> Anyway, let me know if you have any feedback to give and before I go ahead and make the pull request.

From my side you can go ahead and submit the pull request. I haven't yet reviewed the code in much detail, but I think I'll do that over the pull request.

nsdecicco, 2021-09-16:
> Regarding the S-band link, that's quite interesting. How much bandwidth do you use? I had the idea that there is not that much bandwidth available at S-band (maybe with an experimental licence?).

I don't honestly know the details of the license for CUTE's S-band link (though I do think it is an experimental license), but [the bandwidth between the first nulls is approximately 1 MHz](https://user-images.githubusercontent.com/18425292/133689153-cdf38a27-719d-46ef-8a58-d19f98e7e3c2.png). Between the next two sets of nulls [is about 2 MHz](https://user-images.githubusercontent.com/18425292/133689159-979be61d-bce5-4cec-8ba1-0514234929b2.png). Going by power, [99.0% of the transmitted power is contained within about 1.450 MHz](https://user-images.githubusercontent.com/18425292/133689182-90e0a9e1-0bf4-42c8-90a6-8894f2ed59e8.png). For reference, [here is the license for CSIM](https://apps.fcc.gov/els/GetAtt.html?id=212803&x=.), which does not specify bandwidth.

> I think the sequence count is per APID, so since you use an APID for FSW packets only (and put SOH in another APID) this shouldn't happen. But still, I agree that relying on this doesn't seem so [...]

After taking another look at the test data I shared in my previous comment ([cute_fsw_beacons.bin](https://drive.google.com/file/d/1w8kMXS4oTv1h4W50WPeAMXeR7bkxImm8/view?usp=sharing) and [cute_soh_beacons.bin](https://drive.google.com/file/d/1rxt8bYg_W4gtjP9NCuFGeD0q3ABIKUXI/view?usp=sharing)), it appears that BCT has not implemented the sequence count in this way: running these files through the flow graph, the state of health beacons cover sequence counts 915 through 926, and the flight software beacons started with sequence count 927. I am not entirely certain, but I believe we recorded these two files one right after the other, where the two files were separated by changing the beacon from SOH to FSW packets.

>> I pushed this to the master branch of lasp/gr-satellites as commit 9be615c.
>
> On a first glance this looks quite good. It gives you the best of two worlds by using construct for parsing each piece and using a custom state machine in a Python class.

Cool, glad to hear that you approve.

> That seems a great idea. I wouldn't do this probably in the telemetry parser block itself, but just create a different block that does the output to file in the required format. This could be hooked [...]

That's a good point about needing to know how to map containers to NetCDF. I think I will go ahead and make a separate issue to brainstorm this in.

>> If you would like, I could squash the two commits that I made (9be615c and 47bd10d) into one before making a pull request.
>
> I think you can keep these separate if you want, for the first commit will work correctly without the second, so they are somehow independent.
>
>> Anyway, let me know if you have any feedback to give and before I go ahead and make the pull request.
> 
> From my side you can go ahead and submit the pull request. I haven't yet reviewed the code in much detail, but I think I'll do that over the pull request.

Sounds good--I went ahead and submitted a pull request with the two commits separate. One of the tests did fail because the NORAD ID is missing.

Thanks,

Nick

daniestevez, 2021-09-17:
I think something is amiss regarding the S-band downlink. Looking at the spectrum plots you shared, it seems as 500 kbaud (rather than 500 Mbaud) BPSK with square pulse filtering. That seems much more reasonable for S-band.

> One of the tests did fail because the NORAD ID is missing.

You should put a NORAD ID even though there won't be one from 18th SPCS until a few days after launch. Some parts of the code expect the NORAD ID to be there and will give fatal exceptions, probably. You can put a placeholder value such as 99999, or even better, talk to the SatNOGS people on the [Librespace forums](https://community.libre.space/) to get the satellite added to [SatNOGS DB](http://db.satnogs.org/) and a preliminary NORAD ID that you can put in the YAML file. This will allow people submitting telemetry to SatNOGS DB even before the final NORAD ID is assigned.

Thanks for the pull request. I'll take a closer look at it over the weekend.

nsdecicco, 2021-09-17:
> I think something is amiss regarding the S-band downlink. Looking at the spectrum plots you shared, it seems as 500 kbaud (rather than 500 Mbaud) BPSK with square pulse filtering. That seems much more reasonable for S-band.

Whoops, you are correct. Not sure how I made that typo twice in a row. Yes, it is 500 kbaud, not Mbaud. (I think I had Mbaud on the brain because we were originally supposed to have 1 MBaud, so I was thinking 500 kbaud = 0.500 MBaud but writing 500 MBaud, but I digress...)

> One of the tests did fail because the NORAD ID is missing.
>
> You should put a NORAD ID even though there won't be one from 18th SPCS until a few days after launch. Some parts of the code expect the [...]

Good to know--I will reach out to SatNOGs to get a preliminary ID.

Thanks again,

Nick

DL4PD, 2021-09-28:
> It is not clear why the last byte of the source callsign is the lowercase letter 'p', nor why the last byte of the destination callsign is the numeric character '0': these appear to be errors in the [...]

You need to mask and shift the SSID field. See section 3.12 of the AX.25 specs.

>3.12. Address-Field Encoding
The address field of all frames consists of a destination, source and (optionally) two Layer 2 repeater subfields. Each subfield consists of an amateur callsign and a Secondary Station Identifier (SSID). The callsign is made up of upper-case alpha and numeric ASCII characters only. The SSID is a four-bit integer that uniquely identifies multiple stations using the same amateur callsign.
The HDLC address field is extended beyond one octet by assigning the least-significant bit of each octet to be an “extension bit.” The extension bit of each octet is set to “0” to indicate the next octet contains more address information, or to “1”, to indicate that this is the last octet of the HDLC address field. To make room for this extension bit, the amateur radio call- sign information is shifted one bit left.

`0xE1` sets the extension bit to 1. Following the above description this marks the end of the address field(s). When masking the lower 4 bits and shifting one position to the right, `0xE1` turns into `0`.

An example how I am solving this in `satnogs-decoders`is shown below:

```YAML
  ssid_mask:
    seq:
      - id: ssid_mask
        type: u1
    instances:
      ssid:
        value: (ssid_mask & 0x0f) >> 1
```