daniestevez/gr-satellites #315: YUSAT-1 decoder error

N6RFM, 2021-10-31:
Hi Dani,

Using GNU Radio, and gr-satellites master branch.  Trying the YUSAT-1 satyaml decoder.

See the following error -
gr::log :ERROR: thread_body_wrapper - ERROR thread[thread-per-block[24]: <block crop_and_check_crc(18)>]: NameError: name 'packet' is not defined
At:
  /usr/local/lib/python3.8/dist-packages/satellites/components/deframers/yusat_deframer.py(46): handle_msg

Any suggestions?
Thanks,
Bob

daniestevez, 2021-10-31:
Hi! You're right. From taking a look at the code, I see that this was a bug in porting the block from GNU Radio 3.8 to 3.9 and that I never tested it on GNU Radio 3.9. I have just pushed a commit to master that should fix the problem.

N6RFM, 2021-10-31:
Thanks Dani.  Unfortunately, now a different error.
gr::log :ERROR: thread_body_wrapper - ERROR thread[thread-per-block[24]: <block crop_and_check_crc(18)>]: TypeError: 'pmt.pmt_python.pmt_base' object is not subscriptable

At:
  /usr/local/lib/python3.8/dist-packages/satellites/components/deframers/yusat_deframer.py(47): handle_msg

daniestevez, 2021-10-31:
My bad. I make a mistake in the previous commit. I have pushed a new commit that tries to fix this. I think that now it will work correctly.

N6RFM, 2021-10-31:
Thanks Dani.  All set.  Appreciate the great support!!

daniestevez, 2021-10-31:
Great! Can this issue be closed?

N6RFM, 2021-11-01:
Yes. Thanks again Dani.

closed 2021-11-01 by daniestevez