daniestevez/gr-satellites #327: NRZI Decoder needs history

dcajacob, 2022-01-26:
I think the NRZI Decoder block needs history to be enabled since it looks back one bit. I believe the current implementation assumes the first previous bit is zero. I'm guessing the tests pass because the test data isn't long enough to break a work call? Compare the implementation to the core Differential Decoder block when modulus=2. They are equivalent (though maybe inverted). That implementation uses history, I think, correctly.

Apologies if I'm missing something obvious. My python is much stronger than my C++.

daniestevez, 2022-01-26:
Thanks for the comment!

There's the [`d_last` variable](https://github.com/daniestevez/gr-satellites/blob/master/lib/nrzi_decode_impl.cc#L51), which plays the same role as history. Probably it would be better to use history instead. In fact, now that I look at the code, `d_last` is assigned in each iteration of the `for` loop, while it only needs to be assigned in the last iteration.

By the way, starting with GNU Radio 3.10 the Differential Decoder block has an option to do NRZI decoding, so depending on what version of GNU Radio you're using, you might want to use this instead. The only reason the NRZI decoder block is still in gr-satellites is for compatibility with older versions of GNU Radio.

dcajacob, 2022-01-26:
Thanks! I did see you had an implementation for NRZI in the code, but not broken out in GRC yet. I discovered this because I was implementing NRZ-M, which doesn't seem to be covered.

daniestevez, 2022-01-26:
I think [it's already broken out in the GRC](https://github.com/gnuradio/gnuradio/blob/main/gr-digital/grc/digital_diff_decoder_bb.block.yml#L11). I'm guessing NRZ-M decoding could be obtained by NRZ-I decoding followed by an inverter. But having the calculation inside a single block would be more efficient.

dcajacob, 2022-01-26:
I thought it would be an inverter too, but it's not. NRZM is transition on one, NRZS is transition on zero. NRZI is NRZS, inverted, I think.

dcajacob, 2022-01-26:
Also, I am using the apt packages on Ubuntu at the moment, so the NRZI option probably just isn't broken out on that.

daniestevez, 2022-01-26:
Hmmm. I always get confused between all the NRZ variants. My understanding is that (speaking from the encoder perspective) NRZ-I is transition on zero (that's the whole reason why runs of 1's are limited in HDLC). If you invert the output of an NRZ-I encoder, you get the same kind of code (because it's a differential encoding).

If NRZ-M is transition on one, then for the encoder flipping zeros and ones at the input should suffice. So I think
invert + NRZ-I encoder = NRZ-S encoder
So it must be that
NRZ-I decoder + invert = NRZ-S decoder.

dcajacob, 2022-01-26:
Here's a table from Sklar. Frankly, I get confused too and worse, most of my understanding had been from explanations from other folks early in my career. I think NRZ-I can mean different things to different people, but I agree with your description of it. I have always used it in conjunction with HDLC. I have hear NRZ-M and NRZ-S referred to, but recently needed to use NRZ-M and my assumption that it was the same as NRZ-I was tested.

[image]

daniestevez, 2022-01-26:
Very helpful table! That made me put my thinking cap on.

Isn't NRZ-M exactly the same as the "plain old differential encoding" (i.e., what the Differential Encoder/Decoder with modulus 2 do)?

I was speaking all the time about NRZI-I plus an inverter, but that's just differential encoding/decoding. In fact, looking at the decoder formulas, it's clear that the only difference between them is an inversion:
```
if (d_coding == DIFF_NRZI) {
    for (int i = 0; i < noutput_items; i++) {
        out[i] = ~(in[i] ^ in[i - 1]) & 1;
    }
} else if (modulus == 2) {
    // optimized implementation for modulus 2
    for (int i = 0; i < noutput_items; i++) {
        out[i] = (in[i] ^ in[i - 1]) & 1;
    }
}
```

Doesn't the differential decoder with modulus 2 work in your use case? I've also heard people use all these NRZ-* terms somewhat differently, which adds to the confusion.

dcajacob, 2022-01-26:
Yes. In my testing, I also discovered that NRZ-M is equivalent to what the Differential Encoder(mod=2) produces. And looking at my notes more clearly, NRZ-I is equivalent to NRZ-S. I made a quick and dirty "Line Encoder" / "Line Decoder" set of blocks so I could use the NRZ-M / NRZ-I / NRZ-S terms without going nuts while trying to resolve other unknowns.

daniestevez, 2022-02-24:
I have just added a commit that changes the NRZI decoder to use history, since that seems the most logical way of doing things, even though the previous implementation did work correctly.

I guess this issue can be closed now? (I was only keeping it open as a reminder for me to do this change).

dcajacob, 2022-02-24:
Apologies. I didn't clearly see until now that d_last was really doing the same thing.

closed 2022-02-24 by dcajacob