daniestevez/gr-satellites #333: Sample recording wav file of AO-73 doesn't work with gr_satellites command

chouoo, 2022-02-14:
Hi, Daniel Estevez

Thanks for the work! 
I am a newbie for the gr satellite, so if there is a mistake during run of gr_satellites command then excuse me.

I found the sample recording wav file for AO-73 (ao73.wav for Funcube-1) doesn't work with the command line tool, gr_satellites (reference https://gr-satellites.readthedocs.io/en/latest/command_line.html). 
The error is
gr::log :DEBUG: decode_rs0 - Reed-Solomon decode fail (interleaver path 0)

But the recording file of GOMX-3 works for GOMX-3 (ref https://destevez1.rssing.com/chan-64037216/all_p14.html).

I am running it in gr_satellites v3.11.0 on Ubuntu 18.04 x64.

Cheol

daniestevez, 2022-02-14:
Can you check if your problem is the same as #217?

chouoo, 2022-02-15:
Thanks! It is working now by referencing the #217 issue.
Can you leave a note about the volk issue on the document (https://gr-satellites.readthedocs.io/en/latest/command_line.html)?

Cheol

daniestevez, 2022-02-24:
Sorry for the delay.

Yes. I think we could leave a note in the documentation. However, I'm not sure of where exactly to put this note. The "Command line" section doesn't seem a good idea, since this problem affects any use of gr-satellites.

It is perhaps best put inside the Installation section (but somewhere common, because this section is split into "from source", "conda" and "Ubuntu PPA", while this problem potentially affects all these methods). Alternatively, we could add a section called "Known problems" or "Potential problems".

What do you think?

chouoo, 2022-02-25:
No problem!

"Known problems" looks fine.

I suppose you may want to leave a short note at the end of test.sh in the bottom of the main repository, e.g., echo "NOTICE. All of these tests shall be successfully run if installation is successful. If not, please visit this link (https://gr-satellites.readthedocs.io/en/latest/command_line.html) and see Known problems section to find resolutions.". I think the test.sh script is a good place to catch the issue by most starters.

daniestevez, 2022-09-04:
In the end I have added a note about the broken AVX2 Viterbi decoder in Volk to the README in 1eea4426c127d94d16240d95bba5fcb91beec388. I think this is the location were people are more likely to find the note.

I'm closing this.

closed 2022-09-04 by daniestevez