daniestevez/gr-satellites #346: AttributeError: module 'gnuradio.blocks' has no attribute 'udp_source'

dw9ilx, 2022-03-26:
Hello, 
I am using the latest version of gr-satellites.  I am using UDP straight from GQRX on port 7355. I encounter a problem when I run tthis line on gr-satellites: 
$ gr_satellites $HOME/gr-satellites/python/satyaml/AO-73.yml --udp --samp_rate 48e3

The response from the software is: 
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 316, in <module>
    main()
  File "/usr/local/bin/gr_satellites", line 301, in main
    tb = gr_satellites_top_block(parser)
  File "/usr/local/bin/gr_satellites", line 179, in __init__
    self.setup_input()
  File "/usr/local/bin/gr_satellites", line 221, in setup_input
    return self.setup_udp_input()
  File "/usr/local/bin/gr_satellites", line 266, in setup_udp_input
    self.input_int16 = blocks.udp_source(
AttributeError: module 'gnuradio.blocks' has no attribute 'udp_source'

The problem is true to all supported satellites, but when I run the test.sh, and the recordings, it seems to work pretty well. What did I missed here? 

Thank you

dw9ilx, 2022-03-27:
Hi, I am using the GNU Radio version 3.10. is this version good for gr-satellites?

daniestevez, 2022-03-27:
This is indeed a bug that I hadn't noticed before.

I have pushed a new pull request #347 that I think fixes this bug. Can you test this?

dw9ilx, 2022-03-27:
I already checked the pull request that you've made.  However I got another problem when I run the script: 
$ gr_satellites $HOME/gr-satellites/python/satyaml/AO-73.yml --udp --samp_rate 48e3

The response:  
Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 16, in <module>
    if gr.api_version() != '9':
NameError: name 'gr' is not defined

I got the same result when I run the test.sh. What do I missed here?

daniestevez, 2022-03-27:
Sorry. My fault.

I've pushed a couple commits to the pull request fixing this problem and another problem that showed up after fixing this one. Now I think I've tested this properly, and it works for me, but I've be glad if you can test this before I merge the pull request.

dw9ilx, 2022-03-28:
I already checked the new commits, However, the decoder doesn't decode any data even if the signal is good. I tried to listen to the couple of supported satellites, but still no luck of decoding at least  1 frame. The test.sh runs pretty well. 

Everytime I listen to a supported satellite,  I only got this from the decoder: 
udp_source :info: Listening for data on UDP port 7355.

daniestevez, 2022-03-28:
Does this work if you use https://github.com/daniestevez/gr-frontends/blob/master/wav_48kHz.grc to play one of the WAV files in satellite-recordings using UDP?

dw9ilx, 2022-03-28:
I was not able to test the comment above, since I tried reverting my GNU Radio to 3.9.5, and re-installed the gr-satellites v4.6. It is now working well on my end. 
Reverting to previous GR version works for me and it solves my problem. I will settle on this for the time being. Thank you for your big help, Daniel. I learned a lot from your software.

daniestevez, 2022-03-28:
Thanks for testing. I'll merge the pull request now.

closed 2022-03-28 by daniestevez