daniestevez/gr-satellites #348: Support for CSP V2 protocol

skoold2003, 2022-04-07:
I'm getting incorrect data using the Print CSP Header block. I assume it's because gr–satellites is using CSP v1 protocol? Perhaps just add a toggle for which protocol to use?

daniestevez, 2022-04-07:
Indeed gr-satellites only implements CSP v1 currently. Which satellite is using CSP v2?

I think that a good solution for this is to add a construct definition for CSP v2 in [python/telemetry/csp.py](https://github.com/daniestevez/gr-satellites/blob/main/python/telemetry/csp.py). This can be used with the [Telemetry Parser](https://gr-satellites.readthedocs.io/en/latest/components.html#telemetry-parser) block.

The old Print CSP Header block is probably not very useful after the Telemetry Parser block was added, so I think this block should be deprecated.

skoold2003, 2022-04-07:
Well I know I'm only one man but I'm finding it useful. I've been working with CSP and trying to send it through a flow graph. As for what satellites are using v2 at the moment I assume none, but since it's default in libcsp now I assume any newly launched satellites (using CSP) will use the new protocol. 

P.S. great work on building this module!

daniestevez, 2022-04-11:
I have added a CSP v2 header definition in [python/telemetry/csp.py](https://github.com/daniestevez/gr-satellites/blob/main/python/telemetry/csp.py). This can be used with the Telemetry Parser block.

daniestevez, 2022-05-22:
I'm closing this, since a CSP v2 header definition has already been added. Feel free to re-open for any possible follow ups on this topic.

closed 2022-05-22 by daniestevez