daniestevez/gr-satellites #357: Unable to decode FORESAIL-1 using the AX100 ASM+Golay yml option

janvgils, 2022-05-26:
It seems FORESAIL-1 is using a non standard AX100 sync word.

From the below document, the following: `ASM (0x1ACFFC1D) + Golay24 + RS(255,233) (AX.100 Mode 5 compatible)`

https://foresail.github.io/docs/FS1_Space_Ground_Interface_Control_Sheet.pdf

Is there a way to alter this in gr_satellites?

Some sources and test files are available via this [FORESAIL-1](https://www.pe0sat.vgnet.nl/download/FORESAIL-1/)  link.

daniestevez, 2022-05-26:
The AX100 deframer needs to be modified to account for this. It's a simple change. I'll try to do it over the weekend.

daniestevez, 2022-05-27:
I have pushed 8dcfa431cf981e102e8119abeb174251c1398777, which changes the syncword that is used by FORESAIL-1 to `0x1ACFFC1D`. I think this will work, but haven't tested it.

N6RFM, 2022-05-28:
Thanks Dani.  This is working nicely.  
73,  Bob N6RFM

daniestevez, 2022-05-28:
Thanks for testing. Since this works, I'm closing the issue.

closed 2022-05-28 by daniestevez

janvgils, 2022-05-28:
Wonder full, thanks for the continuing support!

Just tested a couple of observations and it looks fine.

```
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length =         73 bytes
pdu vector contents = 
0000: 66 4f 48 32 46 31 53 29 05 a9 26 54 00 fa 00 f4 
0010: 0b 34 0b 34 00 2b 10 03 02 83 11 50 93 01 80 00 
0020: 0a 11 b7 03 00 5c 00 dc 0e 94 6d 5f 0e 10 00 99 
0030: 00 06 53 00 05 07 00 04 05 30 07 31 07 32 07 33 
0040: 07 34 d7 00 e2 e7 db 6d e5 
************************************
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length =         95 bytes
pdu vector contents = 
0000: 66 4f 48 32 46 31 53 29 05 a9 27 54 00 fa 00 78 
0010: 0b 34 0b 34 00 41 10 03 05 83 11 50 93 00 00 e8 
0020: fa 80 47 00 00 00 00 00 00 00 00 00 00 00 00 00 
0030: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
0050: 00 00 00 00 00 00 00 84 7d f8 b7 1d fd 8d 19 
************************************
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length =         77 bytes
pdu vector contents = 
0000: 66 4f 48 32 46 31 53 29 05 a9 28 54 00 fa 00 f6 
0010: 0b 34 0b 34 00 2f 10 03 04 83 11 50 93 f5 16 00 
0020: 00 13 00 00 00 00 da 01 00 00 2b ec 00 00 dd 4a 
0030: 00 00 00 00 00 00 00 00 00 00 00 02 02 ab 00 a2 
0040: 00 f1 00 00 00 c2 a7 b5 5e ec f1 3a 3a 
************************************
***** VERBOSE PDU DEBUG PRINT ******
((transmitter . 9k6 FSK downlink))
pdu length =        165 bytes
pdu vector contents = 
0000: 66 4f 48 32 46 31 53 29 05 a9 29 54 00 fa 00 79 
0010: 0b 34 0b 34 00 87 10 03 03 83 11 50 93 dc bb 03 
0020: 00 05 60 00 2c 01 0a 2d 0a 3f 0a 3f 0a 01 0a 36 
0030: 0a 3f 0a 3f 0a 00 00 00 00 00 00 00 00 00 00 00 
0040: 00 00 00 00 00 be 1f e0 ff f0 ff 05 00 06 00 af 
0050: 00 ad 0e d1 0e 17 0f 00 00 00 00 00 00 00 00 00 
0060: 00 48 00 45 00 00 00 00 00 00 00 00 00 00 00 00 
0070: 00 45 00 3d 00 00 00 00 00 00 00 00 00 00 00 00 
0080: 00 5a 00 65 00 00 00 00 00 06 00 00 01 bc 1f c4 
0090: 0f 83 00 7c 00 96 00 74 00 77 00 00 00 b0 8c 77 
00a0: 96 06 60 2b 6d 
************************************
```