daniestevez/gr-satellites #367: QO-100 hsmodem decoder sometimes dies decoding file 0

phl0, 2022-06-15:
The multimedia beacon decoder sometimes tries to RX a file with name "0" and then dies in processing it.

```
FileReceiver: received sequence 34 for file 0                                                            
thread_body_wrapper :error: ERROR thread[thread-per-block[62]: <block file_receiver(29)>]: TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'
                                                                                                         
At:                                                                                                      
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/qo100_multimedia.py(42): chunk_data
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/filereceiver.py(304): push_chunk
  /usr/local/lib/python3.10/dist-packages/satellites/components/datasinks/file_receiver.py(65): handle_msg
```

daniestevez, 2022-06-19:
Thanks for reporting.

What git commit are you using exactly? In ef7d7912bc40b68c0de26fbe2ddb68a640cf4068 I did some modifications that I think should prevent this problem for happening (but I am thinking this just by reading the code; maybe there's some corner case I'm not seeing).

Regarding the file name "0" this is what the file receiver does when the file name is only carried in the first chunk of the file (as is the case in the QO-100 beacon), but the reception misses that chunk (because the chunk is lost or the receiver starts mid-way in the file). Without having a better name, the receiver assigns files the filenames 0, 1, etc. Probably this is not very useful for the QO-100 beacon, but it's quite useful for some LEO satellites where typically only one or two files are transmitted in a pass and there is often packet loss, so the files need to be pieced from the data received by several stations).

In any case, the decoder shouldn't die, so if there is some possible scenario that makes it die with the current main/maint-3.9/maint-3.8 branch, then it should be fixed.

phl0, 2022-06-19:
I am on https://github.com/daniestevez/gr-satellites/commit/65daf3f97a674669eaabfd688a863f32b72cd02d so that should include the fix.

daniestevez, 2022-06-19:
Are you sure you have the fix included?

I'm testing with the current maint-3.9, where I have introduced a modification to artificially drop the first chunk of every file, to try to replicate the problem.

This is what I see:
```
FileReceiver: new file 0
FileReceiver: received sequence 1 for file 0
FileReceiver: received sequence 2 for file 0
Could not get file data for last chunk: unsupported operand type(s) for -: 'NoneType' and 'int'
FileReceiver: file 0 is finished
FileReceiver: receiving new file 1
FileReceiver: received sequence 1 for file 1
FileReceiver: received sequence 2 for file 1
FileReceiver: received sequence 3 for file 1
FileReceiver: received sequence 4 for file 1
FileReceiver: received sequence 5 for file 1
FileReceiver: received sequence 6 for file 1
FileReceiver: received sequence 7 for file 1
FileReceiver: received sequence 8 for file 1
FileReceiver: received sequence 9 for file 1
FileReceiver: received sequence 10 for file 1
FileReceiver: received sequence 11 for file 1
FileReceiver: received sequence 12 for file 1
FileReceiver: received sequence 13 for file 1
FileReceiver: received sequence 14 for file 1
FileReceiver: received sequence 15 for file 1
FileReceiver: received sequence 16 for file 1
FileReceiver: received sequence 17 for file 1
FileReceiver: received sequence 18 for file 1
FileReceiver: received sequence 19 for file 1
FileReceiver: received sequence 20 for file 1
FileReceiver: received sequence 21 for file 1
FileReceiver: received sequence 22 for file 1
FileReceiver: received sequence 23 for file 1
FileReceiver: received sequence 24 for file 1
FileReceiver: received sequence 25 for file 1
FileReceiver: received sequence 26 for file 1
FileReceiver: received sequence 27 for file 1
FileReceiver: received sequence 28 for file 1
FileReceiver: received sequence 29 for file 1
FileReceiver: received sequence 30 for file 1
FileReceiver: received sequence 31 for file 1
FileReceiver: received sequence 32 for file 1
FileReceiver: received sequence 33 for file 1
Could not get file data for last chunk: unsupported operand type(s) for -: 'NoneType' and 'int'
FileReceiver: file 1 is finished
FileReceiver: receiving new file 2
FileReceiver: received sequence 1 for file 2
FileReceiver: received sequence 2 for file 2
```

Note the "Could not get file data..." message, which is printed by ef7d7912bc40b68c0de26fbe2ddb68a640cf4068 rather than crashing.

phl0, 2022-06-19:
Just re-started the python script:

```
FileReceiver: received sequence 3 for file 0
thread_body_wrapper :error: ERROR thread[thread-per-block[62]: <block file_receiver(29)>]: TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'
                                                                                                         
At:                    
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/qo100_multimedia.py(42): chunk_data
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/filereceiver.py(304): push_chunk
  /usr/local/lib/python3.10/dist-packages/satellites/components/datasinks/file_receiver.py(65): handle_msg
                                                                                                         
correlate_access_code_tag_bb :debug: writing tag at sample 39357            
```

line 64 in python/filereceiver/qo100_multimedia.py is: 

```
print('Could not get file data for last chunk:', e)
```

So I think that fix is included in my code.

daniestevez, 2022-06-19:
Can you add a KISS File Sink block to `qo100_multimedia_beacon.grc` as shown and regenerate the hierarchical flowgraph (Run > Generate menu). This will save all the received frames to a file so that we can the debug by trying to replicate the problem using exactly the same sequence of received frames.

[image]

phl0, 2022-06-19:
And then run the decoder waiting for the error to appear?

daniestevez, 2022-06-19:
Yes.

phl0, 2022-06-19:
Here we go. I killed the script while qo100_info.html was received and started again. It was then receiving file 0 as expected. Then:

```
FileReceiver: received sequence 67 for file 0
thread_body_wrapper :error: ERROR thread[thread-per-block[62]: <block file_receiver(33)>]: TypeError: unsupported operand type(s) for -: 'NoneType' and 'int'

At:
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/qo100_multimedia.py(42): chunk_data
  /usr/local/lib/python3.10/dist-packages/satellites/filereceiver/filereceiver.py(304): push_chunk
  /usr/local/lib/python3.10/dist-packages/satellites/components/datasinks/file_receiver.py(65): handle_msg
```

Shall I send the KISS file via email?

daniestevez, 2022-06-19:
I think you can attach it here. Maybe you need to zip it.

phl0, 2022-06-19:
Ah ok. Here you go.

[qo100_data.zip](https://github.com/daniestevez/gr-satellites/files/8935805/qo100_data.zip)

daniestevez, 2022-06-19:
I'm playing back that KISS file with the attached flowgraph. I get the following:
```
FileReceiver: new file 0
FileReceiver: received sequence 1 for file 0
FileReceiver: received sequence 2 for file 0
FileReceiver: received sequence 3 for file 0
FileReceiver: received sequence 4 for file 0
FileReceiver: received sequence 5 for file 0
FileReceiver: received sequence 6 for file 0
FileReceiver: received sequence 7 for file 0
FileReceiver: received sequence 8 for file 0
FileReceiver: received sequence 9 for file 0
FileReceiver: received sequence 10 for file 0
FileReceiver: received sequence 11 for file 0
FileReceiver: received sequence 12 for file 0
FileReceiver: received sequence 13 for file 0
FileReceiver: received sequence 14 for file 0
FileReceiver: received sequence 15 for file 0
FileReceiver: received sequence 16 for file 0
FileReceiver: received sequence 17 for file 0
FileReceiver: received sequence 18 for file 0
FileReceiver: received sequence 19 for file 0
FileReceiver: received sequence 20 for file 0
FileReceiver: received sequence 21 for file 0
FileReceiver: received sequence 22 for file 0
FileReceiver: received sequence 23 for file 0
FileReceiver: received sequence 24 for file 0
FileReceiver: received sequence 25 for file 0
FileReceiver: received sequence 26 for file 0
FileReceiver: received sequence 27 for file 0
Could not get file data for last chunk: unsupported operand type(s) for -: 'NoneType' and 'int'
```

[play_kiss_file.grc.zip](https://github.com/daniestevez/gr-satellites/files/8935856/play_kiss_file.grc.zip)

---

Another thing:

> line 64 in python/filereceiver/qo100_multimedia.py is:
> print('Could not get file data for last chunk:', e)

Are you sure you are looking at the same code that you're running? The code you're running is probably in `~/.local/lib` or `/usr/local`.

phl0, 2022-06-19:
Ah right. My fault. I guess I have to re-install gr-satellites as I was thinking it uses the lib in the checkout. I will go and try agn. sftn.

phl0, 2022-06-19:
Ok, I think we can close that. After "sudo make install"-ing the gr-satellites again it is using the fixed qo100_multimedia.py. Now it seems to not die on this error:

```
FileReceiver: received sequence 66 for file 0
correlate_access_code_tag_bb :debug: writing tag at sample 274002
decode_rs :debug: Reed-Solomon decode corrected 4 bytes (interleaver path 0)
crc_check :info: CRC OK
correlate_access_code_tag_bb :debug: writing tag at sample 276066
decode_rs :debug: Reed-Solomon decode corrected 3 bytes (interleaver path 0)
crc_check :info: CRC OK
correlate_access_code_tag_bb :debug: writing tag at sample 278130
decode_rs :debug: Reed-Solomon decode corrected 3 bytes (interleaver path 0)
crc_check :info: CRC OK
Could not get file data for last chunk: unsupported operand type(s) for -: 'NoneType' and 'int'
correlate_access_code_tag_bb :debug: writing tag at sample 280194
decode_rs :debug: Reed-Solomon decode corrected 6 bytes (interleaver path 0)
crc_check :info: CRC OK
```
How can I disable the verbose debug output btw?

daniestevez, 2022-06-19:
Take a look at https://wiki.gnuradio.org/index.php/Logging. You can set there the log levels to warn instead of info/debug.

phl0, 2022-06-19:
Roger. Will do that. Closing here.

closed 2022-06-19 by phl0