daniestevez/gr-satellites #383: IQ realtime don't decode but AF replay do

janvgils, 2022-07-20:
Good day,

More and more I see SatNOGS <-> gr-satellites integration that don't decode realtime (the integration uses IQ data), but when I replay the observation with AF as input it does. Below some details.

satnogs <-> gr_satellites realtime IQ cli command:

```
GROPT="$NORAD --samp_rate $SAMP --iq --throttle --udp --udp_port $UDP_DUMP_PORT --udp_raw --start_time $DATEF --kiss_out $KSS --ignore_unknown_args --use_agc"

$GRSBIN $GROPT

```

satnogs <-> gr_satellites replay AF cli command:

```
with_offset(){
  echo "# Decoding Satellite $SAT with OFFSET enabled."
  gr_satellites "${SAT}" --wavfile "${WAV}" --samp_rate 48e3 --f_offset 12e3 --start_time "${DF}" --throttle
}

no_offset(){
  echo "# Decoding Satellite $SAT with OFFSET disabled."
  gr_satellites "${SAT}" --wavfile "${WAV}" --samp_rate 48e3 --start_time "${DF}" --throttle ${GROPTIONS}
}
```

Just a couple of observation examples (sample rate 48k3):
https://network.satnogs.org/observations/6241297/ (BPSK) can be downloaded/used for AF replay
https://network.satnogs.org/observations/6241465/ (GMSK) can be downloaded/used for AF replay

The same obs but the saved IQ files, sample rate 48k3:
[iq_431_6241297.raw](https://www.pe0sat.vgnet.nl/download/SatNOGS/iq_431_6241297.raw)
[iq_431_6241465.raw](https://www.pe0sat.vgnet.nl/download/SatNOGS/iq_431_6241465.raw)

```
gr_satellites --version
gr_satellites v3.14.0-git
Copyright (C) 2020 Daniel Estevez
License GPLv3+: GNU GPL version 3 or later <https://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
```

Is there anything we can do, maybe add some parameters so the IQ will also decode the data realtime?

Jan - PE0SAT

daniestevez, 2022-07-20:
Hi Jan,

I'm trying to replicate this and getting the following with the first obseration (PICSAT):
```
$ gr_satellites PICSAT --rawint16 iq_431_6241297.raw --iq --samp_rate 48e3 | grep "Packet from" | wc -l
gr::debug :INFO: vmcircbuf_sysconfig - Using gr::vmcircbuf_sysv_shm_factory
7
$ gr_satellites PICSAT --wavfile satnogs_6241297_2022-07-20T10-32-28.ogg --f_offset 12e3 | grep "Packet from" | wc -l
gr::debug :INFO: vmcircbuf_sysconfig - Using gr::vmcircbuf_sysv_shm_factory
8
```
So I get 7 decodes with the IQ file and 8 decodes with the OGG file. I guess that one of these packets may be just barely decodable, and so it decodes with the OGG file and not with the IQ file due to luck. Do you get something different with your system.

This is not realtime playback, though. I see that for realtime playback you're using the `--udp` option as well as the `--throttle` option. This is not advisable. If the data coming in from UDP comes at a slightly faster rate than what the throttle does (due to a difference between your SDR clock and your computer clock), then the UDP source will throw away data, which would explain why you're not getting decodes (but I'm not sure if the UDP source is indeed throwing data in your case). Can you try removing the `--throttle` option? `--throttle` should only be used to play back a recording at real time rate, never with UDP.

janvgils, 2022-07-22:
Good afternoon Daniel,

Thanks for the analyses and this sounds like a very good idea.
I will remove the --throttle option and see if that will alter the behavior.

All the best,

Jan van Gils

janvgils, 2022-07-23:
This seemed to be a very good idea, but it isn't the solution.
Here again a very good observation https://network.satnogs.org/observations/6254232/ and no "realtime" decode.
Replaying the audio works fine.

daniestevez, 2022-07-23:
I don't see an easy way to debug this, since both the audio and the IQ recording work fine if processed afterwards. It's only the realtime decoding that fails.

An idea is to use wireshark or tcpdump to capture the UDP traffic being sent to gr-satellites. From this one could extract the data an feed it later as a file or by UDP again. Another idea is to modify gr-satellites to save the data that it receives by UDP to a file. This could be used to check if there are some UDP packets being lost. The `--dump_path` option from gr-satellites could also be useful. It saves some internal signals to files.

Finally, perhaps a silly question: when you say that there are no realtime decodes with gr-satellites, are you able to see the output of gr-satellites directly or are you inferring that because there are no packets shown in the SatNOGS observation? Could it be that gr-satellites decodes fine but the problem is in uploading the packets to SatNOGS?

janvgils, 2022-07-24:
Thank you for the reply and the suggested ways to debug this.

I will try and see if I can produce the data.

Jan - PE0SAT

janvgils, 2022-07-24:
Good afternoon Daniel,

My deepest apologies.

I have made a very "stupid" mistake. As you may remember I am running [multiple](https://www.pe0sat.vgnet.nl/satellite/satnogs/using-a-second-sdr/) satnogs-clients on the same physical system.
Because of this I need to configure multiple UDP ports and there I made a terrible mistake, I configured a value above 65000 and as you know that isn't possible.

So now I have reconfigured this settings and I will check if this was indeed the problem.

Sorry for taking up your time for this stupid mistake.

janvgils, 2022-07-24:
Based on your advice I also created the `dump_path` output and `tcpdump -i lo port 50431 -s0 -w satnogs-client_udp_pcap` these files can be downloaded via the following [link](https://pe0sat.vgnet.nl/download/gr-satellites/issue-383/).

Looking at the pcap output, there are sometimes RTCP messages and those I wouldn't expect.

So maybe at the end my mistake in useful at all.

daniestevez, 2022-07-24:
Does the problem still persist after you fixed the problem with the port above 65536?

janvgils, 2022-07-24:
Yes it does. this obs https://network.satnogs.org/observations/6258833/ didn't decode using the IQ via UDP but is did after AF replay.

janvgils, 2022-07-25:
Well there is the risk we are chasing multiple challenges, as an example:

Looking at this obs: https://network.satnogs.org/observations/6260900/ [files](https://www.pe0sat.vgnet.nl/download/gr-satellites/issue-383/)

```
LUME-1 (NORAD 43908)
    4k8 FSK downlink 437.060 MHz FSK AX100 ASM+Golay
```

`gr_satellites LUME-1 --wavfile AF_48e3_LUME-1_6260900.wav --samp_rate 48e3 --hexdump` successful decoding.
`gr_satellites LUME-1 --rawint16 IQ_48e3_LUME-1_6260900.wav --samp_rate 48e3 --iq --use_agc --hexdump` no decoding.

So there is a possibility this isn't real time IQ UDP related.

I will try to create an obs and save all the debug data like: `dump_path and pcap` and others.

janvgils, 2022-07-28:
Close this issue for now, there is some randomness that makes it very difficult to reproduce.
And therefor very hard to produce all the information for a single failing observation (AF, IQ, dump_path and pcap)

closed 2022-07-28 by janvgils

daniestevez, 2022-07-28:
Ok. Let's leave this closed. The problem seems quite weird and I don't have enough time at the moment to look at it in detail.

I wonder if some of the other people using gr-satellites with SatNOGS also have this problem, since it's the first time I hear about this issue.