daniestevez/gr-satellites #390: Latest master pull: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.3: undefined symbol

janvgils, 2022-07-31:
Just wanted to shared that after a git pull (31-07-2022 12:00 UTC) from the master branch I run into a `undefined symbol` when running gr_satellites.

Error log:

```
-- Installing: /usr/local/share/gnuradio/grc/blocks/satellites_usp_pls_crop.block.yml
janvg@locutus:~/source/git/daniestevez/gr-satellites/build$ sudo ldconfig
janvg@locutus:~/source/git/daniestevez/gr-satellites/build$ gr_satellites --version

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gnuradio/gr/__init__.py", line 26, in <module>
    from .gr_python import *
ImportError: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.3: undefined symbol: _ZN6spdlog5sinks15basic_file_sinkINS_7details10null_mutexEEC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/gr_satellites", line 15, in <module>
    from gnuradio import gr, blocks, audio
  File "/usr/lib/python3/dist-packages/gnuradio/gr/__init__.py", line 30, in <module>
    from .gr_python import *
ImportError: /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.10.3: undefined symbol: _ZN6spdlog5sinks15basic_file_sinkINS_7details10null_mutexEEC1ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEb
```

System information:

```
cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux bookworm/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/
```

GNURadio packages installed:

```
dpkg -l | grep gnuradio
ii  gnuradio                                                    3.10.3.0-2+b1                      amd64        GNU Radio Software Radio Toolkit
ii  gnuradio-dev                                                3.10.3.0-2+b1                      amd64        GNU Software Defined Radio toolkit development
ii  gr-fosphor                                                  3.9~0.974ab2f-1+b5                 amd64        gnuradio fosphor block (GPU spectrum display)
ii  libgnuradio-analog3.10.3:amd64                              3.10.3.0-2+b1                      amd64        gnuradio analog functions
ii  libgnuradio-audio3.10.3:amd64                               3.10.3.0-2+b1                      amd64        gnuradio audio functions
ii  libgnuradio-blocks3.10.2:amd64                              3.10.2.0-1+b3                      amd64        gnuradio blocks functions
ii  libgnuradio-blocks3.10.3:amd64                              3.10.3.0-2+b1                      amd64        gnuradio blocks functions
ii  libgnuradio-channels3.10.3:amd64                            3.10.3.0-2+b1                      amd64        gnuradio channels functions
ii  libgnuradio-digital3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio digital communications functions
ii  libgnuradio-dtv3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio digital TV signal processing blocks
ii  libgnuradio-fec3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio forward error correction support
ii  libgnuradio-fft3.10.2:amd64                                 3.10.2.0-1+b3                      amd64        gnuradio fast Fourier transform functions
ii  libgnuradio-fft3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio fast Fourier transform functions
ii  libgnuradio-filter3.10.2:amd64                              3.10.2.0-1+b3                      amd64        gnuradio filter functions
ii  libgnuradio-filter3.10.3:amd64                              3.10.3.0-2+b1                      amd64        gnuradio filter functions
ii  libgnuradio-fosphor3.9.0:amd64                              3.9~0.974ab2f-1+b5                 amd64        gnuradio fosphor block (GPU spectrum display) (library)
ii  libgnuradio-funcube1.0.0                                    1.0.0-3+b3                         amd64        Funcube Dongle controller for GNU Radio (runtime)
ii  libgnuradio-iio3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        IIO blocks for GnuRadio
ii  libgnuradio-iqbalance3.9.0                                  0.38.2-2+b4                        amd64        GNU Radio Blind IQ imbalance estimator and correction (shared library)
ii  libgnuradio-network3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio network library
ii  libgnuradio-osmosdr0.2.0:amd64                              0.2.3-6+b2                         amd64        Gnuradio blocks from the OsmoSDR project - library
ii  libgnuradio-pdu3.10.2:amd64                                 3.10.2.0-1+b3                      amd64        gnuradio streaming protocol library
ii  libgnuradio-pdu3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio streaming protocol library
ii  libgnuradio-pmt3.10.2:amd64                                 3.10.2.0-1+b3                      amd64        gnuradio pmt container library
ii  libgnuradio-pmt3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio pmt container library
ii  libgnuradio-qtgui3.10.3:amd64                               3.10.3.0-2+b1                      amd64        gnuradio Qt graphical user interface functions
ii  libgnuradio-runtime3.10.2:amd64                             3.10.2.0-1+b3                      amd64        gnuradio core runtime
ii  libgnuradio-runtime3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio core runtime
ii  libgnuradio-soapy3.10.3:amd64                               3.10.3.0-2+b1                      amd64        gnuradio SoapySDR support
ii  libgnuradio-trellis3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio trellis modulation functions
ii  libgnuradio-uhd3.10.3:amd64                                 3.10.3.0-2+b1                      amd64        gnuradio universal hardware driver functions
ii  libgnuradio-video-sdl3.10.3:amd64                           3.10.3.0-2+b1                      amd64        gnuradio video functions
ii  libgnuradio-vocoder3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio vocoder functions
ii  libgnuradio-wavelet3.10.2:amd64                             3.10.2.0-1+b3                      amd64        gnuradio wavelet functions
ii  libgnuradio-wavelet3.10.3:amd64                             3.10.3.0-2+b1                      amd64        gnuradio wavelet functions
ii  libgnuradio-zeromq3.10.3:amd64                              3.10.3.0-2+b1                      amd64        gnuradio zeromq functions
```

daniestevez, 2022-07-31:
Hi Jan,

Thanks for the bug report. It seems to me that this is not related to gr-satellites, but perhaps a problem of how the GNU Radio package in Debian sid was built (maybe you updated the package recently). The file with an undefined symbol is libgnuradio-runtime.so.3.10.3, which comes from the libgnuradio-runtime3.10.3 Debian package.

Can you test if a simple flowgraph that doesn't involve gr-satellites works? If libgnuradio-runtime.so.3.10.3 has missing symbols, nothing would work at all.

Something strange that I notice is that you seem to have packages for both GNU Radio 3.10.2 and 3.10.3 installed. Perhaps the previous time that you built gr-satellites it was linked against 3.10.2, and now it has linked against 3.10.3?

janvgils, 2022-07-31:
Thanks for the reply, will first purge the old ones

janvgils, 2022-07-31:
After removing the 3.10.2 packages, I rebuild gr-satellites, the error was the same, and as what you expected running a flowgraph also give this error, something seem to be wrong with the package.

`gnuradio-companion bpsk.grc`

[image]

So this needs to be fixed by the Debian maintainers.

Thanks for the support.

janvgils, 2022-08-01:
Just tested this on another system and we get the same error.
Now lets see how to register a Debian GNURadio "bug"

janvgils, 2022-08-01:
The bug is registered https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1016466

janvgils, 2022-08-04:
> The bug is registered https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1016466

@daniestevez Do you have any experience with the Debian bug report system?
I get the impression this bug isn't recognized as a problem.

daniestevez, 2022-08-04:
I'm only vaguely familiar with how the Debian bug tracking system and the maintainer procedures work. Looking at the bug report I'd say that everything is going normally and the problem will be eventually fixed. I guess that it isn't too rare for a package to be broken in sid for a while until the problem is fixed, particularly if the best fix is not straightforward. That's the purpose of sid, really.

janvgils, 2022-08-14:
The following, for now, has solved the issue `sudo apt-get -t sid install gnuradio-dev` currently sid has version `3.10.3.0-3`

daniestevez, 2022-08-14:
Great. Then I guess this issue can be closed (you can close the issue yourself).

closed 2022-08-15 by janvgils