daniestevez/gr-satellites #391: Flowgraph error after upgrading to v5

lsdta, 2022-08-03:
Thank you for your work on this great module.

I have a working flowgraph that failed upon upgrading gr-satellites to v5 (from v4.6.0)
The CCSDS R-S deframer block (in particular the sync_to_pdu.py) no longer works and gives the following error:
`ValueError: invalid msg port in connect() or disconnect()`

My System:
ubuntu: 22.04
gnuradio: 3.10.2.0 (from gnuradio PPA)
python: 3.10.4
gr-satellites: v5.0.0 (from daniestevez PPA)

Error messsage in full:

```...

correlate_access_code_tag_bb :debug: Access code: 1acffc1d
correlate_access_code_tag_bb :debug: Mask: ffffffff
flowgraph :warning: Could not find port pdus in:
flowgraph :warning:   system
Traceback (most recent call last):
  File "/redacted.py", line 491, in <module>
    main()
  File "/redacted.py", line 469, in main
    tb = top_block_cls()
  File "/redacted.py", line 132, in __init__
    self.satellites_ccsds_concatenated_deframer_0 = satellites.components.deframers.ccsds_concatenated_deframer(frame_size = 223, precoding = None, rs_basis = "conventional", rs_interleaving = interleave_depth, scrambler = "CCSDS", convolutional = "CCSDS", syncword_threshold = 0, options="")
  File "/usr/lib/python3/dist-packages/satellites/components/deframers/ccsds_concatenated_deframer.py", line 59, in __init__
    self.rs0 = ccsds_rs_deframer(
  File "/usr/lib/python3/dist-packages/satellites/components/deframers/ccsds_rs_deframer.py", line 69, in __init__
    self.deframer = deframe_func(
  File "/usr/lib/python3/dist-packages/satellites/hier/sync_to_pdu.py", line 50, in __init__
    self.msg_connect(
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 51, in wrapped
    func(self, src.to_basic_block(), srcport, dst.to_basic_block(), dstport)
  File "/usr/lib/python3/dist-packages/gnuradio/gr/hier_block2.py", line 127, in msg_connect
    self.primitive_msg_connect(*args)
ValueError: invalid msg port in connect() or disconnect()```

daniestevez, 2022-08-03:
Hi, can you give a flowgraph that reproduces this problem?

I don't think this problem happens on my system, since it would cause everything using the `sync_to_pdu` block to fail.

Note that the gr-satellites package in the daniestevez PPA is built against the gnuradio package from Ubuntu 22.04, rather that the package in the gnuradio PPA, so that might be the cause of this problem.

lsdta, 2022-08-03:
Thanks, I will test against the Ubuntu gnuradio build and report back with a simplified flowgraph if I still get an error.

lsdta, 2022-08-04:
After running the same flowgraph after only changing to the Ubuntu build of gnuradio (3.10.1.1, rather than the PPA build of 3.10.2.0), I do not encounter the error. Problem solved, at least for now. Thanks!

closed 2022-08-04 by lsdta