daniestevez/gr-satellites #398: TypeError: in method 'doppler_correction_make'

K4KDR, 2022-08-24:
Hello!

When attempting to run the example flowgraph **doppler_correction_example.grc**, I get the following:

```
Traceback (most recent call last):
  File "/home/k4kdr/grc/doppler_correction_example.py", line 228, in <module>
    main()
  File "/home/k4kdr/grc/doppler_correction_example.py", line 204, in main
    tb = top_block_cls()
  File "/home/k4kdr/grc/doppler_correction_example.py", line 81, in __init__
    self.satellites_doppler_correction_0 = satellites.doppler_correction('/home/k4kdr/gr-satellites/examples/doppler_correction/doppler_file.txt', samp_rate, 0)
  File "/usr/local/lib/python3/dist-packages/satellites/satellites_swig.py", line 4217, in make
    return _satellites_swig.doppler_correction_make(filename, samp_rate, t0)
TypeError: in method 'doppler_correction_make', argument 1 of type 'std::string &'

>>> Done (return code 1)
```

... I have searched online but so far have not found anything to indicate what is causing this.  

My relevant system info:

python --version
**Python 3.6.9**

swig -version
**SWIG Version 4.0.2**

uname -a
**Linux 3010i5 5.4.0-124-generic #140~18.04.1-Ubuntu SMP Fri Aug 5 11:43:34 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux**

Thank you!

daniestevez, 2022-08-24:
I think that #399 should fix this. Can you test this branch and check if it works properly with GNU Radio 3.8?

K4KDR, 2022-08-24:
Thanks so much for checking on this!!

The "make" step fails as shown below:

```
[ 94%] Building CXX object lib/CMakeFiles/gnuradio-satellites.dir/viterbi/viterbi.cc.o
[ 95%] Linking CXX shared library libgnuradio-satellites.so
[ 95%] Built target gnuradio-satellites
[ 95%] Built target satellites_swig_swig_compilation
[ 96%] Building CXX object swig/CMakeFiles/satellites_swig.dir/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx.o
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1847:36: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*)’ {aka ‘_object* (*)(_object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
   {(char *)"disown",  (PyCFunction)SwigPyObject_disown,  METH_NOARGS,  (char *)"releases ownership of the pointer"},
                                    ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1848:36: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*)’ {aka ‘_object* (*)(_object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
   {(char *)"acquire", (PyCFunction)SwigPyObject_acquire, METH_NOARGS,  (char *)"acquires ownership of the pointer"},
                                    ^~~~~~~~~~~~~~~~~~~~
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1851:36: warning: cast between incompatible function types from ‘PyObject* (*)(PyObject*)’ {aka ‘_object* (*)(_object*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
   {(char *)"next",    (PyCFunction)SwigPyObject_next,    METH_NOARGS,  (char *)"returns the next 'this' object"},
                                    ^~~~~~~~~~~~~~~~~
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1852:36: warning: cast between incompatible function types from ‘PyObject* (*)(SwigPyObject*)’ {aka ‘_object* (*)(SwigPyObject*)’} to ‘PyCFunction’ {aka ‘_object* (*)(_object*, _object*)’} [-Wcast-function-type]
   {(char *)"__repr__",(PyCFunction)SwigPyObject_repr,    METH_NOARGS,  (char *)"returns object representation"},
                                    ^~~~~~~~~~~~~~~~~
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function ‘PyObject* _wrap_doppler_correction_make(PyObject*, PyObject*, PyObject*)’:
/home/k4kdr/gr-sats-swig-test/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:36920:72: error: no matching function for call to ‘gr::satellites::doppler_correction::make(std::__cxx11::string&, double&, double&)’
       result = gr::satellites::doppler_correction::make(*arg1,arg2,arg3);
```

K4KDR, 2022-08-24:
I was starting with an empty folder... a new folder tree created just for testing this single branch.    I will see what I can find about that error specifically!

daniestevez, 2022-08-24:
That's weird. I've just tested again and this build for me. I get some warnings, but not the "no matching function for call" error:

```
[ 95%] Linking CXX shared library libgnuradio-satellites.so
[ 95%] Built target gnuradio-satellites
Scanning dependencies of target satellites_swig
[ 96%] Building CXX object swig/CMakeFiles/satellites_swig.dir/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx.o
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function 'PyTypeObject* SwigPyObject_TypeOnce()':
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1770:5: warning: missing initializer for member '_typeobject::tp_vectorcall' [-Wmissing-field-initializers]
 1770 |     };
      |     ^
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1770:5: warning: missing initializer for member '_typeobject::tp_print' [-Wmissing-field-initializers]
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function 'PyTypeObject* SwigPyPacked_TypeOnce()':
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1931:5: warning: missing initializer for member '_typeobject::tp_vectorcall' [-Wmissing-field-initializers]
 1931 |     };
      |     ^
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:1931:5: warning: missing initializer for member '_typeobject::tp_print' [-Wmissing-field-initializers]
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx: In function 'PyTypeObject* swig_varlink_type()':
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:97303:7: warning: missing initializer for member '_typeobject::tp_vectorcall' [-Wmissing-field-initializers]
97303 |       };
      |       ^
/gr-satellites/build/swig/CMakeFiles/satellites_swig.dir/satellites_swigPYTHON_wrap.cxx:97303:7: warning: missing initializer for member '_typeobject::tp_print' [-Wmissing-field-initializers]
[ 97%] Linking CXX shared module _satellites_swig.so
[ 97%] Built target satellites_swig
Scanning dependencies of target pygen_swig_b71b8
[100%] Generating satellites_swig.pyc
[100%] Generating satellites_swig.pyo
[100%] Built target pygen_swig_b71b8
```

I think that somehow you have a leftover from the old code, since your error is mentioning `std::__cxx11::string&`, which is the thing I replaced by `const char*`.

K4KDR, 2022-08-24:
I will have to continue researching - no luck so far.   I see many references to this error online from various projects.

I did try adding compile option "-Wno-cast-function-type", but the "make" step continues to fail.   I will see what else I can find!!

[image]

daniestevez, 2022-08-24:
Maybe look in the files `include/satellites/doppler_correction.h`, `lib/doppler_correction_impl.h` and `lib/doppler_correction_impl.cc`. There should be no `std::string&` there in the code you're trying to build.

K4KDR, 2022-08-24:
I wonder if I did not obtain the latest, updated files using the correct method.

In an empty directory, I ran:

`git clone -b swig-doppler-correction --single-branch https://github.com/daniestevez/gr-satellites.git`

... is there some other method that I should use to obtain your most recent updates??

Thanks!

daniestevez, 2022-08-24:
That seems. Correct. The only other thing that comes to mind is that maybe you have the old `doppler_correction.h` installed in `/usr/local/include/satellites/` or similar. In principle this shouldn't conflict, but all the tests I'm doing are on a fresh docker image, so I don't have an older version installed. Perhaps try to delete the `.h` files from the old installed version?

K4KDR, 2022-08-24:
Thank you - I would have never found that conflict!   Yes, I renamed the **/usr/local/include/satellites/** folder and the build-make-install process completed successfully.

Also, the **doppler_correction_example.grc** flowgraph runs properly now on my system:

```
Ubuntu 18.04
GNU Radio 3.8.1.0 (Python 3.6.9)
gr-satellites branch 'swig-doppler-correction'
```

daniestevez, 2022-08-25:
Thanks for testing. I have now merge the fix to the maint-3.8 branch, so I am closing the issue.

PR #399 (2022-08-25): Fix SWIG bindings for Doppler correction block
The Doppler correction block used a std::string& parameter to indicate
the filename, but this gives problems with the SWIG bindings. Therefore,
we use const char* instead.

This fixes #398
commit 03cdc4f3 (2022-08-25): Fix SWIG bindings for Doppler correction block

The Doppler correction block used a std::string& parameter to indicate
the filename, but this gives problems with the SWIG bindings. Therefore,
we use const char* instead.

This fixes #398
closed 2022-08-25 by daniestevez